USC-ACTLab/crazyswarm #1: Build Error

mjc401, 2017-01-10:
I am interested in using your setup for the crazyswarm.  I pulled the master branch and added the Vicon sdk files to the externalDependencies folder as specified.  I have been running into some issues during the build script:

1. The build script still references some of the submodules that were removed, e.g., vicon_ros

1. The crazyflie_driver cmake and package.xml still references vicon_ros, & the crazyflie_server.cpp file includes "vicon_ros/NamedPoseArray.h".

1. I commented all those out, since vicon_ros appears to be a submodule that is no longer used, but during the build I am getting many c++ errors, such as

/usr/include/c++/4.8/bits/basic_ios.h:66:11: note: ‘std::basic_ios<char>::basic_ios(const std::basic_ios<char>&)’ is implicitly deleted because the default definition would be ill-formed:
     class basic_ios : public ios_base
           ^
In file included from /usr/include/c++/4.8/ios:42:0,
                 from /usr/include/c++/4.8/ostream:38,
                 from /usr/include/c++/4.8/iostream:39,
                 from /opt/ros/indigo/include/ros/time.h:54,
                 from /opt/ros/indigo/include/ros/ros.h:38,
                 from /home/rasl/catkin_ws/src/crazyswarm/ros_ws/src/crazyflie_ros/crazyflie_driver/src/crazyflie_server.cpp:1:
/usr/include/c++/4.8/bits/ios_base.h:792:5: error: ‘std::ios_base::ios_base(const std::ios_base&)’ is private
     ios_base(const ios_base&);
     ^

I was wondering what version of c++ you were using, or if had an idea of what the underlying issue may be.  I tried with gcc 4.8.4 and 4.9 with the same errors.  I've had success interfacing with the standard crazyflie_ros driver you created, and found that it worked very well.

whoenig, 2017-01-10:
* We are using Ubuntu 16.04 with ROS Kinetic. That includes gcc 5.4.0. That being said, we had earlier versions running with Indigo just fine and I can't think of a reason why it shouldn't work with Indigo in the latest version.

* It looks like the master-branch didn't reference the correct/latest version of the crazyflie_ros submodule. I fixed that and the errors which you manually corrected should be gone now.

* For your issue, make sure you run a "catkin_make clean" before you run "catkin_make" in "crazyswarm/ros_ws". Also, don't checkout the crazyswarm repository as part of a ROS workspace. The repository contains it's own ROS workspace (namely in the ros_ws folder). Putting a workspace inside another workspace might create some unwanted side-effects. (Not sure if that explains the errors you are seeing, but might be worth a shot.)

* If you are using OptiTrack, or you are interested in using another external localization system other than VICON, you can look at our "libmotioncapture" branch.

Hope that helps - let me know if you run into more issues!
Wolfgang

mjc401, 2017-01-10:
I'm still getting the same class errors.  There always while building the crazyflie_server.cpp file, and in the libobjectracker included files.  The vicon_ros issues are all fixed.

Let me know if there is anything else you think I should try or any outputs that might be useful.

Matt

whoenig, 2017-01-10:
Could you pipe the full output of catkin_make to a file and attach it here?

mjc401, 2017-01-10:
[build_error.txt](https://github.com/USC-ACTLab/crazyswarm/files/697286/build_error.txt)

whoenig, 2017-01-10:
Seems to be related to this: http://stackoverflow.com/questions/28775673/why-cant-i-move-stdofstream
If you remove all occurrences/uses of "m_outputCSVs" in crazyflie_server.cpp it might compile just fine even with an older gcc.

mjc401, 2017-01-10:
Commenting the instances of m_outputCSV looks to have fixed the issue.  I'm no longer getting any errors during the build.  

Thanks for all the help,

Matt

closed 2017-01-10 by mjc401