USC-ACTLab/crazyswarm #2: Setup

mjc401, 2017-01-11:
Hi Wolfgang,

I have flashed the nrf and crazyflie-firmwares to the crazyflie, but I have a few questions regarding the setup in getting everything flying:

1. Is there any further setup that needs to be done (dependencies) or files that need to be changed to account for our configuration?

I now to change the radio channels is in the crazyflies.yaml file.  The reason I ask is that I am getting an rosparam error when trying to launch the crazy swarm_hover_swarm.launch file.  The full error is as follows:

load_parameters: unable to set parameters (last param was [/crazyflie_server/markerConfigurations/0/points/2=[-0.0328889, -0.02757, 0.0390601]]): cannot marshal None unless allow_none is enabled

I didn't know if the vicon marker points needed to be set or any other configuration settings that need to be changed.

1. When I run chooser.py, anytime I hit one of the buttons on the bottom, the channel value changes from what I set it to, e.g., I set it to channel 80, check the battery and the yaml file resets the value to channel 110.

- Matt

whoenig, 2017-01-11:
* The address of each Crazyflie must match the following format 0xE7E7E7E7<x>, where <x> is the crazyflie id in hex. E.g. cf10 will be 0xE7E7E7E70A. You can change the address using the cfclient.
* For initial testing, you should use VICON tracker with one CF. Enable "use_vicon_tracker" (https://github.com/USC-ACTLab/crazyswarm/blob/master/ros_ws/src/crazyswarm/launch/hover_swarm.launch#L48) and name the object in Tracker "cf01" (or whatever id you are using). Also enable just this one cf in your crazyflies.yaml file.
* To use the pointcloud-mode, switch "use_vicon_tracker" to false. Adjust the position of your markers by placing the CF in the center of your MoCap area facing forward and change the values accordingly here: https://github.com/USC-ACTLab/crazyswarm/blob/master/ros_ws/src/crazyswarm/launch/hover_swarm.launch#L54-L60
* The error you are seeing seems weird to me - I haven't seen that one before. You can follow the tip at http://answers.ros.org/question/165686/how-to-solve-the-error-annot-marshal-none-unless-allow_none-is-enabled/ and comment out a few lines in the launch file to find the offending line and report back.
* chooser.py actually reads https://github.com/USC-ACTLab/crazyswarm/blob/master/ros_ws/src/crazyswarm/launch/all49.yaml and dynamically creates crazyflies.yaml. So you should edit all49.yaml instead (the name is misleading - of course you don't need to have 49 CFs; any number will do).

mjc401, 2017-01-13:
I was able to get the hover and figure 8 script working for multiple crazyflies.  There was a ROS path issue that was causing the prior error.  Thanks for explaining the setup.  

One note:  When I am flashing the nrf-firmware, I am getting an unending stream of "Error:  written patch does not match! Retry" and the crazyflie never reaches reset in firmware mode.  Did you ever come across this issue?  I did not have this for the first crazyflie I used.  I have been able to use these crazyflies with your code, so it doesn't seem to affect anything.  

Otherwise, this issue can be closed.  Great job on the code, it works great.  I'm looking forward to using it.

whoenig, 2017-01-17:
Sorry for the late response. I am glad you got it working on your side!

For the error you are seeing: We updated the cfclient in an effort to do flash&verify (the default client just flashes and does not verify). However, the code is not very stable and you might just ran into a bug. We are currently working on a cleaner (and faster version), which we'll hopefully have upstream in ~1month. In practice you can just ignore the issue or use the official client for flashing.

closed 2017-01-18 by mjc401