USC-ACTLab/crazyswarm #3: Crazyflie channel setup

arjavdesai, 2017-01-17:
Hi,

I am currently trying to use your software for coordinating multiple crazyflies. At the moment, I am trying to acheive hover with 10 crazyflies. When I launch chooser.py and try to read the voltage values, I get a 'timeout' message for two crazyflies  (id:10 and 11).  However, the voltage values are available when we use crazyflie tools. I have ensured that all the radio channel values correspond to the correct robot id in 'all49.yaml'. For your convenience, the 'all49.yaml' file is attached  (as a.txt file) . Currently,  robot ids [1, 3, 4, 5, 6, 7, 8, 9, 11] are being used. Can you please let me know what the issue might be? 

all49 file: 
[all49.txt](https://github.com/USC-ACTLab/crazyswarm/files/712259/all49.txt)

Another issue was noticed during channel assignment to individual crazyflies. I followed the following procedure:

 First - assigned the channels using the setChannel functionality available in crazyflie_tools. 

 Second - verified that the channel has actually been assigned using the crazyflie gui interface.

However, while testing with chooser.py, I noticed that the channel numbers were not really assigned. 

E.g. I changed the channel of a crazyflie from 80 to 100, verified that the channel was changed using the  GUI interface. However, on running chooser.py, I was not able to read the voltage values. However, when I changed the channel number back to 80 in the 'all49.yaml' file,  I was able to get the voltage values. Can you please let me know what the problem might be?

Thanks.
Arjav

whoenig, 2017-01-17:
Hi Arjav,

About chooser.py issue: The script actually just runs crazyflie_tools commands internally and parses the output. As a first step, make sure that you either delete unused entries in all49.yaml (it doesn't need to have 49) or only select the ones you use in the GUI. Also, please provide which command line actually worked for you, otherwise it is hard for me to judge if the channel and address on the CF is set correctly.

About the channel assignment: Unfortunately, the setChannel script doesn't work very well in practice. First, it will only set the channel until you reboot - after a reboot it will have the default channel again. Second, it changes the channel in an best-effort way only, so there is no guarantee that it actually goes through.
Instead, I recommend using the cfclient (which changes a setting in the EEPROM, i.e. it will use that channel even after a reboot) or the following script (adjusted for your needs): https://github.com/bitcraze/crazyflie-lib-python/blob/master/examples/write-eeprom.py. Also, it is easiest to actually connect the CF using a USB cable for that purpose, because the official python client doesn't deal very well with possible packet drops for the channel changing at the moment.

Hope that helps!
Wolfgang

arjavdesai, 2017-01-17:
Hi Wolfgang, 

Thanks a lot for your response.

The command which works from the terminal is as follows (for robot id 11, channel 100):

rosrun crazyflie_tools battery --uri radio://0/100/2M/E7E7E7E711

However, chooser.py times out.

Arjav

whoenig, 2017-01-17:
You will need to use hexadecimal notation, so robot with id 10 should actually use the address E7E7E7E70A. Or you keep what you have, but then you ID in all49.yaml is 16 and 17 (rather than 10 or 11).

arjavdesai, 2017-01-17:
Hi Wolfgang,

Thanks a lot! That worked. Closing this issue.

Arjav

closed 2017-01-17 by arjavdesai