USC-ACTLab/crazyswarm #4: Point Cloud Object Tracker

mjc401, 2017-01-31:
Hi Wolfgang,

I have some questions regarding getting the marker configuration for the point cloud tracking.  I set the crazyflie at the origin and made a model.  Should I use the local marker locations that the tracker provides or the absolute locations?  Additionally, what are the offset values in relation towards?  Any help you could provide on how to set up the point cloud tracker would be appreciated.

Matt

whoenig, 2017-01-31:
Hi Matt,

The coordinates are in the global coordinate system, assuming that the CF is facing forward (=aligned with x-axis). Here, the offset defines the center of mass of the CF. For example, if you place the CF at the origin, the offset would be [0,0,0], but if you place it at [1,0,0], the offset should be [-1,0,0].

If the performance is not as good as with VICON Tracker, your center of mass might be too low. We achieved the best performance by setting it slightly above the battery. You can achieve that by decreasing the z-part of the offset.

Hope that helps,
Wolfgang

arjavdesai, 2017-01-31:
Hi Wolfgang,

Thanks a lot for your response.Based on your previous comment, we defined the marker configurations in the global coordinate system. 

The main issue that we are facing at the moment is that the ICP algorithm doesn't seem to converge when we launch hover_swarm.launch. The message printed in the terminal is as follows:

[pcl::IterativeClosestPoint::computeTransformation] Not enough correspondences found. Relax your threshold parameters.
[ WARN] [1485892013.271483293]: ICP did not converge!
[ WARN] [1485892013.271513546]: No updated pose for CF cf2 for 0.010167 s.

Further, the model does not show up in RVIZ. 

Can you please give us some pointers on how to solve this issue? Please let me know if you require any other information from my side.

Thanks.
Arjav

whoenig, 2017-01-31:
What are your relevant entries in hover_swarm.launch? Also, do you see the markers in RVIZ? Currently, only unlabled markers are tracked, i.e. make sure you have no object enabled in VICON Tracker (and you see the respective unlabled markers in Tracker as well.)

jpreiss, 2017-09-19:
@mjc401 @arjavdesai I am trying to close dead issues. Are you still having any problems related to this?

arjavdesai, 2017-09-20:
@jpreiss Hi. No, we are not having any more problems. You can close the issue. Thanks for your help.

closed 2017-09-20 by jpreiss