USC-ACTLab/crazyswarm #5: Build error

byronxubin, 2017-04-12:
Hi Wolfgang,

Your work is really amazing. I have just followed the instructions as you said. But unexpected error came out when I build in the workspace. The build errors are as follows. It is about the crazyflie_ros.
```
/home/binx/crazyswarm/ros_ws/src/crazyflie_ros/crazyflie_manager/src/manager.cpp:10:47: fatal error: crazyflie_driver/UploadTrajectory.h: No such file or directory
compilation terminated.
crazyflie_ros/crazyflie_manager/CMakeFiles/crazyflie_manager.dir/build.make:62: recipe for target 'crazyflie_ros/crazyflie_manager/CMakeFiles/crazyflie_manager.dir/src/manager.cpp.o' failed
make[2]: *** [crazyflie_ros/crazyflie_manager/CMakeFiles/crazyflie_manager.dir/src/manager.cpp.o] Error 1
CMakeFiles/Makefile2:3806: recipe for target 'crazyflie_ros/crazyflie_manager/CMakeFiles/crazyflie_manager.dir/all' failed
make[1]: *** [crazyflie_ros/crazyflie_manager/CMakeFiles/crazyflie_manager.dir/all] Error 2
make[1]: *** Waiting for unfinished jobs....
/home/binx/crazyswarm/ros_ws/src/crazyflie_ros/crazyflie_controller/src/controller.cpp:9:58: fatal error: crazyflie_controller/ExecuteTrajectoryAction.h: No such file or directory
compilation terminated.
crazyflie_ros/crazyflie_controller/CMakeFiles/crazyflie_controller.dir/build.make:62: recipe for target 'crazyflie_ros/crazyflie_controller/CMakeFiles/crazyflie_controller.dir/src/controller.cpp.o' failed
make[2]: *** [crazyflie_ros/crazyflie_controller/CMakeFiles/crazyflie_controller.dir/src/controller.cpp.o] Error 1
CMakeFiles/Makefile2:4619: recipe for target 'crazyflie_ros/crazyflie_controller/CMakeFiles/crazyflie_controller.dir/all' failed
make[1]: *** [crazyflie_ros/crazyflie_controller/CMakeFiles/crazyflie_controller.dir/all] Error 2
[  9%] Linking CXX executable /home/binx/crazyswarm/ros_ws/devel/lib/crazyflie_tools/reboot
[  9%] Linking CXX executable /home/binx/crazyswarm/ros_ws/devel/lib/crazyflie_tools/packetDropTest
[  9%] Built target packetDropTest
[  9%] Built target reboot
Makefile:138: recipe for target 'all' failed
make: *** [all] Error 2
Invoking "make -j4 -l4" failed
```
It seems like that the srv files in crazyflie_driver do not really converted to the header files which could be included in cpp.

byronxubin, 2017-04-12:
Hi,

I've just modified somewhere in two `CMake.txt` in both `crazyflie_manager` and `crazyflie_controller`. I thinks there might be something wrong you have to check. 

In the `CMake.txt` of `crazyflie_manager`, `add_dependencies` is missing. I think it should be like this.
```
add_dependencies(crazyflie_manager
 crazyflie_driver_generate_messages_cpp)
```

In the `CMake.txt` of `crazyflie_controller`, 
```
add_action_files(
  FILES
  ExecuteTrajectory.action
)
```
should be commented out.

Thanks

byronxubin, 2017-04-12:
One more issue is that there is some warnings came out. Looks like this:
```
In file included from /home/binx/crazyswarm/ros_ws/src/crazyflie_ros/crazyflie_driver/src/crazyflie_server.cpp:57:0:
/home/binx/crazyswarm/ros_ws/src/crazyflie_ros/externalDependencies/libobjecttracker/include/libobjecttracker/cloudlog.hpp: In member function ‘void libobjecttracker::PointCloudPlayer::play(libobjecttracker::ObjectTracker&) const’:
/home/binx/crazyswarm/ros_ws/src/crazyflie_ros/externalDependencies/libobjecttracker/include/libobjecttracker/cloudlog.hpp:103:57: warning: format ‘%d’ expects argument of type ‘int’, but argument 2 has type ‘std::size_t {aka long unsigned int}’ [-Wformat=]
     printf("\n  %d  ------------------------------\n", i);
                                                         ^
/home/binx/crazyswarm/ros_ws/src/crazyflie_ros/externalDependencies/libobjecttracker/include/libobjecttracker/cloudlog.hpp: In member function ‘void libobjecttracker::PointCloudDebugger::convert(libobjecttracker::ObjectTracker&, std::vector<boost::shared_ptr<pcl::PointCloud<pcl::PointXYZ> > >&)’:
/home/binx/crazyswarm/ros_ws/src/crazyflie_ros/externalDependencies/libobjecttracker/include/libobjecttracker/cloudlog.hpp:139:57: warning: format ‘%d’ expects argument of type ‘int’, but argument 2 has type ‘std::size_t {aka long unsigned int}’ [-Wformat=]
     printf("\n  %d  ------------------------------\n", i);
                                                         ^
/home/binx/crazyswarm/ros_ws/src/crazyflie_ros/externalDependencies/libobjecttracker/include/libobjecttracker/cloudlog.hpp:152:55: warning: format ‘%d’ expects argument of type ‘int’, but argument 2 has type ‘std::vector<libobjecttracker::Object>::size_type {aka long unsigned int}’ [-Wformat=]
      printf("Object vector size: %d\n", objects.size());
```

I hope it would not be a big deal.

jpreiss, 2017-09-19:
@binxxx I am trying to close dead issues. Are you still having any trouble related to this?

closed 2017-12-13 by jpreiss