USC-ACTLab/crazyswarm #6: Crazyradio Make error

byronxubin, 2017-04-12:
Hi,

After flashing the `Firmware(NRF)`, I got an error when wanting to `make CRPA=1`. The error message is as follows. Could you help with this problem? Without the update of CrazyRadio, I am not able to use other parts. Thanks in advance.

```
binx@binx-ubuntu:~/crazyswarm/crazyradio-firmware/firmware$ make CRPA=1
sdcc -Iinc/ --model-large --std-sdcc99 -DCRPA -c src/main.c -o bin/main.rel
make: sdcc: Command not found
Makefile:65: recipe for target 'main.rel' failed
make: *** [main.rel] Error 127
```

whoenig, 2017-04-13:
Try `sudo apt-get install sdcc binutils`

byronxubin, 2017-04-13:
Thanks @whoenig . I've installed sdcc and then it works. One more thing is that for the last procedure, `python ../usbtools/nrfbootload.py flash bin/cradio.bin`, I would need add `sudo` in front to make it work. 

I would close this issue.

closed 2017-04-13 by byronxubin