USC-ACTLab/crazyswarm #7: Build Script

lakehanne, 2017-05-06:
Hi there,

It seems one of the git repos that you recursively clone with the `build.sh` script is either not available or is a private repo. 

When it `cd`s into the `externalDependencies/` directory, it tries to clone a `phasespace_sdk` submodule but then it prompts for authentication access:

```bash
Cloning into 'externalDependencies/phasespace_sdk'...
Username for 'https://github.com': lakehanne
Password for 'https://lakehanne@github.com': 
remote: Repository not found.
fatal: repository 'https://github.com/USC-ACTLab/phasespace_sdk.git/' not found
fatal: clone of 'https://github.com/USC-ACTLab/phasespace_sdk.git' into submodule path 'externalDependencies/phasespace_sdk' failed
```

It also does not find `object_tracker/:` e.g.

```bash
./build.sh: line 17: cd: ros_ws/src/object_tracker/: No such file or directory
```

My second question is why didn't you just use the ros interface to `vicon_sdk` available [here](https://github.com/ethz-asl/ros-drivers/tree/master/vicon_bridge) instead of requiring the download of the sdk from vicon itself?

whoenig, 2017-05-07:
The two dependencies which are not included are VICON and Phasespace SDKs; both are excluded for legal reasons. Phasespace is not built by default, so you won't need that repository. For Vicon follow the instruction on github or http://crazyswarm.readthedocs.io/en/latest/installation.html.

I am not sure about your problem with the missing object_tracker. The installation instructions on readthedocs were recently created on a freshly installed machine.

We don't use vicon_bridge because it doesn't provide low-level access. Our solution gets the raw marker point cloud and does the tracking using ICP.

lakehanne, 2017-05-13:
I am either missing something or something is not right. The build script cd's into each subfolder within crazyswarm and attempts to recursively clone the submodules. If the `phasesdk` is not required, why did you include it as part of the installation script? The script prompts for authentication credentials when it gets to the instructions for `phase_sdk` and since the repo is either not public or non-existent, users' credentials do not suffice in cloning the repo.

Aside from that, I did set up the vicon libraries and include files as you described [here](http://crazyswarm.readthedocs.io/en/latest/installation.html) but still catkin_make fails in finding `ExecuteTrajectoryAction.h`. I did a quick search on your repo and this header is not included so I am assuming it is a message_generation file or something.

Here are the relevant portions of the trace:

```bash
home/hivemind/crazyswarm/ros_ws/src/crazyflie_ros/externalDependencies/libmotioncapture/src/vicon.cpp:4:30: fatal error: vicon_sdk/Client.h: No such file or directory
compilation terminated.
crazyflie_ros/externalDependencies/libmotioncapture/CMakeFiles/libmotioncapture.dir/build.make:86: recipe for target 'crazyflie_ros/externalDependencies/libmotioncapture/CMakeFiles/libmotioncapture.dir/src/vicon.cpp.o' failed
make[2]: *** [crazyflie_ros/externalDependencies/libmotioncapture/CMakeFiles/libmotioncapture.dir/src/vicon.cpp.o] Error 1
make[2]: *** Waiting for unfinished jobs....
```

It's looking for `Client.h` in `vicon_sdk` and not `vicon_sdk/include` as your instructions describe. Also, 

```bash
[  5%] Building CXX object crazyflie_ros/crazyflie_controller/CMakeFiles/crazyflie_controller.dir/src/controller.cpp.o
CMakeFiles/Makefile2:1049: recipe for target 'crazyflie_ros/externalDependencies/libmotioncapture/CMakeFiles/libmotioncapture.dir/all' failed
make[1]: *** [crazyflie_ros/externalDependencies/libmotioncapture/CMakeFiles/libmotioncapture.dir/all] Error 2
make[1]: *** Waiting for unfinished jobs....
/home/hivemind/crazyswarm/ros_ws/src/crazyflie_ros/crazyflie_controller/src/controller.cpp:9:58: fatal error: crazyflie_controller/ExecuteTrajectoryAction.h: No such file or directory
compilation terminated.
crazyflie_ros/crazyflie_controller/CMakeFiles/crazyflie_controller.dir/build.make:62: recipe for target 'crazyflie_ros/crazyflie_controller/CMakeFiles/crazyflie_controller.dir/src/controller.cpp.o' failed
make[2]: *** [crazyflie_ros/crazyflie_controller/CMakeFiles/crazyflie_controller.dir/src/controller.cpp.o] Error 1
CMakeFiles/Makefile2:4619: recipe for target 'crazyflie_ros/crazyflie_controller/CMakeFiles/crazyflie_controller.dir/all' failed
make[1]: *** [crazyflie_ros/crazyflie_controller/CMakeFiles/crazyflie_controller.dir/all] Error 2
```

lakehanne, 2017-05-13:
It seems it is looking for vicon_sdk in `crazyflie_ros/externalDependencies/libmotioncapture`. Placing the files there seems to fix the Client.h issue. One other thing that I think should be addressed is that the vicon sdk downloaded files does not have `Client.h` but rather a file named `DataStreamClient.h`. Renaming this header to Client.h seems a good enough fix for now.

whoenig, 2017-05-21:
Hi,
Thank you for your detailed analysis! If you already made the changes, a PR would be very helpful for issues 3 and 4.
Regarding 1 and 2: Which SDK are you using? We tested with 1.6, and things might have changed slightly with the newer version.
Thanks,
Wolfgang

lakehanne, 2017-05-21:
> a PR would be very helpful for issues 3 and 4.
Yes, I will be sending the PRs to the submodules that 3 and 4 affect. Thanks!

> Regarding 1 and 2: Which SDK are you using? We tested with 1.6, and things might have changed slightly with the newer version.

I downloaded the most recent sdk from the vicon website: vicon_sdk 1.7. Perhaps this was the cause of the error and it is perhaps worth adding a caveat in the documentation about this.

closed 2017-05-22 by lakehanne