USC-ACTLab/crazyswarm #8: UWB compatibility

4rzael, 2017-05-11:
Bitcraze recently released the Loco Positioning System, a tracking system for the crazyflie working with the Ultra-Wide-Band technology.

It now supports (in TdoA experimental mode) multiple crazyflies.

Does crazyswarm support the Loco Positioning System instead of VICON ? And if not, do you think achieving the compatibility between thoses two would be hard ?

whoenig, 2017-05-11:
The Crazyswarm does not support UWB at the moment. It should be possible by merging our firmware changes to the Bitcraze crazyflie-firmware master branch (and adding a few other things to enable switching modes). It is a long-term goal for us to have that support, but we can't give any ETA on it. Any help would be welcome of course.

jpreiss, 2017-09-19:
Closing. We are working towards merging the Crazyswarm fork with the main Bitcraze firmware, which would be the first step towards UWB support.

closed 2017-09-19 by jpreiss

NicksonYap, 2018-12-16:
@jpreiss @whoenig 
now that the the fork is merged, can Crazyswarm now support UWB / DWM1000 / Loco deck?

Thanks!