USC-ACTLab/crazyswarm #9: crazyswarm Pointcloud issues

txlin, 2017-05-23:
We're currently in the process of trying to run the crazyswarm stack but we're hitting a few issues. When we run hover_swarm.launch no pointcloud and no tf data is available. 

A potential issue we thought of is that we're running vicon tracker on a separate computer in the network so we thought vicon_host_name: was meant for the ip address but when changing it we get a std::runtime_error. 

It seems like we've followed the setup and documentation appropriately, is there something else we're missing in setup?

whoenig, 2017-05-23:
Did you follow the documentation on readthedocs (http://crazyswarm.readthedocs.io/en/latest/). What you are describing sounds indeed like a communication issue. The typical setup is to have two machines, one Windows and one Linux box on the same network. You can try your setup with the vicon_bridge package (https://github.com/ethz-asl/vicon_bridge). Also make sure that the ports are open (disable the firewall on your windows box) and that you can ping the vicon machine from the linux box.

vss2sn, 2017-05-26:
We've checked the setup by pinging the host machine and using Vicon bridge. Is there some place we could specify the IP address of the host machine?

jpreiss, 2017-09-19:
@txlin @vss2sn I am trying to close dead issues. Is your Vicon communication working properly now?

closed 2017-09-19 by txlin