USC-ACTLab/crazyswarm #10: Update params

vss2sn, 2017-06-05:
We're currently trying to run crazyswarm but we're having a bit of trouble. While we are able run hover_swarm.launch and see the crazyflie in rviz, the crazyflie does not move. The channel and the marker configuration have both been set as shown in the documentation. We get the following errors:

[ INFO] [1496673760.698281845]: [cf1] Update parameters
[ERROR] [1496673760.698336362]: Could not find param ctrlMel/i_range_m_z
[ERROR] [1496673760.698350819]: Could not find param ctrlMel/i_range_xy
[ERROR] [1496673760.698360302]: Could not find param ctrlMel/i_range_z
[ERROR] [1496673760.698369223]: Could not find param ctrlMel/kR_xy
[ERROR] [1496673760.698377789]: Could not find param ctrlMel/kR_z
[ERROR] [1496673760.698387129]: Could not find param ctrlMel/kd_omega_rp
[ERROR] [1496673760.698396218]: Could not find param ctrlMel/kd_xy
[ERROR] [1496673760.698407280]: Could not find param ctrlMel/kd_z
[ERROR] [1496673760.698415656]: Could not find param ctrlMel/ki_m_z
[ERROR] [1496673760.698424105]: Could not find param ctrlMel/ki_xy
[ERROR] [1496673760.698432548]: Could not find param ctrlMel/ki_z
[ERROR] [1496673760.698440915]: Could not find param ctrlMel/kp_xy
[ERROR] [1496673760.698449300]: Could not find param ctrlMel/kp_z
[ERROR] [1496673760.698460612]: Could not find param ctrlMel/kw_xy
[ERROR] [1496673760.698470677]: Could not find param ctrlMel/kw_z
[ERROR] [1496673760.698478898]: Could not find param ctrlMel/mass
[ERROR] [1496673760.698487425]: Could not find param ctrlMel/massThrust
[ERROR] [1496673760.698496481]: Could not find param flightmode/posCtrl
[ INFO] [1496673760.719825891]: Update params: 0.036021 s

Could you give any pointers as to what the issue might be, and its solution? Thanks in advance.

lakehanne, 2017-06-06:
I also have problems with the pid gains of the crazyflie. I am using version 2.0. Each time I fly, the swarms shoot up so high that they crash after a few minutes of launch. How do you set up the robots in your environment such that they work just fine?

whoenig, 2017-06-06:
@vss2sn Please update the firmware on your Crazyflie with the one provided in the crazyswarm repository. The required parameters are not part of the stock firmware.

@lakehanne Please open another issue for that. The gains are pre-tuned for a CF 2.0 with a total mass of 33g. If it happens "after a few minutes" maybe there is a tracking issue?

vss2sn, 2017-06-07:
Thanks a lot! That worked; a ppreciate the quick response.

closed 2017-06-07 by vss2sn

byronxubin, 2017-06-07:
@vss2sn  What is the right way to update the firmware. I follow this procedure as follows. But after I have updated the firmware, I can still find the parameter errors. Could you give me some advice? 

```
git clone https://github.com/USC-ACTLab/crazyflie-firmware.git
git submodule init
git submodule update

make

make cload
```

whoenig, 2017-06-08:
You need to use the firmware as provided in the crazyswarm repository (specifically, that uses a different branch).

`
crazyswarm$ cd crazyflie-firmware
make
make cload
`