USC-ACTLab/crazyswarm #11: Build Script Issue: Missing Object Tracker

kgslovak, 2017-06-06:
Hey guys,

First and foremost, thank you so much for making your project open source and well documented. The effort you all put into making your work reproducible is awesome. 

I am trying to use the build.sh script as instructed on the installation page of the readthedocs site, but i'm running into issues. I'll just start with my first issue in this post: submodule handling in the build script.

The submodule init and update for ".../externalDependencies/libmotioncapture/" pulls up the github credentials prompt. I just commented it out to avoid the issue, but thought I'd mention it because it seems unnecessary.

The submodule init and update for "ros_ws/src/object_tracker/" has me really confused though: The src folder on github only contains the crazyflie_ros and crazyswarm directories...

Was object_tracker intentionally removed or do I need to clone it from another project? Is commenting this block out going to cause me any problems?

Thanks,
Kyle

whoenig, 2017-06-06:
Hi Kyle,
The update in libmotioncapture pulls public and private repositories, so just commenting out this line might not work. Some repos are private for licensing issues. Our current recommendation is to just ctrl+c out of the credential and manually copy the files from the respective SDKs. I agree, that this is not very nice. However, you only need to use build.sh initially. Afterwards, you can just execute `catkin_make`.

Your second issue was a bug and I just pushed a fix. The object_tracker resides in libobjecttracker now and is simply in a different folder.

Cheers,
Wolfgang

kgslovak, 2017-06-07:
Good to know, thanks for the information and the fix Wolfgang.

Kyle

closed 2017-06-07 by kgslovak