USC-ACTLab/crazyswarm #12: Submodule Link: crazyflie_ros

kgslovak, 2017-06-07:
I noticed the submodule link for the ROS-crazyflie library is: "crazyflie_ros @ bac1f8f". Should it point to the "crazyswarm_master" branch instead?

The bac1f8f tree does not include the fixes committed to the CMakeList.txt files of crazyflie_driver and crazyflie_manager.

Thanks,
Kyle

kgslovak, 2017-06-08:
`git submodule update` yields the following: 

Submodule path 'ros_ws/src/crazyflie_ros': checked out 'bac1f8f19237a1bed8bcffd0c0f60e19a074f41f'

If this correct?

Thanks,
Kyle

whoenig, 2017-06-09:
The head of `crazyswarm_master` should be fine. Git submodules point to a specific version rather than a branch, which can be annoying (an update to crazyflie_ros also requires to update crazyswarm to point to the newest version).

kgslovak, 2017-06-09:
Okay. Are the changes to the CMakeLists of crazyflie_manager and crazyflie_driver not supposed to show up in my files? I tried `git submodule update` in my crazyswarm directory after going through the build script, but the changes are still not included.

whoenig, 2017-06-12:
I didn't update the crazyswarm project itself yet. You would need to manually run `git checkout crazyswarm_master` in `ros_ws/src/crazyflie_ros` to fetch those changes.

kgslovak, 2017-06-12:
Oh okay, got it. Thanks Wolfgang

closed 2017-06-12 by kgslovak