USC-ACTLab/crazyswarm #13: runtime error when launch hover_swarm

byronxubin, 2017-06-07:
Hi, I've got an error when launching the launch file of crazyswarm. I really don't know which part is wrong here. I think I've followed the procedure in the documentation page. 
```
[ INFO] [1496861669.955738149]: Adding CF: cf1 (radio://0/100/2M/E7E7E7E701, cf1)...
[ INFO] [1496861669.968280750]: CF ctor: 0.012442 s
terminate called after throwing an instance of 'std::runtime_error'
  what():  timeout
[crazyflie_server-2] process has died [pid 30320, exit code -6, cmd /home/binx/crazyswarm/ros_ws/devel/lib/crazyflie_driver/crazyflie_server __name:=crazyflie_server __log:=/home/binx/.ros/log/bc3ae7da-4bb2-11e7-b2e0-e4a4714af4e3/crazyflie_server-2.log].
log file: /home/binx/.ros/log/bc3ae7da-4bb2-11e7-b2e0-e4a4714af4e3/crazyflie_server-2*.log
``

whoenig, 2017-06-08:
That happens if no connection can be established. Try connecting to the CF using the official Python client and your uri (radio://0/100/2M/E7E7E7E701). Make sure that your CF is configured with the specified channel, bandwidth, and address.

glennliu, 2017-07-13:
Hi 

I met similar problem. When I start hover_swarm.launch, the terminal display as below, 

```
PARAMETERS
 * /crazyflie_manager/csv_file: /home/xiaohao/bit...
 * /crazyflie_manager/timescale: 0.8
 * /crazyflie_server/broadcast_uri: radio://0/100/2M/...
 * /crazyflie_server/dynamicsConfigurations/0/maxFitnessScore: 0.001
 * /crazyflie_server/dynamicsConfigurations/0/maxPitch: 1.4
 * /crazyflie_server/dynamicsConfigurations/0/maxPitchRate: 20.0
 * /crazyflie_server/dynamicsConfigurations/0/maxRoll: 1.4
 * /crazyflie_server/dynamicsConfigurations/0/maxRollRate: 20.0
 * /crazyflie_server/dynamicsConfigurations/0/maxXVelocity: 2.0
 * /crazyflie_server/dynamicsConfigurations/0/maxYVelocity: 2.0
 * /crazyflie_server/dynamicsConfigurations/0/maxYawRate: 10.0
 * /crazyflie_server/dynamicsConfigurations/0/maxZVelocity: 3.0
 * /crazyflie_server/enable_logging: False
 * /crazyflie_server/enable_parameters: True
 * /crazyflie_server/firmwareParams/ctrlMel/i_range_m_z: 1500
 * /crazyflie_server/firmwareParams/ctrlMel/i_range_xy: 2.0
 * /crazyflie_server/firmwareParams/ctrlMel/i_range_z: 0.4
 * /crazyflie_server/firmwareParams/ctrlMel/kR_xy: 70000
 * /crazyflie_server/firmwareParams/ctrlMel/kR_z: 60000
 * /crazyflie_server/firmwareParams/ctrlMel/kd_omega_rp: 200
 * /crazyflie_server/firmwareParams/ctrlMel/kd_xy: 0.2
 * /crazyflie_server/firmwareParams/ctrlMel/kd_z: 0.4
 * /crazyflie_server/firmwareParams/ctrlMel/ki_m_z: 500
 * /crazyflie_server/firmwareParams/ctrlMel/ki_xy: 0.05
 * /crazyflie_server/firmwareParams/ctrlMel/ki_z: 0.05
 * /crazyflie_server/firmwareParams/ctrlMel/kp_xy: 0.4
 * /crazyflie_server/firmwareParams/ctrlMel/kp_z: 1.25
 * /crazyflie_server/firmwareParams/ctrlMel/kw_xy: 20000
 * /crazyflie_server/firmwareParams/ctrlMel/kw_z: 12000
 * /crazyflie_server/firmwareParams/ctrlMel/mass: 0.032
 * /crazyflie_server/firmwareParams/ctrlMel/massThrust: 132000
 * /crazyflie_server/firmwareParams/flightmode/posCtrl: 1
 * /crazyflie_server/firmwareParams/ring/effect: 14
 * /crazyflie_server/firmwareParams/ring/headlightEnable: 0
 * /crazyflie_server/firmwareParams/ring/solidBlue: 20
 * /crazyflie_server/firmwareParams/ring/solidGreen: 0
 * /crazyflie_server/firmwareParams/ring/solidRed: 0
 * /crazyflie_server/force_no_cache: False
 * /crazyflie_server/genericLogTopicFrequencies: [10]
 * /crazyflie_server/genericLogTopic_log1_Variables: ['stabilizer.yaw'...
 * /crazyflie_server/genericLogTopics: ['log1']
 * /crazyflie_server/interactive_object: active-wand
 * /crazyflie_server/markerConfigurations/0/numPoints: 4
 * /crazyflie_server/markerConfigurations/0/offset: [0.0, -0.01, -0.04]
 * /crazyflie_server/markerConfigurations/0/points/0: [0.0177184, 0.013...
 * /crazyflie_server/markerConfigurations/0/points/1: [-0.0262914, 0.05...
 * /crazyflie_server/markerConfigurations/0/points/2: [-0.0328889, -0.0...
 * /crazyflie_server/markerConfigurations/0/points/3: [0.0431307, -0.03...
 * /crazyflie_server/motion_capture_type: optitrack
 * /crazyflie_server/numDynamicsConfigurations: 1
 * /crazyflie_server/numMarkerConfigurations: 1
 * /crazyflie_server/object_tracking_type: libobjecttracker
 * /crazyflie_server/optitrack_local_ip: 10.0.0.4
 * /crazyflie_server/optitrack_server_ip: 10.0.0.2
 * /crazyflie_server/print_latency: False
 * /crazyflie_server/save_point_clouds: ~/pointCloud.ot
 * /crazyflie_server/world_frame: /world
 * /crazyflie_server/write_csvs: False
 * /crazyflies: [{'initialPositio...
 * /rosdistro: kinetic
 * /rosversion: 1.12.7

NODES
  /
    crazyflie_manager (crazyflie_manager/crazyflie_manager)
    crazyflie_server (crazyflie_driver/crazyflie_server)
    rqt (rqt_gui/rqt_gui)
    rviz (rviz/rviz)

auto-starting new master
process[master]: started with pid [16418]
ROS_MASTER_URI=http://localhost:11311

setting /run_id to 71b13c3e-67d5-11e7-931c-408d5c095ae7
process[rosout-1]: started with pid [16431]
started core service [/rosout]
process[crazyflie_server-2]: started with pid [16448]
process[crazyflie_manager-3]: started with pid [16449]
process[rviz-4]: started with pid [16450]
[ INFO] [1499955209.547520707]: Wait for services...
[ INFO] [1499955209.547907452]: waitForService: Service [/emergency] has not been advertised, waiting...
process[rqt-5]: started with pid [16467]
[ INFO] [1499955209.589614202]: waitForService: Service [/emergency] is now available.
NatNetClient connecting: 10.0.0.4 to 10.0.0.2...
WARNING: Could not set receive buffer size. Asked for 1048576B got 0B
[ INFO] [1499955209.592903141]: Manager ready.
[Client] Server Software: Motive
[Client] NatNetVersion: 2.9
[Client] ServerVersion: 1.90
NatNet server version: 33554442
ch: 1
terminate called after throwing an instance of 'std::runtime_error'
  what():  timeout
[crazyflie_server-2] process has died [pid 16448, exit code -6, cmd /home/xiaohao/bitcraze/crazyswarm/ros_ws/devel/lib/crazyflie_driver/crazyflie_server __name:=crazyflie_server __log:=/home/xiaohao/.ros/log/71b13c3e-67d5-11e7-931c-408d5c095ae7/crazyflie_server-2.log].
log file: /home/xiaohao/.ros/log/71b13c3e-67d5-11e7-931c-408d5c095ae7/crazyflie_server-2*.log

```

I can connect to crazyflie through client, with radio setting,

> addresss: "0xE7E7E7E701"
> channel: "100"

I write swarm_hover.launch as below,
```
<node pkg="crazyflie_driver" type="crazyflie_server" name="crazyflie_server" output="screen">
    <rosparam>
      #broadcast_address: "FFE7E7E701"
      broadcast_uri: "radio://0/100/2M/E7E7E7E701"
      world_frame: "/world"
```
any advice appreciated

whoenig, 2017-07-14:
How does your crazyflies.yaml look like?

glennliu, 2017-07-17:
crazyflies.yaml as below,
crazyflies:
- channel: 120
  id: 45
  initialPosition: [-1.5, 0.5, 0.0]

whoenig, 2017-07-17:
If you can connect to 0xE7E7E701 and channel 100, you should change the crazyflies.yaml accordingly to

```
- channel: 100
  id: 1
```

Otherwise make sure you can connect using the client to address 0xE7E7E72D, channel 120.

glennliu, 2017-07-18:
I changed crazyflies.yaml
```
crazyflies:
- channel: 120
  id: 45
  initialPosition: [-1.5, 0.5, 0.0]
```
It still met "timeout()" error

glennliu, 2017-07-19:
Hi Whoenig

There are different channels and id numbers for each crazyflie. How does crazyflies.yaml work? It can only set the same paramters. 

Thanks
Chuhao

whoenig, 2017-07-20:
* If you can connect using the cfclient to addresss: "0xE7E7E7E701" and channel: "100", then the above should work.
* Crazyflies.yaml is a yaml file containing an array of all CFs you want to connect to. For testing, start with just a single entry. You can add more entries with different channels/ids to your liking.
* I believe the broadcast_uri is not used in the code - it is computed dynamically; Thus you can ignore the portion in hover_swarm.launch
* Try the command line tools: `roslaunch crazyflie_tools reboot` etc.

jpreiss, 2017-09-19:
@glennliu I am trying to close dead issues. Did you get communication working?

javieryu, 2017-10-10:
I am having the same issue, so far I have followed all of the advice on this thread. However, I am still running into the timeout error. I can also successfully connect to the CF with cfclient. 
[image]

whoenig, 2017-10-10:
How does crazyflies.yaml look like? Do the command line tools/python GUI (choser.py) work?

jpreiss, 2017-10-11:
@whoenig maybe we should modify the server to print a more meaningful message besides `std::runtime_error`...

anchekuri, 2017-10-12:
I am trying to fly 2 crazyflies with the same radio dongle and with different address to each crazyflie 

roslaunch bitcraze_lps_estimator dwm_loc_ekf_multi_hover.launch uri0:=radio://0/30/2M/E7E7E7E701 uri1:=radio://0/80/2M/E7E7E7E702

And I can connect the crazyflies to the CF client but when I run the above command the crazyflies are not connected showing an error status in the rviz. with an error 

[crazyflie0/crazyflie_add-3] process has finished cleanly
log file: /root/.ros/log/4bb9c042-aef1-11e7-ba7c-b88a60e5eba6/crazyflie0-crazyflie_add-3*.log
[ WARN] [1507773705.730807229]: Link Quality low (0.370000)

terminate called after throwing an instance of 'std::runtime_error'
  what():  timeout
[crazyflie_server-18] process has died [pid 3320, exit code -6, cmd /root/catkin_ws/devel/lib/crazyflie_driver/crazyflie_server __name:=crazyflie_server __log:=/root/.ros/log/4bb9c042-aef1-11e7-ba7c-b88a60e5eba6/crazyflie_server-18.log].
log file: /root/.ros/log/4bb9c042-aef1-11e7-ba7c-b88a60e5eba6/crazyflie_server-18*.log

^C[link1_broadcaster-17] killing on exit

I am having the same kind of error : Run time error time out 

can anyone know how to rectify this error 

any help will be appreciated thank you

whoenig, 2017-10-12:
If you want to use a single radio, the channels need to be identical (you seem to use channel 30 and channel 80). Also, how does crazyflies.yaml look like and can you connect to the CF using the GUI or the command line tools (e.g. `rosrun crazyflie_tools reboot --uri radio://0/30/2M/E7E7E7E701`)?

anchekuri, 2017-10-12:
If I use the same radio signal for 2 crazyflies there will not be any interference issues.

anchekuri, 2017-10-12:
roslaunch bitcraze_lps_estimator dwm_loc_ekf_multi_hover.launch uri0:=radio://0/30/2M/E7E7E7E701 uri1:=radio://0/30/2M/E7E7E7E702 
So you are saying that I can use the command for the launch file as the above like 30 for each..???

whoenig, 2017-10-12:
There won't be any interference. On the host side the radio is used in a round-robin fashion. On the CF side packets with the wrong address are ignored.

There is no support for bitcraze_lps_estimator dwm_loc_ekf_multi_hover.launch in the crazyswarm package (only hover_swarm.launch is supported). You should use the official crazyflie_ros instead. However, even in that case it is recommended to use the same channel for both CFs to avoid that the radio needs to switch channels.

anchekuri, 2017-10-12:
I am using the crazyflie_ros for this and make the radio channel same but the crazyflies are not recognized on the rviz screen 

i used this command 

roslaunch bitcraze_lps_estimator dwm_loc_ekf_multi_hover.launch uri0:=radio://0/30/2M/E7E7E7E701 uri1:=radio://0/30/2M/E7E7E7E702

I have three now. I even saw this earlier too 

I think If you show you the screen shot I will be clear for you

[image]

this was the error I have Can you check it once 

Don't know ack: Port: 13 Channel: 0 Len: 3
Don't know ack: Port: 15 Channel: 0 Len: 3
Don't know ack: Port: 15 Channel: 0 Len: 3
Don't know ack: Port: 13 Channel: 0 Len: 3
Don't know ack: Port: 13 Channel: 0 Len: 3
Don't know ack: Port: 15 Channel: 0 Len: 3
Don't know ack: Port: 15 Channel: 0 Len: 3
[INFO] [1507776040.534758]: Anchor 1 at [0, 2, 1.85]
[ INFO] [1507776040.539097293]: Update parameters
[INFO] [1507776040.641626]: Anchor 1 at [0, 2, 1.85]
[ INFO] [1507776040.642644976]: Update parameters
terminate called after throwing an instance of 'std::runtime_error'
  what():  Could not find acc.x in log toc!
Traceback (most recent call last):
  File "/root/catkin_ws/src/lps-ros/scripts/lps_ekf_bridge.py", line 77, in <module>
Traceback (most recent call last):
  File "/root/catkin_ws/src/lps-ros/scripts/lps_ekf_bridge.py", line 77, in <module>
    update_params([name + 'x', name + 'y', name + 'z'])
  File "/opt/ros/kinetic/lib/python2.7/dist-packages/rospy/impl/tcpros_service.py", line 435, in __call__
    update_params([name + 'x', name + 'y', name + 'z'])
  File "/opt/ros/kinetic/lib/python2.7/dist-packages/rospy/impl/tcpros_service.py", line 435, in __call__
    return self.call(*args, **kwds)
  **File "/opt/ros/kinetic/lib/python2.7/dist-packages/rospy/impl/tcpros_service.py", line 525, in call
    return self.call(*args, **kwds)
  File "/opt/ros/kinetic/lib/python2.7/dist-packages/rospy/impl/tcpros_service.py", line 525, in call
    raise ServiceException("transport error completing service call: %s"%(str(e)))
rospy.service.ServiceException: transport error completing service call: unable to receive data from sender, check sender's logs for details
    raise ServiceException("transport error completing service call: %s"%(str(e)))
rospy.service.ServiceException: transport error completing service call: unable to receive data from sender, check sender's logs for details
[crazyflie_server-18] process has died [pid 14526, exit code -6, cmd /root/catkin_ws/devel/lib/crazyflie_driver/crazyflie_server __name:=crazyflie_server __log:=/root/.ros/log/bac0f668-aef6-11e7-ba7c-b88a60e5eba6/crazyflie_server-18.log].
log file: /root/.ros/log/bac0f668-aef6-11e7-ba7c-b88a60e5eba6/crazyflie_server-18*.log
[crazyflie1/lps_efk_bridge-15] process has died [pid 14497, exit code 1, cmd /root/catkin_ws/src/lps-ros/scripts/lps_ekf_bridge.py __name:=lps_efk_bridge __log:=/root/.ros/log/bac0f668-aef6-11e7-ba7c-b88a60e5eba6/crazyflie1-lps_efk_bridge-15.log].
log file: /root/.ros/log/bac0f668-aef6-11e7-ba7c-b88a60e5eba6/crazyflie1-lps_efk_bridge-15*.log
[crazyflie0/lps_efk_bridge-8] process has died [pid 14446, exit code 1, cmd /root/catkin_ws/src/lps-ros/scripts/lps_ekf_bridge.py __name:=lps_efk_bridge __log:=/root/.ros/log/bac0f668-aef6-11e7-ba7c-b88a60e5eba6/crazyflie0-lps_efk_bridge-8.log].
log file: /root/.ros/log/bac0f668-aef6-11e7-ba7c-b88a60e5eba6/crazyflie0-lps_efk_bridge-8*.log**
^C[link1_broadcaster-17] killing on exit
[crazyflie1/pose-14] killing on exit
[crazyflie1/lps_viz-16] killing on exit
[crazyflie1/controller_bridge-13] killing on exit
[crazyflie1/joystick_controller-12] killing on exit
[crazyflie0/joystick_controller-5] killing on exit
[crazyflie0/lps_viz-9] killing on exit
[crazyflie0/pose-7] killing on exit[crazyflie1/joy-11] killing on exit

[crazyflie0/controller_bridge-6] killing on exit
[crazyflie0/joy-4] killing on exit
[rviz-2] killing on exit
[rosout-1] killing on exit
[master] killing on exit
shutting down processing monitor...
... shutting down processing monitor complete
done

Check the screen shot for the issue I was facing and I dont know I saw 2 rviz open toolbar

whoenig, 2017-10-12:
Please open a new issue for crazyflie_ros (https://github.com/whoenig/crazyflie_ros) instead of using this one. Also, in this new thread, provide some information on the firmware you use. The error you have indicates that your firmware doesn't have the "acc.x" logging variable, which is strange.

anchekuri, 2017-10-12:
@whoenig  I tried to install the new crazyflie_ros and run catkin_make in the workspace and ran the same command but it was showing the same error that 

[INFO] [1507776838.819946]: Anchor 2 at [4, 2, 1.85]
[ INFO] [1507776838.829502301]: Update parameters
terminate called after throwing an instance of 'std::runtime_error'
  what():  Could not find acc.x in log toc!
Traceback (most recent call last):
Traceback (most recent call last):

And the same three error were repeated .

 **File "/opt/ros/kinetic/lib/python2.7/dist-packages/rospy/impl/tcpros_service.py", line 525, in call
  File "/opt/ros/kinetic/lib/python2.7/dist-packages/rospy/impl/tcpros_service.py", line 525, in call
        raise ServiceException("transport error completing service call: %s"%(str(e)))
raise ServiceException("transport error completing service call: %s"%(str(e)))
rospy.servicerospy.service..ServiceExceptionServiceException: : transport error completing service call: unable to receive data from sender, check sender's logs for detailstransport error completing service call: unable to receive data from sender, check sender's logs for details

[crazyflie_server-18] process has died [pid 16536, exit code -6, cmd /root/catkin_ws/devel/lib/crazyflie_driver/crazyflie_server __name:=crazyflie_server __log:=/root/.ros/log/96955368-aef8-11e7-ba7c-b88a60e5eba6/crazyflie_server-18.log].
log file: /root/.ros/log/96955368-aef8-11e7-ba7c-b88a60e5eba6/crazyflie_server-18*.log
[crazyflie0/lps_efk_bridge-8] process has died [pid 16459, exit code 1, cmd /root/catkin_ws/src/lps-ros/scripts/lps_ekf_bridge.py __name:=lps_efk_bridge __log:=/root/.ros/log/96955368-aef8-11e7-ba7c-b88a60e5eba6/crazyflie0-lps_efk_bridge-8.log].
log file: /root/.ros/log/96955368-aef8-11e7-ba7c-b88a60e5eba6/crazyflie0-lps_efk_bridge-8*.log
[crazyflie1/lps_efk_bridge-15] process has died [pid 16514, exit code 1, cmd /root/catkin_ws/src/lps-ros/scripts/lps_ekf_bridge.py __name:=lps_efk_bridge __log:=/root/.ros/log/96955368-aef8-11e7-ba7c-b88a60e5eba6/crazyflie1-lps_efk_bridge-15.log].
log file: /root/.ros/log/96955368-aef8-11e7-ba7c-b88a60e5eba6/crazyflie1-lps_efk_bridge-15***.log

Can I know How to add the acc.x file to got out the error 

Thank you,

Appala Chekuri

whoenig, 2018-01-05:
This is unrelated to crazyflie_ros. Closing.

closed 2018-01-05 by whoenig