USC-ACTLab/crazyswarm #14: crazyswarm tested with OptiTrack

byronxubin, 2017-06-07:
Hi guys, I'm using optitrack as the motion capture system. So I need to modify the hover_swarm.launch to make it compatible with the firmware. After reading the documentation, I do know which part I need to modify while it still not work for me. Can anyone give me some advice on that?
```
# tracking
      motion_capture_type: "optitrack" # one of vicon,optitrack
      object_tracking_type: "libobjecttracker" # one of motionCapture,libobjecttracker
<!--       vicon_host_name: "vicon"
      # optitrack_local_ip: "localhost"
      # optitrack_server_ip: "optitrack" -->
      optitrack_local_ip: "192.168.1.253"
      optitrack_server_ip: "192.168.1.147"
      save_point_clouds: ~/pointCloud.ot
      interactive_object: active-wand # virtual #active-wand
```

whoenig, 2017-06-08:
What does "not work" mean? The configuration looks correct (assuming that you updated the IP addresses accordingly).

neudinglei, 2017-09-07:
@whoenig @binxxx 
Hi guys, I'm using optitrack as the motion capture system. I follow the example above to modify the hover_swarm.launch.I use the mocap_optitrack-master package, which is located at /crazyswarm/ros_ws/src/crazyflie_ros.   However, it still not work for me,Can anyone give me some advice?

The error is as follows：
ERROR: cannot launch node of type [crazyflie_driver/crazyflie_server]: can't locate node [crazyflie_server] in package [crazyflie_driver]
ERROR: cannot launch node of type [joy/joy_node]: joy
ROS path [0]=/opt/ros/kinetic/share/ros
ROS path [1]=/home/neu/crazyswarm/ros_ws/src
ROS path [2]=/opt/ros/kinetic/share
ERROR: cannot launch node of type [crazyflie_manager/crazyflie_manager]: can't locate node [crazyflie_manager] in package [crazyflie_manager]

whoenig, 2017-09-07:
You need to install joy (`sudo apt install ros-kinetic-joy`). Otherwise it looks like there might have been a build error in your workspace. Try running `catkin_make` in `crazyswarm/ros_ws` and check if there are any issues.

jpreiss, 2017-09-07:
also make sure to `source devel/setup.bash` :)

neudinglei, 2017-09-08:
@whoenig @jpreiss
Hello,I encountered the following questions:
There were a lot of bugs when executing this command: ./build.sh.
The error is as follows：
Username for 'https://github.com':
Password for 'https://github.com':
remote: Repository not found.
fatal: Authentication failed for 'https://github.com/USC-ACTLab/phasespace_sdk.git/'
fatal: clone of 'https://github.com/USC-ACTLab/phasespace_sdk.git' into submodule path 'externalDependencies/phasespace_sdk' failed

/home/neu/crazyswarm/ros_ws/src/crazyflie_ros/externalDependencies/libmotioncapture/src/vicon.cpp:4:30: fatal error: vicon_sdk/Client.h: No such file or directory
compilation terminated.

CMakeFiles/Makefile2:1049: recipe for target 'crazyflie_ros/externalDependencies/libmotioncapture/CMakeFiles/libmotioncapture.dir/all' failed
make[1]: *** [crazyflie_ros/externalDependencies/libmotioncapture/CMakeFiles/libmotioncapture.dir/all] Error 2
make[1]: *** Waiting for unfinished jobs....
[ 5%] Built target _crazyflie_driver_generate_messages_check_deps_StartEllipse
[ 5%] Linking CXX shared library /home/neu/crazyswarm/ros_ws/devel/lib/libcrazyflie_cpp.so
[ 5%] Built target crazyflie_cpp
[ 5%] Linking CXX shared library /home/neu/crazyswarm/ros_ws/devel/lib/liblibobjecttracker.so
[ 5%] Built target libobjecttracker
Makefile:138: recipe for target 'all' failed
make: *** [all] Error 2

[image]
[image]

Because my system is optitrack, so I did not download the VICON DataStream SDK.

whoenig, 2017-09-08:
Those errors are "normal" and documented. Essentially we were not allowed to distribute the vicon SDK and you have to download it manually. Since you are using optitrack only, you can also decide to disable VICON support in the build: https://github.com/USC-ACTLab/libmotioncapture/blob/master/CMakeLists.txt#L7

neudinglei, 2017-09-10:
@whoenig 
Thank you for your quick reply, I still can not run the swarm project. After the implementation of the compiler command：
$ cd ros_ws/
$ catkin_make
The error is as follows：
[image]

/home/neu/crazyswarm/ros_ws/src/crazyflie_ros/externalDependencies/libmotioncapture/src/vicon.cpp:4:30: fatal error: vicon_sdk/Client.h: No such file or directory
compilation terminated.
crazyflie_ros/externalDependencies/libmotioncapture/CMakeFiles/libmotioncapture.dir/build.make:86: recipe for target 'crazyflie_ros/externalDependencies/libmotioncapture/CMakeFiles/libmotioncapture.dir/src/vicon.cpp.o' failed
make[2]: *** [crazyflie_ros/externalDependencies/libmotioncapture/CMakeFiles/libmotioncapture.dir/src/vicon.cpp.o] Error 1
CMakeFiles/Makefile2:1049: recipe for target 'crazyflie_ros/externalDependencies/libmotioncapture/CMakeFiles/libmotioncapture.dir/all' failed
make[1]: *** [crazyflie_ros/externalDependencies/libmotioncapture/CMakeFiles/libmotioncapture.dir/all] Error 2
make[1]: *** Waiting for unfinished jobs....
/home/neu/crazyswarm/ros_ws/src/crazyflie_ros/crazyflie_manager/src/manager.cpp:10:47: fatal error: crazyflie_driver/UploadTrajectory.h: No such file or directory
compilation terminated.
crazyflie_ros/crazyflie_manager/CMakeFiles/crazyflie_manager.dir/build.make:62: recipe for target 'crazyflie_ros/crazyflie_manager/CMakeFiles/crazyflie_manager.dir/src/manager.cpp.o' failed
make[2]: *** [crazyflie_ros/crazyflie_manager/CMakeFiles/crazyflie_manager.dir/src/manager.cpp.o] Error 1
CMakeFiles/Makefile2:3806: recipe for target 'crazyflie_ros/crazyflie_manager/CMakeFiles/crazyflie_manager.dir/all' failed
make[1]: *** [crazyflie_ros/crazyflie_manager/CMakeFiles/crazyflie_manager.dir/all] Error 2
/home/neu/crazyswarm/ros_ws/src/crazyflie_ros/crazyflie_controller/src/controller.cpp:9:58: fatal error: crazyflie_controller/ExecuteTrajectoryAction.h: No such file or directory
compilation terminated.
crazyflie_ros/crazyflie_controller/CMakeFiles/crazyflie_controller.dir/build.make:62: recipe for target 'crazyflie_ros/crazyflie_controller/CMakeFiles/crazyflie_controller.dir/src/controller.cpp.o' failed
make[2]: *** [crazyflie_ros/crazyflie_controller/CMakeFiles/crazyflie_controller.dir/src/controller.cpp.o] Error 1
CMakeFiles/Makefile2:4619: recipe for target 'crazyflie_ros/crazyflie_controller/CMakeFiles/crazyflie_controller.dir/all' failed
make[1]: *** [crazyflie_ros/crazyflie_controller/CMakeFiles/crazyflie_controller.dir/all] Error 2
[  6%] Linking CXX executable /home/neu/crazyswarm/ros_ws/devel/lib/crazyflie_tools/packetDropTest
[  6%] Built target packetDropTest
Makefile:138: recipe for target 'all' failed
make: *** [all] Error 2
Invoking "make -j4 -l4" failed

Can you provide some guidance? Thank you!

jpreiss, 2017-09-19:
@neudinglei did you modify `libmotioncapture/CMakeLists.txt`? For Optitrack lines 6-8 should look like:

    set(ENABLE_OPTITRACK True CACHE STRING "Enable Optitrack")
    set(ENABLE_VICON False CACHE STRING "Enable Vicon")
    set(ENABLE_PHASESPACE False CACHE STRING "Enable Phasespace")

jpreiss, 2017-12-18:
I am trying to close dead issues. @neudinglei are you still having any problems building?

samandehghan, 2018-02-09:
Hi everyone,

I am using optitrack motion capture and I changed CMakeLists.txt.  but after implementing ./build.sh or catkin_make -k I got this error: 

[ 95%] Built target crazyflie_driver_generate_messages
[ 95%] Built target crazyflie_controller_generate_messages
/home/sam/project/crazyswarm/ros_ws/src/crazyflie_ros/externalDependencies/libmotioncapture/src/vicon.cpp:4:40: fatal error: vicon_sdk/DataStreamClient.h: No such file or directory
compilation terminated.
crazyflie_ros/externalDependencies/libmotioncapture/CMakeFiles/libmotioncapture.dir/build.make:86: recipe for target 'crazyflie_ros/externalDependencies/libmotioncapture/CMakeFiles/libmotioncapture.dir/src/vicon.cpp.o' failed
make[2]: *** [crazyflie_ros/externalDependencies/libmotioncapture/CMakeFiles/libmotioncapture.dir/src/vicon.cpp.o] Error 1
CMakeFiles/Makefile2:1049: recipe for target 'crazyflie_ros/externalDependencies/libmotioncapture/CMakeFiles/libmotioncapture.dir/all' failed
make[1]: *** [crazyflie_ros/externalDependencies/libmotioncapture/CMakeFiles/libmotioncapture.dir/all] Error 2
Makefile:138: recipe for target 'all' failed
make: *** [all] Error 2
Invoking "make -j8 -l8" failed

any help would be appreciated. 
Thanks

whoenig, 2018-02-09:
Either download the VICON SDK (as described [here](http://crazyswarm.readthedocs.io/en/latest/installation.html#simulation-and-physical-robots)) or disable VICON support [here](https://github.com/USC-ACTLab/libmotioncapture/blob/62fb73f2ba51cc39705d4c7cfb7978bff662ea87/CMakeLists.txt#L7).

samandehghan, 2018-02-09:
@whoenig
I disabled VICON support  but I still got this error.

whoenig, 2018-02-09:
You might need to delete your cmakecache (aka full rebuild after that change).

whoenig, 2018-06-05:
This sounds like you try to compile in a VM that doesn't have enough memory allocated? You can give the VM more main memory or try compiling using "catkin_make -j1" to avoid multiple parallel build processes.

closed 2018-12-07 by whoenig