USC-ACTLab/crazyswarm #16: CMake Warning: libusb not found

kgslovak, 2017-06-09:
Hey Wolfgang,

I'm running into the following issue whenever I try to build the crazyflie_ros workspace for the crazyswarm package: 

`CMake Warning at /opt/ros/kinetic/share/catkin/cmake/catkin_package.cmake:166 (message):
  catkin_package() DEPENDS on '/usr/lib/x86_64-linux-gnu/libusb-1.0.so' but
  neither '/usr/lib/x86_64-linux-gnu/libusb-1.0.so_INCLUDE_DIRS' nor
  '/usr/lib/x86_64-linux-gnu/libusb-1.0.so_LIBRARIES' is defined.
Call Stack (most recent call first):
  /opt/ros/kinetic/share/catkin/cmake/catkin_package.cmake:102 (_catkin_package)
  crazyflie_ros/crazyflie_cpp/CMakeLists.txt:19 (catkin_package)`

Running the command `dpkg -l libusb*` gives me:

```
Desired=Unknown/Install/Remove/Purge/Hold
| Status=Not/Inst/Conf-files/Unpacked/halF-conf/Half-inst/trig-aWait/Trig-pend
|/ Err?=(none)/Reinst-required (Status,Err: uppercase=bad)
||/ Name           Version      Architecture Description
+++-==============-============-============-=================================
ii  libusb-0.1-4:a 2:0.1.12-28  amd64        userspace USB programming library
ii  libusb-1.0-0:a 2:1.0.20-1   amd64        userspace USB programming library
ii  libusb-1.0-0-d 2:1.0.20-1   amd64        userspace USB programming library
ii  libusb-1.0-doc 2:1.0.20-1   all          documentation for userspace USB p
un  libusb-dev     <none>       <none>       (no description available)
un  libusbmuxd-too <none>       <none>       (no description available)
ii  libusbmuxd4:am 1.0.10-2ubun amd64        USB multiplexor daemon for iPhone
```

I am using a "fresh" installation of ubuntu 16.04 and ROS Kinetic. I've been searching online for answers but I am coming up short (I probably don't know the right questions to ask).

Do you have any ideas?

Thanks,
Kyle

whoenig, 2017-06-12:
Very strange - I have not seen that before. What happens if you delete the build and devel folders and ran `catkin_make` again, i.e. what is the initial output of CMake?

kgslovak, 2017-06-12:
Deleting my build and devel spaces and running `catkin_make` again gives:

```
labuser@labuser-ThinkCentre-M91p:~/Desktop/Crazyflie/ROS/crazyswarm/ros_ws$ catkin_make
Base path: /home/labuser/Desktop/Crazyflie/ROS/crazyswarm/ros_ws
Source space: /home/labuser/Desktop/Crazyflie/ROS/crazyswarm/ros_ws/src
Build space: /home/labuser/Desktop/Crazyflie/ROS/crazyswarm/ros_ws/build
Devel space: /home/labuser/Desktop/Crazyflie/ROS/crazyswarm/ros_ws/devel
Install space: /home/labuser/Desktop/Crazyflie/ROS/crazyswarm/ros_ws/install
####
#### Running command: "cmake /home/labuser/Desktop/Crazyflie/ROS/crazyswarm/ros_ws/src -DCATKIN_DEVEL_PREFIX=/home/labuser/Desktop/Crazyflie/ROS/crazyswarm/ros_ws/devel -DCMAKE_INSTALL_PREFIX=/home/labuser/Desktop/Crazyflie/ROS/crazyswarm/ros_ws/install -G Unix Makefiles" in "/home/labuser/Desktop/Crazyflie/ROS/crazyswarm/ros_ws/build"
####
-- The C compiler identification is GNU 5.4.0
-- The CXX compiler identification is GNU 5.4.0
-- Check for working C compiler: /usr/bin/cc
-- Check for working C compiler: /usr/bin/cc -- works
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - done
-- Detecting C compile features
-- Detecting C compile features - done
-- Check for working CXX compiler: /usr/bin/c++
-- Check for working CXX compiler: /usr/bin/c++ -- works
-- Detecting CXX compiler ABI info
-- Detecting CXX compiler ABI info - done
-- Detecting CXX compile features
-- Detecting CXX compile features - done
-- Using CATKIN_DEVEL_PREFIX: /home/labuser/Desktop/Crazyflie/ROS/crazyswarm/ros_ws/devel
-- Using CMAKE_PREFIX_PATH: /home/labuser/Desktop/Crazyflie/ROS/cf_ros/devel;/opt/ros/kinetic
-- This workspace overlays: /home/labuser/Desktop/Crazyflie/ROS/cf_ros/devel;/opt/ros/kinetic
-- Found PythonInterp: /usr/bin/python (found version "2.7.12") 
-- Using PYTHON_EXECUTABLE: /usr/bin/python
-- Using Debian Python package layout
-- Using empy: /usr/bin/empy
-- Using CATKIN_ENABLE_TESTING: ON
-- Call enable_testing()
-- Using CATKIN_TEST_RESULTS_DIR: /home/labuser/Desktop/Crazyflie/ROS/crazyswarm/ros_ws/build/test_results
-- Looking for pthread.h
-- Looking for pthread.h - found
-- Looking for pthread_create
-- Looking for pthread_create - not found
-- Looking for pthread_create in pthreads
-- Looking for pthread_create in pthreads - not found
-- Looking for pthread_create in pthread
-- Looking for pthread_create in pthread - found
-- Found Threads: TRUE  
-- Found gtest sources under '/usr/src/gtest': gtests will be built
-- Using Python nosetests: /usr/bin/nosetests-2.7
-- catkin 0.7.6
-- BUILD_SHARED_LIBS is on
-- ~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
-- ~~  traversing 11 packages in topological order:
-- ~~  - crazyflie (metapackage)
-- ~~  - crazyflie_cpp
-- ~~  - crazyflie_demo
-- ~~  - crazyflie_description
-- ~~  - crazyflie_tools
-- ~~  - crazyswarm
-- ~~  - libmotioncapture
-- ~~  - libobjecttracker
-- ~~  - crazyflie_driver
-- ~~  - crazyflie_manager
-- ~~  - crazyflie_controller
-- ~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
-- +++ processing catkin metapackage: 'crazyflie'
-- ==> add_subdirectory(crazyflie_ros/crazyflie)
-- +++ processing catkin package: 'crazyflie_cpp'
-- ==> add_subdirectory(crazyflie_ros/crazyflie_cpp)
CMake Warning at /opt/ros/kinetic/share/catkin/cmake/catkin_package.cmake:166 (message):
  catkin_package() DEPENDS on '/usr/lib/x86_64-linux-gnu/libusb-1.0.so' but
  neither '/usr/lib/x86_64-linux-gnu/libusb-1.0.so_INCLUDE_DIRS' nor
[... 92 lines cut ...]
-- looking for PCL_RECOGNITION
-- Found PCL_RECOGNITION: /usr/lib/x86_64-linux-gnu/libpcl_recognition.so  
-- looking for PCL_KEYPOINTS
-- Found PCL_KEYPOINTS: /usr/lib/x86_64-linux-gnu/libpcl_keypoints.so  
-- looking for PCL_VISUALIZATION
-- Found PCL_VISUALIZATION: /usr/lib/x86_64-linux-gnu/libpcl_visualization.so  
-- looking for PCL_PEOPLE
-- Found PCL_PEOPLE: /usr/lib/x86_64-linux-gnu/libpcl_people.so  
-- looking for PCL_OUTOFCORE
-- Found PCL_OUTOFCORE: /usr/lib/x86_64-linux-gnu/libpcl_outofcore.so  
-- looking for PCL_TRACKING
-- Found PCL_TRACKING: /usr/lib/x86_64-linux-gnu/libpcl_tracking.so  
-- looking for PCL_APPS
-- Could NOT find PCL_APPS (missing:  PCL_APPS_LIBRARY) 
-- looking for PCL_MODELER
-- Found PCL_MODELER: /usr/include/pcl-1.7  
-- looking for PCL_IN_HAND_SCANNER
-- Found PCL_IN_HAND_SCANNER: /usr/include/pcl-1.7  
-- looking for PCL_POINT_CLOUD_EDITOR
-- Found PCL_POINT_CLOUD_EDITOR: /usr/include/pcl-1.7
```

I copied the full output [here](https://drive.google.com/open?id=0B6rEh5VzICgjemRlSE9ITllIVkU).

After doing some digging around online this weekend, I think it may not be finding the dev version of the libusb package?

whoenig, 2017-06-15:
I think the build error might be caused by some missing dependencies (which we'll need to fix on our side). As workaround, you can try `catkin_make -k` two or three times.

kgslovak, 2017-06-18:
So after doing some digging around, I think I've figured out what is going on. The appearance of this warning is due to a change in the catkin_package.cmake files of catkin from version 0.6.19 (included with indigo) to version 0.7.6 (included with kinetic). When parsing through the list of provided dependencies, this script checks a different set of conditions based on which version is being used:

catkin 0.6.19 (line 160): `if(NOT ${${depend_name}_FOUND})`
catkin 0.7.6 (line 165): `if(NOT DEFINED ${depend_name}_INCLUDE_DIRS AND NOT DEFINED ${depend_name}_LIBRARIES)`

The "issue" is that the CMakeLists.txt files of the crazyswarm package use `find_library(USB_LIB usb-1.0)` which only sets the variable `USB_LIB` to the full path of the desired library. It does not define `USB_LIB_FOUND`, `USB_LIB_INCLUDE_DIRS`, or `USB_LIB_LIBRARIES`. So Catkin 0.6.19 did not throw a warning (which is probably why this condition was changed), however the Kinetic version does. 

A search utilizing something like:
```
find_package(PkgConfig REQUIRED)
pkg_search_module(USB_LIB REQUIRED usb-1.0)
```
might be more appropriate to "properly" link the library? 

As it is, since `${USB_LIB}` assess to the full path of the library's '`.so`' file and the CMakeLists.txt files use:

```
target_link_libraries(crazyflie_cpp
  ${USB_LIB}
)
```

I don't think this warning is pointing to any issues linking the usb library.

whoenig, 2017-06-19:
Thats a great analysis - thank you! I think you are right that using package config might be the better choice (or getting someone to fix the Find-USB module.) I'll keep this open until we fix the issue accordingly.

`-k` stands for keep going. You can find all the options if you look in the cmake manual; `catkin_make` is a cmake-wrapper.

kgslovak, 2017-06-19:
You're welcome, happy to help.

That makes sense why I wasn't finding anything with my google-searching. Thanks for the info!

ricardohalfeld, 2019-03-22:
Hello folks. I'm trying to install this stack, but my setup is a bit odd: I'm running ROS Melodic 1.14.3 in WSL with Ubuntu 18.04.2 LTS. This means I'm running into what @kgslovak mentioned in his analysis:

I'm fairly new to Linux, ROS and pretty much everything coming along with this endeavor, so please bear with me. Am I right in my understanding of @kgslovak's suggestion? Should I replace:

`find_library(USB_LIB usb-1.0)`

with:

`find_package(PkgConfig REQUIRED)`
`pkg_search_module(USB_LIB REQUIRED usb-1.0)`

In every occurrence in crazyflie_ros directory? 

- crazyflie_cpp
  - Cmake.cmake
  - ROS.cmake

- crazyflie_tools
  - Cmake.cmake
  - ROS.cmake

whoenig, 2019-03-22:
WSL is not supported, see https://crazyswarm.readthedocs.io/en/latest/installation.html.

ricardohalfeld, 2019-03-22:
> Using ubuntu in Windows Subsystem for Linux (WSL) is not supported since WSL does not have USB support and so Crazyradio will not work. You must install Ubuntu either directly on the computer or in a VM.

I see. I believe things have changed recently, though. I'll let you know if manage to get around that.
https://devblogs.microsoft.com/commandline/bashwsl-insiders-build-now-supports-usbserial-comms-and-mounting-removable-storage-network-shares/

Anyway, aside from the WSL part, is my understanding correct?

whoenig, 2019-03-22:
Your understanding is correct. Actually you only need to change ROS.cmake files (the other ones are used if you want to build crazyflie_tools standalone, i.e. without any ROS [which might be useful for debugging/trouble shooting]).

If it works with WSL, please let us know and we can update the documentation accordingly.

closed 2019-11-12 by whoenig