USC-ACTLab/crazyswarm #17: Build Error: Missing Header Files

kgslovak, 2017-06-18:
I feel like I'm finally getting close to being able to successfully compile the crazyswarm package after hunting down and resolving a handful of dependencies... but the build is now failing because header files are missing:

```
/home/labuser/Desktop/Crazyflies/ROS/crazyswarm/ros_ws/src/crazyflie_ros/crazyflie_controller/src/controller.cpp:9:58: fatal error: crazyflie_controller/ExecuteTrajectoryAction.h: No such file or directory
 #include <crazyflie_controller/ExecuteTrajectoryAction.h>
```
Commenting `#include <crazyflie_controller/ExecuteTrajectoryAction.h>` reveals another missing header, GenericLogData.h, is also missing. I cannot find these files anywhere within the crazyswarm package and commenting the 'include's generates a long list of compilation errors.

Is there something I'm overlooking?

whoenig, 2017-06-19:
You shouldn't need to change anything. However, I believe that there are some wrong dependencies, which can cause the built to build targets in the wrong order. This leads to such "missing" header files not being generated *before* they are actually included. If you run 

```
catkin_make -k -j1
```
For two or three times it should work out of the box. We tested the build in a VM with a freshly installed Ubuntu 16.04 and ROS Kinetic. Here `-k` stands for keep going and `-j1` disables parallel builds (you can find all the options if you look in the cmake manual; `catkin_make` is a cmake-wrapper.)

kgslovak, 2017-06-19:
Thanks Wolfgang, running `catkin_make -k -j1` twice did the trick. The compilation made it to 100%!

I've been keeping track of all my steps to reach this point "from scratch" on Ubuntu 14+Indigo (I switched over from Ubuntu 16+Kinetic because it felt like I was running in circles trying to resolve PCL/VTK dependencies). Would this information be useful to you for supplementing or modifying the install instructions? I plan to do the same for 16+Kin. but for now I'm going to spend a few weeks testing everything out.

closed 2017-06-22 by kgslovak