USC-ACTLab/crazyswarm #18: CF Firmware: MD5 Values?

kgslovak, 2017-06-21:
Hey Wolfgang,

Could you provide the correct MD5 values for the crazyflie firmware provided in the crazyswarm package? 

Using the CF-PC client, I've been trying to flash the cf2.bin and cf2_nrf.bin files, but when I do the dialogue states that the NRF firmware is also being written to the STM microcontroller... On restarting the 'flie, only the M2 LED lights up and stays lit (only after flashing the NRF firmware -- no issues when flashing the crazyswarm-STM firmware).  

I received a verification when using the build script commands, but I think I remember reading that this may occur even when using an incorrectly configured toolchain. I'll re-investigate my GCC toolchain if needed... but I figured I'd check those values first.

The MD5 values for my binaries are:
```
d2c05f9fed6fe70c58c7b139f9400324  cf2.bin
d12bfa6076b350a4d9a5b799c262e412  cf2_nrf.bin
```

Thanks,
Kyle

kgslovak, 2017-06-22:
Using "`make cload`" worked. The prompt indicated each firmware was flashing to the correct MCU and the 'flie restarted correctly. I'm guessing my previous issues were due to a PC-client bug(?)

However, I noticed something rather unsettling when doing a bit of experimentation:
I backed up my builds when attempting to use Ubuntu 16 (I am currently back on 14) and the firmware binaries have different MD5 values than my current U14-built binaries (I assume due to differences in the toolchain). Their MD5 values are:

```
452f9ff46cf526e7a2af88213918f985  cf2.bin
72b51e089147da04f89dd82dc12affa6  cf2_nrf.bin
```

When I swapped the .bin files out and used "`make cload`" to flash the U16-built binaries there were, again, no issues flashing or restarting. Which doesn't make sense to me, but I've been learning everything on-the-fly so maybe I'm missing something...

When you (or someone else) is able, would you mind listing the correct firmware MD5 values? (If nothing else, just for my peace of mind!)

Much appreciated,
Kyle

kgslovak, 2017-06-27:
While attempting to use the `hover_swarm.launch` script I'm not able to get the 'flies to connect and keep getting the runtime error. I've tried both of my firmware builds with no success... The correct MD5 values would really help in troubleshooting this issue.

Additionally, after updating the Crazyradio firmware and using:
 `lsusb -d 1915:7777 -v | grep bcdDevice`

I get:
`bcdDevice = 99.55`

Is that correct? (MD5 values would be helpful if not)

Thanks,
Kyle

jpreiss, 2017-09-19:
@kgslovak Sorry about radio silence on this issue. Are you still having trouble connecting to the Crazyflies?

kgslovak, 2017-09-27:
@jpreiss No worries, I know you and Wolfgang had other things going on over the summer. 

I have not yet achieved connection with the 'flies using the Crazyswarm package, however I put that goal on the back-burner shortly after posting this issue. For now we are just using the crazyflie_ros package in our lab with custom scripts.

I'm sure there have been changes since I tried building the package last so hopefully I can give it another try soon. Would it be possible for you guys to post the MD5 sums for the CF-firmware as I had requested?

jpreiss, 2017-12-18:
@kgslovak we just added precompiled binaries for the STM32, NRF, and Crazyradio in https://github.com/USC-ACTLab/crazyswarm/tree/master/prebuilt . If you are still having trouble building the firmware, these may be helpful. I'm not sure the MD5 sum will be meaningful since we have not made any effort towards deterministic builds.

closed 2017-12-18 by jpreiss