USC-ACTLab/crazyswarm #19: hover_swarm.launch: 'XmlRpc::XmlRpcException'

kgslovak, 2017-06-26:
I'm having issues when trying to run hover_swarm.launch. My full terminal output can be seen [here](https://drive.google.com/open?id=1uSnyMoHu5oCnV_0R7UdXMKt8V3ETgP--6Qt5NfNMsw8).

Jumping ahead to the error:
```
...
[ INFO] [1498513544.008199879]: waitForService: Service [/emergency] is now available.
[ INFO] [1498513544.015521940]: Manager ready.
ch: 1
terminate called after throwing an instance of 'XmlRpc::XmlRpcException'
[ INFO] [1498513544.139372619]: Opened joystick: /dev/input/js0. deadzone_: 0.050000.
[crazyflie_server-2] process has died [pid 13386, exit code -6, cmd 
/home/labuser/Desktop/Crazyflies/ROS/crazyswarm/ros_ws/devel/lib/crazyflie_driver/crazyflie_server
 __name:=crazyflie_server __log:=/home/labuser/.ros/log/cdfce1f2-5ab8-11e7-9a12-
4437e6a9fa65/crazyflie_server-2.log].
log file: /home/labuser/.ros/log/cdfce1f2-5ab8-11e7-9a12-4437e6a9fa65/crazyflie_server-2*.log
```

Rviz does not show any objects (just the floor grid). Please let me know if any other information can help diagnose the issue.

whoenig, 2017-06-27:
Could you attach crazyflies.yaml? The YAML library in ROS has some nasty bugs - you most likely have somewhere an integer where you must have a floating point value, e.g. the position entries need to be of the form "[0.0, 1.0, 2.0]" rather than "[0, 1, 2]". If that's the issue feel free to leave the bug open - we should probably switch to another library (e.g. yaml-cpp), which does have proper YAML support.

kgslovak, 2017-06-27:
Ah, I see. That's it -- I have a marker centered on the battery and entered it's position as [0, 0, 0]. Thanks for the quick response Wolfgang.

closed 2017-09-19 by jpreiss