USC-ACTLab/crazyswarm #21: Optitrack coordinates transformation

Negvet, 2017-07-17:
Hey Whoenig,

I followed the setup procedure in details. My mocap system is Optitrack.

The problem is that in ROS we have X pointed forward, Y left and Z up. But in Motive (Optitrack tracker) Z pointed forward, X left and Y upwards. The result is that, for example, when I move CF forward (along X axes in ROS) I can see in RVIZ that CF moves up. Also, although I modified marker arrangement according to the documentation, I am getting errors like `Not enough correspondences found. Relax your threshold parameters` and `ICP did not converge!`.
May be you took into account that issue but I do not see how to enable it? Otherwise what is the best place to make changes?

Additional info below.
The console output is the following:

> negvet@asuslab:~$ roslaunch crazyswarm hover_swarm.launch
> ... logging to /home/negvet/.ros/log/ae21116e-6aee-11e7-929e-2c56dcbc6198/roslaunch-asuslab-14476.log
> Checking log directory for disk usage. This may take awhile.
> Press Ctrl-C to interrupt
> Done checking log file disk usage. Usage is <1GB.
> 
> started roslaunch server http://asuslab:44439/
> 
> SUMMARY
> ========
> 
> PARAMETERS
>  * /crazyflie_manager/csv_file: /home/negvet/Down...
>  * /crazyflie_manager/timescale: 0.8
>  * /crazyflie_server/broadcast_address: FFE7E7E7E7
>  * /crazyflie_server/dynamicsConfigurations/0/maxFitnessScore: 0.001
>  * /crazyflie_server/dynamicsConfigurations/0/maxPitch: 1.4
>  * /crazyflie_server/dynamicsConfigurations/0/maxPitchRate: 20.0
>  * /crazyflie_server/dynamicsConfigurations/0/maxRoll: 1.4
>  * /crazyflie_server/dynamicsConfigurations/0/maxRollRate: 20.0
>  * /crazyflie_server/dynamicsConfigurations/0/maxXVelocity: 2.0
>  * /crazyflie_server/dynamicsConfigurations/0/maxYVelocity: 2.0
>  * /crazyflie_server/dynamicsConfigurations/0/maxYawRate: 10.0
>  * /crazyflie_server/dynamicsConfigurations/0/maxZVelocity: 3.0
>  * /crazyflie_server/enable_logging: False
>  * /crazyflie_server/enable_parameters: True
>  * /crazyflie_server/firmwareParams/ctrlMel/i_range_m_z: 1500
>  * /crazyflie_server/firmwareParams/ctrlMel/i_range_xy: 2.0
>  * /crazyflie_server/firmwareParams/ctrlMel/i_range_z: 0.4
>  * /crazyflie_server/firmwareParams/ctrlMel/kR_xy: 70000
>  * /crazyflie_server/firmwareParams/ctrlMel/kR_z: 60000
>  * /crazyflie_server/firmwareParams/ctrlMel/kd_omega_rp: 200
>  * /crazyflie_server/firmwareParams/ctrlMel/kd_xy: 0.2
>  * /crazyflie_server/firmwareParams/ctrlMel/kd_z: 0.4
>  * /crazyflie_server/firmwareParams/ctrlMel/ki_m_z: 500
>  * /crazyflie_server/firmwareParams/ctrlMel/ki_xy: 0.05
>  * /crazyflie_server/firmwareParams/ctrlMel/ki_z: 0.05
>  * /crazyflie_server/firmwareParams/ctrlMel/kp_xy: 0.4
>  * /crazyflie_server/firmwareParams/ctrlMel/kp_z: 1.25
>  * /crazyflie_server/firmwareParams/ctrlMel/kw_xy: 20000
>  * /crazyflie_server/firmwareParams/ctrlMel/kw_z: 12000
>  * /crazyflie_server/firmwareParams/ctrlMel/mass: 0.032
>  * /crazyflie_server/firmwareParams/ctrlMel/massThrust: 132000
>  * /crazyflie_server/firmwareParams/flightmode/posCtrl: 1
>  * /crazyflie_server/firmwareParams/ring/effect: 14
>  * /crazyflie_server/firmwareParams/ring/headlightEnable: 0
>  * /crazyflie_server/firmwareParams/ring/solidBlue: 20
>  * /crazyflie_server/firmwareParams/ring/solidGreen: 0
>  * /crazyflie_server/firmwareParams/ring/solidRed: 0
>  * /crazyflie_server/force_no_cache: False
>  * /crazyflie_server/genericLogTopicFrequencies: [10]
>  * /crazyflie_server/genericLogTopic_log1_Variables: ['stabilizer.yaw'...
>  * /crazyflie_server/genericLogTopics: ['log1']
>  * /crazyflie_server/interactive_object: virtual
>  * /crazyflie_server/markerConfigurations/0/numPoints: 4
>  * /crazyflie_server/markerConfigurations/0/offset: [0.0, 0.0, -0.04]
>  * /crazyflie_server/markerConfigurations/0/points/0: [0.001376, 0.0055...
>  * /crazyflie_server/markerConfigurations/0/points/1: [-0.044365, 0.043...
>  * /crazyflie_server/markerConfigurations/0/points/2: [-0.042596, -0.04...
>  * /crazyflie_server/markerConfigurations/0/points/3: [0.044267, -0.044...
>  * /crazyflie_server/motion_capture_type: optitrack
>  * /crazyflie_server/numDynamicsConfigurations: 1
>  * /crazyflie_server/numMarkerConfigurations: 1
>  * /crazyflie_server/object_tracking_type: libobjecttracker
>  * /crazyflie_server/optitrack_local_ip: 192.168.0.67
>  * /crazyflie_server/optitrack_server_ip: 192.168.0.100
>  * /crazyflie_server/print_latency: False
>  * /crazyflie_server/save_point_clouds: ~/pointCloud.ot
>  * /crazyflie_server/world_frame: /world
>  * /crazyflie_server/write_csvs: False
>  * /crazyflies: [{'initialPositio...
>  * /rosdistro: kinetic
>  * /rosversion: 1.12.7
> 
> NODES
>   /
>     crazyflie_manager (crazyflie_manager/crazyflie_manager)
>     crazyflie_server (crazyflie_driver/crazyflie_server)
>     rqt (rqt_gui/rqt_gui)
>     rqt_plot_yaw (rqt_plot/rqt_plot)
>     rviz (rviz/rviz)
> 
> auto-starting new master
> process[master]: started with pid [14487]
> ROS_MASTER_URI=http://localhost:11311
> 
> setting /run_id to ae21116e-6aee-11e7-929e-2c56dcbc6198
> process[rosout-1]: started with pid [14500]
> started core service [/rosout]
> process[crazyflie_server-2]: started with pid [14517]
> process[crazyflie_manager-3]: started with pid [14518]
> process[rviz-4]: started with pid [14525]
> process[rqt-5]: started with pid [14531]
> process[rqt_plot_yaw-6]: started with pid [14532]
> [ INFO] [1500295902.050838196]: Wait for services...
> [ INFO] [1500295902.051987784]: waitForService: Service [/emergency] has not been advertised, waiting...
> [ INFO] [1500295902.080295086]: waitForService: Service [/emergency] is now available.
> [ INFO] [1500295902.094723908]: Manager ready.
> NatNetClient connecting: 192.168.0.67 to 192.168.0.100...
> WARNING: Could not set receive buffer size. Asked for 1048576B got 0B
> [Client] Server Software: Motive
> [Client] NatNetVersion: 2.10
> [Client] ServerVersion: 1.10
> NatNet server version: 1677764800
> ch: 1
> [ INFO] [1500295902.234850503]: Adding CF: cf1 (radio://0/80/2M/E7E7E7E701, cf1)...
> [ INFO] [1500295902.251270257]: CF ctor: 0.016309 s
> [ INFO] [1500295902.259970576]: [cf1] Requesting parameters...
> [ INFO] [1500295902.276363232]: Found variables in cache.
> [ INFO] [1500295902.557709479]: [cf1] reqParamTOC: 0.306374 s
> [ INFO] [1500295902.557747435]: [cf1] Ready. Elapsed: 0.306439 s
> [ INFO] [1500295902.557765324]: CF run: 0.306520 s
> [ INFO] [1500295902.601801716]: [cf1] Update parameters
> [ INFO] [1500295902.667709764]: Update params: 0.109815 s
> [ INFO] [1500295902.667966633]: Started 1 threads
> [pcl::IterativeClosestPoint::computeTransformation] Not enough correspondences found. Relax your threshold parameters.
> [ WARN] [1500295902.670972982]: ICP did not converge!
> [ WARN] [1500295902.671025576]: No updated pose for CF cf1 for 0.002638 s.
> [pcl::IterativeClosestPoint::computeTransformation] Not enough correspondences found. Relax your threshold parameters.
> [ WARN] [1500295902.671268235]: ICP did not converge!
> [ WARN] [1500295902.671335318]: No updated pose for CF cf1 for 0.002909 s.

hover_swarm.launch file:

>       # tracking
>       motion_capture_type: "optitrack" # one of vicon,optitrack
>       object_tracking_type: "libobjecttracker" # one of motionCapture,libobjecttracker
>       # vicon_host_name: "vicon"
>       optitrack_local_ip: "192.168.0.67"
>       optitrack_server_ip: "192.168.0.100"
>       save_point_clouds: ~/pointCloud.ot
>       interactive_object: virtual # virtual #active-wand
>       numMarkerConfigurations: 1
>       markerConfigurations:
>         "0":
>           numPoints: 4
>           offset: [0.0, 0.0, -0.04]
>           points:
>             "0": [0.001376,0.005547,0.034077]
>             "1": [-0.044365,0.043484,0.009906]
>             "2": [-0.042596,-0.043791,0.008427]
>             "3": [0.044267,-0.044304,0.009137]

crazyflies.yaml file:

> crazyflies:
> - channel: 80
>   id: 1
>   initialPosition: [0.0, 0.0, 0.0]

Have to say that the crazyswarm project is awesome. I am able to fly two CF using pure ROS package crazyflie_ros, but now I want to switch to more CF, for that I need crazyswarm. In case of crazyflie_ros, I modified vrpn_client_ros to remap the coordinates from Optitrack to ROS, but I am not sure if I know how to do it with crazyswarm.
Thanks in advance!

whoenig, 2017-07-17:
The crazyswarm should take care of that, but it is possible that you have to change the coordinate system in Motive. You can also adjust the mapping by changing the following code: https://github.com/USC-ACTLab/libmotioncapture/blob/master/src/optitrack.cpp#L24-L25

Adding @naus3a, who implemented and tested the OptiTrack backend.

naus3a, 2017-07-18:
> libmotioncapture/blob/master/src/optitrack.cpp#L24-L25

The highlighted lines should already work with Motive/Optitrack; it should
be easy to remap/scale to any other coordinate system.
In case of problems, I'm here.

Enrico

Negvet, 2017-07-18:
@whoenig , Thanks for your reply. I will do it and report back.

@naus3a , could you please share some Optitrack settings, like streaming engine parameters (network interface, streaming settings, and advanced network settings)? 
The problem is that Optitrack is used by other VR team and they change some settings - today I cannot get even wrong position of CF and no markers, instead, I am getting the error 

> ...
> [ INFO] [1500368335.187809572]: [cf1] Update parameters
> [ INFO] [1500368335.278951562]: Update params: 0.156564 s
> [ INFO] [1500368335.279203858]: Started 1 threads
> [ WARN] [1500368336.279493411]: VICON Latency high: 2.72609e+20 s.
>   Latency: : 2.72609e+20 s.
> 
> [ WARN] [1500368336.279692751]: No updated pose for CF cf1 for 1500368336.279626 s.
> [ WARN] [1500368337.281413040]: VICON Latency high: 2.72609e+20 s.
>   Latency: : 2.72609e+20 s.
> 
> [ WARN] [1500368337.281816987]: No updated pose for CF cf1 for 1500368337.281648 s.
> [ WARN] [1500368338.283294912]: VICON Latency high: 2.72609e+20 s.
>   Latency: : 2.72609e+20 s.
> ...

and only world frame and virtual frame are visible in RVIZ. No CF and no markers in rviz.

My Optitrack settings:
-No rigid body is selected/enabled
-I can see all four markers in Optitrack near the origin
Streaming settings:
-network interface. Local interface: 192.168.0.100
-streaming settings. All options are True, Scale: 1, Bone naming convention: Motive
-advanced network settings. Up axis: Z, Remote trigger: False, Type: multicast, Command port: 1510, Data port: 1511, Multicast interface: 192.168.1.188

naus3a, 2017-07-18:
I don't have a Optitrack system here, but your settings look reasonable.
The lines Wolfgang highlighted should be changed if you experience flipped
axis or scaled up/down translations.

The warnings you're getting seem to point out you're not getting pose
updates in a long time (or not getting pose at all): did you already check
your network setup and performance?

jpreiss, 2017-09-19:
@Negvet I am trying to close dead issues. Did you resolve the coordinate system / tracking problems with Optitrack?

closed 2017-12-13 by jpreiss

prbhtm, 2018-04-03:
I am using vrpn_client_ros for optitrack. When I use hover_optitrack.launch, which is the modified version of hover_vicon.launch, crazyflie goes in some direction. It does not stay at a fixed height. I think this is due to the coordinate system of optitrack. Can somebody tell me how to fix this issue?

jpreiss, 2018-04-05:
The Crazyflie's internal coordinate system is: X forward, Y left, Z up. The motion capture system should be configured to match. We do not have an OptiTrack system to test, but it is an option in Motive. Or you can modify the code as explained above by @whoenig.

Other things to check are:
- make sure your physical marker configuration matches the one defined in  crazyswarm/ros_ws/src/crazyswarm/launch/crazyflieTypes.yaml (line 79 onward) (an image of this configuration is in our ICRA 2017 paper "Crazyswarm: A Large Nano-Quadcopter Swarm")
- make sure the true initial position matches the one specified in  crazyswarm/ros_ws/src/crazyswarm/launch/allCrazyflies.yaml
- make sure the Crazyflie faces in the positive X direction when it is on the floor