USC-ACTLab/crazyswarm #23: Connection issue

math4356, 2017-07-20:
Hi !

When I was running hover_swarm.launch, I got these msgs !!! The pose information should have been sent from the motion capture system. 

> [ WARN] [1500535268.047072608]: No updated pose for motion capture object active-wand
> [ WARN] [1500535268.047446726]: No updated pose for CF cf1 for 1500535268.047005 s.
> [ WARN] [1500535268.055758347]: No updated pose for motion capture object active-wand
> [ WARN] [1500535268.055929900]: No updated pose for CF cf1 for 1500535268.055579 s.
> [ WARN] [1500535268.064536358]: No updated pose for motion capture object active-wand
> [ WARN] [1500535268.064662063]: No updated pose for CF cf1 for 1500535268.064438 s.
> [ WARN] [1500535268.072331422]: No updated pose for motion capture object active-wand
> [ WARN] [1500535268.072488554]: No updated pose for CF cf1 for 1500535268.072249 s.
> [ WARN] [1500535268.082445309]: VICON Latency high: 1.79446e+35 s.
>   Latency: : 1.79446e+35 s.
> 
> [ WARN] [1500535268.082805783]: No updated pose for motion capture object active-wand
> [ WARN] [1500535268.083112103]: No updated pose for CF cf1 for 1500535268.082738 s.
> [ WARN] [1500535268.089371782]: No updated pose for motion capture object active-wand
> [ WARN] [1500535268.089510510]: No updated pose for CF cf1 for 1500535268.089289 s.
> [ WARN] [1500535268.097756121]: No updated pose for motion capture object active-wand
> [ WARN] [1500535268.097857519]: No updated pose for CF cf1 for 1500535268.097697 s.
> [ WARN] [1500535268.105778800]: VICON Latency high: 4.2277e+22 s.
>   Latency: : 4.2277e+22 s.
> 
> [ WARN] [1500535268.106075287]: No updated pose for motion capture object active-wand
> [ WARN] [1500535268.106196562]: No updated pose for CF cf1 for 1500535268.106008 s.
> [ WARN] [1500535268.118769200]: No updated pose for motion capture object active-wand
> [ WARN] [1500535268.118908088]: No updated pose for CF cf1 for 1500535268.118691 s.
> [ WARN] [1500535268.122286767]: No updated pose for motion capture object active-wand
> [ WARN] [1500535268.122409086]: No updated pose for CF cf1 for 1500535268.122217 s.
> [ WARN] [1500535268.130102276]: VICON Latency high: 4.40615e+33 s.
>   Latency: : 4.40615e+33 s.
> 
> [ WARN] [1500535268.130462971]: No updated pose for motion capture object active-wand
> [ WARN] [1500535268.130643764]: No updated pose for CF cf1 for 1500535268.130376 s.
> [ WARN] [1500535268.139446908]: No updated pose for motion capture object active-wand
> [ WARN] [1500535268.139578887]: No updated pose for CF cf1 for 1500535268.139377 s.
> [ WARN] [1500535268.148501833]: No updated pose for motion capture object active-wand
> [ WARN] [1500535268.148657113]: No updated pose for CF cf1 for 1500535268.148416 s.
> [ WARN] [1500535268.155389313]: No updated pose for motion capture object active-wand
> [ WARN] [1500535268.155501778]: No updated pose for CF cf1 for 1500535268.155321 s.
> [ WARN] [1500535268.166161358]: No updated pose for motion capture object active-wand
> [ WARN] [1500535268.166661199]: No updated pose for CF cf1 for 1500535268.166007 s.
> [ WARN] [1500535268.171431744]: VICON Latency high: 2.24364e+29 s.
>   Latency: : 2.24364e+29 s.
> 
> [ WARN] [1500535268.171908429]: No updated pose for motion capture object active-wand
> [ WARN] [1500535268.172015267]: No updated pose for CF cf1 for 1500535268.171850 s.
> [ WARN] [1500535268.180322459]: No updated pose for motion capture object active-wand
> [ WARN] [1500535268.180847097]: No updated pose for CF cf1 for 1500535268.180239 s.
> [ WARN] [1500535268.190807302]: VICON Latency high: 6.58311e+10 s.
>   Latency: : 6.58311e+10 s.
> 
> 

My setting like these !!!

> 
>  motion_capture_type: "optitrack" # one of vicon,optitrack
>       object_tracking_type: "libobjecttracker" # one of motionCapture,libobjecttracker
>       #vicon_host_name: "vicon"
>       optitrack_local_ip: "10.0.0.18"
>       optitrack_server_ip: "10.0.0.2"
>       save_point_clouds: ~/pointCloud.ot
>       interactive_object: active-wand # virtual #active-wand
>       numMarkerConfigurations: 1
>       markerConfigurations:
>         "0":
>           numPoints: 4
>           offset: [0.0, -0.01, -0.04]
>           points:
>             "0": [-115.638420,1.984801,83.374680]
>             "1": [123.918533,-2.160125,56.46.5626]
>             "2": [-132.688553,1.471121,-55.763245]
>             "3": [124.408455,-1.295798,-84.077003]

jpreiss, 2017-07-20:
You should disable the interactive object by commenting out the `interactive_object: ...` line.

It seems that your network communication with your Optitrack server is not set up correctly. Are you sure the `optitrack_local_ip` and `optitrack_server_ip` are correct?

math4356, 2017-07-20:
These IP are correct.  I tried to comment out the interactive_object line but the issue still exist.

math4356, 2017-07-20:
Hi jpreiss @jpreiss 

Do you know what means of these command 

> if (m_tracker->objects()[i].lastTransformationValid()) 
in crazyflie_server.cpp.

jpreiss, 2017-09-07:
Hi @math4356, are you still having trouble connecting to Optitrack? The code you referenced is related to the internal frame-by-frame object tracking algorithm, but it is irrelevant if you can't even get point cloud data from the mocap server.

jpreiss, 2017-09-29:
Closing for now. If you still have trouble, please comment on this issue and we can reopen.

closed 2017-09-29 by jpreiss