USC-ACTLab/crazyswarm #24: On-line path planning

DarioMariniello, 2017-07-24:
Dear Whoenig,
I've studied your project since you suggested me to use it (I'm the one who was having issiues with control signal of the basic PID in the crazyflie_controller) and I think it is awesome! But since I have to adapt it for Qualisys system and spend more time on it, I would like to understand if it is the right solution for my project. I'm working on a decentralized collective task project with 3, maximum 4 Crazyflies; my drones are supposed to fly always on trajectories that consist of quarters of circles with always the same radius, but variable center (asynchronously changed but "really seldom", respect to the frequencies of the system, let's say every time a quarter of circle is spanned by the CF) and piecewise constant angular velocities (varied asynchronously a couple time during a quarter of circle). The problem is that the algorithm is supposed to generate the changes of centers and angular velocities online, based on the position of the CF (no worries about attitude, I usually keep it fixed). The question is: even if I could use the ellipses to get circular primitives, is there any chance to make the changes I said on-line, I.e. not planning the complete path off-line?what's your opinion about it?I'm worried about it because seems you use broadcasting communications during the flight, and I don't know if it could be compatible with these directed communications. For the discontinuities in the path, I could start trying to find initial conditions that minimize the difference between the consecutive centers.
Another question: is it possible to still use the modified firmware and the EKF with the merged state infos from both the mocap and the IMU but skipping the cloud point body identification, so using the identification from the Qualisys? Because in my application I'm OK with that, and I don't have the need to solve a large number of unique marker disposition.
Thanks in advance!

jpreiss, 2017-07-24:
Hi Dario,

It is possible to implement your project in an online way using ellipse trajectory commands, but it will require some changes to our firmware. Currently there is nowhere to save the next ellipse parameters while the crazyflie is already executing the current ellipse. Since the ellipse parameters and the "start ellipse" command are not in the same radio packet, you would need a "holding area" for the next ellipse. We implemented this for piecewise polynomials, but not for ellipses.

Right now, the `startEllipse` command is only available as a broadcast command in Python, but it's possible to add it as an individual command too. We just didn't need it yet :) Since the position of the crazyflies is also available in Python, you could probably write your online planner as a python script.

Alternately, if the logic to change the circle parameters is simple enough, you could implement in firmware if you want a truly distributed system. (In firmware, every crazyflie has access to the position of every other crazyflie via broadcasting; however we do not use this anywhere yet.)

I would be willing to write the ellipsoid "holding area" firmware change and do code review on the ROS/python changes, but I can't commit to any real robot testing; you would need to do that.

One possible problem is smoothness. Suppose you want to implement an "S" curve. If you implement it with two circles of equal period and length, but different centers, you will get continuous position and velocity but discontinuous acceleration. Since the quadrotor's acceleration is coupled with its orientation, when you transition between circles you will be asking the quadrotor to flip its orientation instantaneously. However, this might not be a problem because the firmware already uses onboard polynomial planning to transition into an ellipse trajectory. The severity of the problem increases with speed/aggressiveness.

Re. Qualisys: yes, you can use the motion capture system's native object tracking by changing `object_tracking_type` to `motionCapture` in the launch file, see lines 54-55 [here](https://github.com/USC-ACTLab/crazyswarm/blob/master/ros_ws/src/crazyswarm/launch/hover_swarm.launch).

jpreiss, 2017-09-19:
Hi Dario, I am trying to close dead issues. Are you still interested in this project?

DarioMariniello, 2017-09-20:
Dear James, yes I am, but you can close the issue, becuse I will probably
tackle it again in a few months, and when I will I'll open a new issue or
contact you or Whoenig directly if I really need your help.
Thanks a lot!
Dario

closed 2017-12-13 by jpreiss