USC-ACTLab/crazyswarm #25: Crazyswarm- No sensor data from Crazyflies

rbellido1, 2017-08-30:
Hi Everyone,
I would like to test the Crazyswarm project from USC-ACTLab with my Crazyflies. At this point, I was able to compile the working space, updated the firmwares for Crazyradio and Crazyfiles and Changed the address. Basically I followed the instructions of this link: http://crazyswarm.readthedocs.io/en/latest/
Before testing using MOCAP system, I tested the communication part between Crazyradio and Crazyflie using the Python-Client GUI. The GUI found the new interface (new addres) and connected successfully but when I moved the crazyflie around it seems that there is no data coming from crazyflie sensors since the "Flight Data" window never got updated. But when I tested with another crazyflie, using the same crazyradio, new firmware, the "Flight Data" window got updated everytime I moved around the crazyflie. It seems that with the new crazyflie firmware there is no data coming out. I tried another test using the new crazyradio (new firmware from crazyswarm project) and a default crazyflie firmware and the result showed that there is sensor data from the crazyflie.
Does anyone know if all sensor data coming from the crazyflies, new firmware-crazyswarm, was disable, maybe for reducing traffic? 
I think I am flashing the Crazyflie using the right method since I used this link: https://www.bitcraze.io/getting-started ... #latest-fw.
Basically, I put the two bin files, nrf and stm, generated by crazyswarm project plus the manifets.json and Zip it.
I'd very much appreciate any help. Many thanks
Robert B.

jpreiss, 2017-09-01:
Hi Robert, it's expected that the Crazyswarm firmware will not work with the Bitcraze client. The Crazyswarm firmware basically will refuse to do anything until it starts getting pose data from the motion capture system. 

It's easier to flash firmware by using the `make cload` command. Then you don't need to make the .zip

jpreiss, 2017-09-01:
By the way, there is no intentional reason why manual mode doesn't work. It's just that we are focused on the mocap, autonomous case, so we neglected manual mode as we were adding the necessary safety checks and other features to support our needs. It's totally possible to add some runtime mode switch to allow both.

rbellido1, 2017-09-01:
Hi James,
Many thanks for your reply and clarification about the Bitcraze client.
Also, I have been testing the Crazyswarm firmware with a single Crazyflie and when I run the hover_swarm.launch file I was able to get the /pointCloud data but also many warning lines, like below:

[WARN] [1504221811.727503164]: Dynamic check failed
wyaw: 13.6166 >= 10

[WARN] [1504221811.727549794]: No updated for CF cf1 for 0.003114s.
[WARN] [1504221811.865071562]: VICON Latency high: 0.0371055 s.
 Latency: Data Delivered: 0.0137766 s.
 Latency: Data Synchronized: 0.00044014 s.
 Latency: Data Processed: 0.0225299 s.
 Latency: Data Delivered To DataStream: 0.000358925 s.
Could you please provide some feedback to fix this issue?
I think I have a communication between my ViconPC and LinuxPC (pinged for test). On the hover_swarm.launch, for the "vicon_host_name" used "Vicon_pc_IP:801" and for the "object_tracking_type" used "libobjecttracker".  My understanding is that "vicon_host_name" is used by ViconDataStreamSDK libraries (Client.h) and I don't need a vicon_bridge involve. Could please confirm or clarify this point?

I am using just one Crazyflie and place at [1.5, 1.5, 0] and used, inside the hover_swarm.launch, an "offset": [-1.5, -1.5, 0.0]. For the markers location, I used the global coordinate values, like below:
numMarkerConfigurations: 1
      markerConfigurations:
        "0":
          numPoints: 4
          offset: [-1.5, -1.5, 0.0]
          points:
            "0": [1.509525,1.50635,0.03175]
            "1": [1.476595,1.521454,0.019075]
            "2": [1.47885178,1.479728477,0.019075]
            "3": [1.522235,1.479619,0.019075]
      numDynamicsConfigurations: 1
For this part, I followed the recommendation from Wolfgang's post , 
https://github.com/USC-ACTLab/crazyswarm/issues/4

Could you please let me know if i am doing correctly this part ? The Dynamic check failed warning doesn't make sense for me since I aligned the CF with X-axis positive.
Many thanks for your help again.
Robert Bellido

jpreiss, 2017-09-07:
Hi Robert, I believe your current progress is all related to #26, so I can close this issue, correct?

rbellido1, 2017-09-19:
Hi James,
Sorry. I just saw your msg. Please close the the issue.
Thanks,
Robert

closed 2017-09-19 by jpreiss