USC-ACTLab/crazyswarm #26: Markers layout on the hover_swarm.launch

rbellido1, 2017-09-02:
Hi Everybody,
I have been testing the Crazyswarm firmware using a single Crazyflie. Today, I was able to fly the crazyflie, running the hover_swarm.lauch and then figure8_canned.py. The hovering part was good  but the fly was not stable. I got too many warning messages during the fly, like below:

[pcl::IterativeClosestPoint::computeTransformation] Not enough correspondences found. Relax your threshold parameters. 
[ WARN] [1504308616.146888169]: ICP did not converge! 
[ WARN] [1504308616.146942180]: No updated pose for CF cf1 for 0.013919 s. 
[pcl::IterativeClosestPoint::computeTransformation] Not enough correspondences found. Relax your threshold parameters. 
[ WARN] [1504308616.172143550]: ICP did not converge! 
[ WARN] [1504308616.172176086]: No updated pose for CF cf1 for 0.012231 s. 
[ WARN] [1504308616.197821675]: Dynamic check failed 
wroll: 153.948 >= 20 
wyaw: -50.948 >= 10 
roll: 2.03238 >= 1.4 

[ WARN] [1504308616.197870392]: No updated pose for CF cf1 for 0.015990 s. 
[pcl::IterativeClosestPoint::computeTransformation] Not enough correspondences found. Relax your threshold parameters. 
[ WARN] [1504308616.214641650]: ICP did not converge! 
[ WARN] [1504308616.214687627]: No updated pose for CF cf1 for 0.009467 s. 
[pcl::IterativeClosestPoint::computeTransformation] Not enough correspondences found. Relax your threshold parameters. 
[ WARN] [1504308616.232144194]: ICP did not converge! 
[ WARN] [1504308616.232182803]: No updated pose for CF cf1 for 0.010545 s. 

I placed my crazyflie at [1.5,1.5, 0.0] and inside the hover_swarm.launch file, my markers layout is:

markerConfigurations:
        "0":
          numPoints: 4
          offset: [-1.5, -1.5, 0.0]
          points:
            "0": [1.509525,1.50635,0.03175]
            "1": [1.476595,1.521454,0.019075]
            "2": [1.47885178,1.479728477,0.019075]
            "3": [1.522235,1.479619,0.019075]

Markers locations are in global coordinate frame and I used an offset of [-1.5, -1.5, 0.0]. Basically for this part, I followed this previous post: https://github.com/USC-ACTLab/crazyswarm/issues/4
Does anyone notice what is the error with this marker configuration?
I'd very much appreciate any help. Many thanks
Robert Bellido

jpreiss, 2017-09-02:
Is your marker configuration symmetrical?

rbellido1, 2017-09-02:
Thanks James again for your reply.
My marker configuration is asymmetrical, three markers over the arms and one over the battery area. Does the configuration has to be symmetrical with this tracking method approach? I thought that it was the same concept that the object method is used for tracking, use an assymetrical configuration. Please clarify this part.
Many thanks,
Robert

whoenig, 2017-09-02:
The marker configuration should be asymmetrical. In your other post you also mentioned that there was high VICON latency reported - make sure that the space is calibrated, no objects are selected in VICON Tracker, and that the latency is below 20ms. You should also verify in Tracker (or RViz) that all 4 markers are nearly always visible while you are flying - if one of them is frequently occluded, the frame-by-frame tracking can get into local minima.

rbellido1, 2017-09-02:
I have been using old calibrated space for my test. Also, I haven't checked VICON latency values before flying or how visible are the markers during the flying. 
Do you think if I change my markers configuration to a more assymetrical configuration, it would help with reducing the warning messages?
Anyway, I will test again with a new calibrated space, checking the markers visibility, first and then with a new more assymetrical configuration. I will let you know the results. 
Many thanks again for your help
Robert.

whoenig, 2017-09-02:
I don't think the symmetry matters very much for your case - it is less critical than for objects tracked by VICON Tracker, because we track frame-by-frame.
We need to calibrate frequently, because the markers only show up as ~3 pixels in some of the cameras. Small camera disturbances can then lead to bad and slow marker position estimates. You should check that on your side as well - I assume you have to use smaller markers than for other tracked objects.

rbellido1, 2017-09-04:
Thanks Whoenig for your reply.
I am using smaller markers, 6.3 mm,  and this is the smallest size that i bought it. Do you recommend smaller than 6.3 mm? I noticed that the weight of this markers (4) reduce around 40-50% of the battery time. Could please tell what size and type you use?. Thanks for your feedback again.
Robert

whoenig, 2017-09-04:
We use 7.9mm traditional spherical markers from B&L Engineering (no base option). Smaller is better iff your Vicon space can handle it. Look at the grayscale camera image in VICON Tracker to check how many pixels each marker occupies in the camera view.

rbellido1, 2017-09-04:
Whoenig,
This is a good info so I will check how many pixels each marker occupies in the cameras view. If I would get less than 3, I would increase my marker size to 7.9mm. Also, I got my makers from B&L Engineering, soft base.
Many thanks again.
Robert

jpreiss, 2017-09-04:
btw, I don't think we are losing that much battery life from markers. Are
you also adding other mass like decks, prop guards, etc?

rbellido1, 2017-09-05:
James,
I didn't add extra accessory to crazyflie but I use litlte heavier batteries than the original that i bought from Amazon. Also, I think my marker is heavier than yours. My marker comes with soft base, like light lether, plus little metal screw to keep the base attached to the base. For my next test, I will use the original batteries to see just the weight effect of the markers. Thanks for your reply.
Robert

rbellido1, 2017-09-08:
Hi Whoenig and James,
I followed all your previous recommendations and my single Crazyflie fly test improved. Keeping the VICON latency less than 20 was critical. 
But I still get warnings and I think is due to the number of pixels per marker since some markers get less than 3 pixels . I am planning to increase the size of my marker to 7.9mm to get more pixels per marker and hope this change will get rid of some of the warnings.
Another question guys, Today I tried to fly two crazyflies using one radio so I changed the address to CF1 and CF4 (E7E7E7E701, E7E7E7E704) and also the crazyflies.yaml with CF1 and CF4.
The part is not clear for me is the hover_swarm.launch file so I reviewed the source code (readMarkerConfiguration()) and noticed the value for numMarkerConfigurations and numDynamicsConfigurations has to be 2 and also I need to add another block of values for the points. Also, the function readObjects() showed that the address are created reading the "id" value from the crazyflies.yaml file. Thus, I didn't change the "broadcast_address" value on the hover_swarm.launch.
So when I run the hover_swarm.launch the two crazyflies are added but just one established radio communication and flied, please see below my terminal output:

[ INFO] [1504830865.258652102]: Adding CF: cf1 (radio://0/100/2M/E7E7E7E701, cf1)... 
[ INFO] [1504830865.262238064]: CF ctor: 0.003481 s 
[ INFO] [1504830865.270021106]: [cf1] Requesting parameters... 
[ INFO] [1504830865.277859280]: Found variables in cache. 
[ INFO] [1504830865.432825654]: [cf1] reqParamTOC: 0.170527 s 
[ INFO] [1504830865.432860976]: [cf1] Ready. Elapsed: 0.170594 s 
[ INFO] [1504830865.432878926]: CF run: 0.170667 s 
[ INFO] [1504830865.444521612]: [cf1] Update parameters 
[ INFO] [1504830865.491383620]: Update params: 0.058466 s 
[ INFO] [1504830865.491408117]: Adding CF: cf4 (radio://0/100/2M/E7E7E7E704, cf4)... 
[ INFO] [1504830865.494393236]: CF ctor: 0.002956 s 
[ INFO] [1504830865.582290535]: [cf4] Requesting parameters... 
[ WARN] [1504830865.583914191]: [cf4] Link Quality low (0.180000) 
[ INFO] [1504830865.590772775]: Found variables in cache. 
[ WARN] [1504830865.667932620]: [cf4] Link Quality low (0.380000) 
[ WARN] [1504830865.751503132]: [cf4] Link Quality low (0.300000) 
[ WARN] [1504830865.836887987]: [cf4] Link Quality low (0.360000) 
[ WARN] [1504830865.917099597]: [cf4] Link Quality low (0.360000) 
[ INFO] [1504830865.955337005]: [cf4] reqParamTOC: 0.460907 s 
[ INFO] [1504830865.955359194]: [cf4] Ready. Elapsed: 0.460944 s 
[ INFO] [1504830865.955371295]: CF run: 0.460997 s 
[ INFO] [1504830865.965143035]: [cf4] Update parameters 
[ WARN] [1504830866.057617684]: [cf4] Link Quality low (0.330000) 
[ WARN] [1504830866.143241205]: [cf4] Link Quality low (0.210000) 
[ INFO] [1504830866.212833692]: Update params: 0.257408 s 
[ INFO] [1504830866.212982580]: Started 1 threads 

Could you please tell me what I am missing on the hover_swarm.launch or another file? I can fly each crazyflie individually using the same hover_swarm.launch, just changing the broadcast_address value.
Many thanks for your help again.
Robert Bellido

whoenig, 2017-09-08:
* You should only need a single block for marker and dynamics configurations (assuming that the marker configuration for your two CFs is identical)
* The ID has to match the address in hex (e.g. E7E7E7E70A would be id 10)
* You don't need to change the broadcast address. This address is identical (and hard-coded) for all CFs.
* According to your log, the connection to both cf1 and cf4 were fine, but cf4 had worse connectivity (i.e. high packet drop rate). You can try a different channel, or a different CF (we have one where the antenna is worse than for all the others.)

rbellido1, 2017-09-08:
Whoenig,
Thanks for your reply. My marker configuration is identical for both CFs  and the crazyflies. yaml file includes both CFs IDs, 1 and 4.
But the part I still confuse is about the offset[ ] values. In one of your previous post (https://github.com/USC-ACTLab/crazyswarm/issues/4) you said if the CF is place at [1, 0, 0] then the offset for this CF should be [-1,0,0]. Now what happen when you have two CFs with different locations. How do you apply the offset on launch file? For instance my CF1 is at [1.5, 1.5, 0] and my CF4 is at [1.5, 0, 0], based on my understanding I need two offset values [-1.5,-1.5,0]  and [-1.5,0,0]. Also, the block of markers, the location of the markers are in global coordinate, origin of space at [0,0,0]. Could you please clarify this part? I am missing something on this part. Regarding to hig packet drop rate, I am using Crazyradio but i just bought a new Crazyradio PA, the one that you use. I will use this new radio since the strength of the signal is greater to see if link quality improve. Last question, as reference, do you remember your threshold value for your cameras? I am currently using 0.2
Many thanks for yor help.
Robert Bellido

whoenig, 2017-09-09:
1. The offset in the launch file is just to make it easier to specify the marker configuration. a) place one of your CFs at a known position, b) use VICON Tracker to find the global coordinates of your 4 markers, c) the offset is essentially where you placed your CF and the coordinates of your markers are the marker configuration. Once you actually fly, you use the initialPosition in crazyflies.yaml (not the launch file) to specify where each CF is located at initially.
2. I don't remember our threshold value, but this also highly depends on your physical setup, the exact camera model (& lens) you use etc. One way is to just look at the grayscale image of the different cameras and place markers in the area of interest and make sure that the raw images look good - some settings need to be changed on the cameras themselves (aperture and focus).  If you still have a service contract, VICON is likely able to give you some more tips.

rbellido1, 2017-09-11:
Hi Whoenig.
Thanks for your reply. 
With your explanation and reviewing again the readMarkersConfigurations() and readObject() functions, I finally got it. Many thanks again.
Robert Bellido

jpreiss, 2017-09-19:
@rbellido1 I am trying to close dead issues. All your current problems are covered by #28, correct?

rbellido1, 2017-09-20:
Hi James,
Yes, that's correct. Please close this issue.
Thanks.

closed 2017-09-20 by jpreiss