USC-ACTLab/crazyswarm #27: Missing binaries in a for VICON DataStream SDK

javieryu, 2017-09-18:
The currently available version of the SDK does not include:_Client.h_, or _libDebugServices.so_. I tried renaming the _DataStreamClient.h_ in the current SDK (version 1.7) to _Client.h_ and moved the other files to the appropriate locations, but the package doesn't seem to be working. I am able to _catkin\_make_ and launch _hover\_swarm.launch_, but no CF TF frames are published.

I have tested my VICON setup with the vicon_bridge package, and that is working (that package uses an older version of the SDK too).

whoenig, 2017-09-18:
The Crazyswarm is tested with VICON Datastream SDK 1.5. VICON explicitly disallowed us to distribute the DataStream SDK as part of the Crazyswarm repositories. You can contact their support to get version 1.5.

rbellido1, 2017-09-19:
Hi Javieryu,
I had the same problem before and I ended up renaming the DataStreamClient.h to Client.h, using SDK version 1.55.  Just make sure that in the hover_swarm.launch the "vicon_host_name" parameter is update with "ViconIP:801" instead of just  "vicon". Otherwise you won't establish communication with DataStreamSDK.

closed 2017-09-25 by javieryu