USC-ACTLab/crazyswarm #28: Communication issue with multiple crazyflies and one radio

rbellido1, 2017-09-19:
Hi Whoenig and James,
I still have a problem with running the crazyswarm firmware with two Crazyflies (IDs:1 and 4, Channel:100) control by one Crazyradio. I followed your previous recomendations, for instance I kept the broadcast_address: "E7E7E7E7E7" on the hover_swarm.launch and the crazyflie.yaml files includes the Ids 1 and 4. I changed the address of each crazyflie, 100/2M/FFE7E7E701 and 100/2M/FFE7E7E704. When I launched the hover_swarm.launch, i got a clean log terminal output. Please see below
[ INFO] [1505518638.809883208]: Wait for services... 
[ INFO] [1505518638.810348943]: waitForService: Service [/emergency] has not been advertised, waiting... 
ch: 1 
[ INFO] [1505518638.831057555]: waitForService: Service [/emergency] is now available. 
[ INFO] [1505518638.839883336]: Manager ready. 
[ INFO] [1505518638.943498040]: Adding CF: cf1 (radio://0/100/2M/E7E7E7E701, cf1)... 
[ INFO] [1505518638.948093670]: CF ctor: 0.004514 s 
[ INFO] [1505518638.955816293]: [cf1] Requesting parameters... 
[ INFO] [1505518638.963545717]: Found variables in cache. 
[ INFO] [1505518639.085565069]: [cf1] reqParamTOC: 0.137428 s 
[ INFO] [1505518639.085584616]: [cf1] Ready. Elapsed: 0.137465 s 
[ INFO] [1505518639.085594584]: CF run: 0.137528 s 
[ INFO] [1505518639.096141981]: [cf1] Update parameters 
[ INFO] [1505518639.139911742]: Update params: 0.054281 s 
[ INFO] [1505518639.139935655]: Adding CF: cf4 (radio://0/100/2M/E7E7E7E704, cf4)... 
[ INFO] [1505518639.142968205]: CF ctor: 0.003005 s 
[ INFO] [1505518639.163414395]: [cf4] Requesting parameters... 
[ INFO] [1505518639.171424352]: Found variables in cache. 
[ INFO] [1505518639.305536762]: [cf4] reqParamTOC: 0.162530 s 
[ INFO] [1505518639.305565768]: [cf4] Ready. Elapsed: 0.162576 s 
[ INFO] [1505518639.305578519]: CF run: 0.162628 s 
[ INFO] [1505518639.317867972]: [cf4] Update parameters 
[ INFO] [1505518639.363879406]: Update params: 0.058251 s 
[ INFO] [1505518639.364018701]: Started 1 threads 

Based on log info, I could say that the crazyflie.yaml was parsed correctly but the part is not working the radio link connection between CFs and crazyflies. When I checked the M4 LEDs, the one get lit up with yellow light indicating that the connection was established, both M4 LEDs were off. Thus, no communication established. But If I changed the broadcast_address to  "E7E7E7E701" or "E7E7E7E704", I can establish radio communication individually, just one CF.

Also, I tested the same CFs with same radio using the "crazyflie_ros" package and the result was positive, communication was established, both M4 LEDs were lit up. Based on this results, it seems I have a problem updating the addresses coming from CFs on the crazyswarm working space.
Could please provide any help where I could take a look for debugging this issue?

Many thanks,
Robert Bellido

whoenig, 2017-09-19:
The communication in the Crazyswarm is using broadcasting (except for the initial connection) and thus your LED indicators might not work as expected. If you have an LED ring attached to the CFs, they show the communication status by default (red = no packets received; green = good communication). Your initial log shows that there was no problem connecting to both CFs.

Also, make sure that the tracking is working properly (e.g. by looking at the tf frames in rviz) - if the CFs are not tracked, no packets are being send to the CFs.

rbellido1, 2017-09-22:
Thanks for your reply Whoenig.
I followed your recommendations so checked tf frames in rviz and I am getting data for both CFs. I don't have LED ring attached to CFs. Also I flashed the firmwares for two more crazyflies (CF2 and CF5) and tested with the same results. Please see the pic for the rviz results.
[image].
But there is one difference compare with the previous logs that I haven't got the thread info like"[ INFO] [1505519597.433256325]: Started 1 threads ".
At this point I think my problem is that i haven't flashed the firmwares properly. To flash the firmwares I used the Crazyflie PC Client and followed the procedure from your paper "Flying Multiple UAVs using ROS". In order to create the zip file, I get the bin files for STM32 and NRF51 that I generated on crazyswarm after running the ./build.sh. But first I renamed the bin files accordingly to the names  on manifest.json and finally zip them. My understanding is when the PC Client is used both firmwares need to upload at the same time.
Yesterday, I tried to followed the procedure from crazyswarm documentation for flashing, https://wiki.bitcraze.io/doc:crazyflie:dev:starting. But It was not clear for me, the first thing that I tried understand what does "make cload"?.
I googled "make cload" and I couldn't find any information. Is this something that you created? .Also I found one of your previous post regarding to "make cload", https://github.com/USC-ACTLab/crazyswarm/issues/10, and tried to use it, like this
crazyswarm$ cd crazyflie-firmware
make
make cload
But I got errors: 
rbellido@holmes:~/crazyswarm/crazyflie-firmware$ make cload 
../crazyflie-clients-python/bin/cfloader  flash cf2.bin stm32-fw 
Restart the Crazyflie you want to bootload in the next 
 10 seconds ... 
Cannot connect the bootloader! 
make: *** [cload] Error 255 
With this method, do you need to "make cload" each bin file separate (crazyswarm/crazyflie-firmware AND crazyswarm/crazyflie2-nrf-firmware)? Do I need to use USB cable between the CF and PC for the communication or a Crazyradio in order to "make cload"? 
I don't have at this moment any of the programming cables mentioned on the paper but I could buy them if it is necessary.  Could you please provide some details about how you flash the firmware using "make cload" from the terminal. if it is possible?.
Last question, on my previous tests with multiple CFs, I used the  script "figure8_canned.py" for flying. Is this the correct script for flying multiples CFs?
Thank you for all your help.
Robert Bellido

whoenig, 2017-09-22:
1. `make cload` is the recommended method. The idea with the zip file should technically work, but we didn't test it for the Crazyswarm. The flashing will use the Crazyradio dongle as well. The steps are the following: a) Put the battery into the CF. b) Press the power button for 3 seconds. The blue LED should start blinking. The firmware is now in bootloading mode. c) run `make cload` on the command line. For best results, place the CF close to the radio. You will need to repeat this procedure for STM32 and NRF51.
2. figure8_canned is a good starting point. You can also just press start on your xbox360 controller to trigger a takeoff (and press back to land again), without running a python script.
3. There will be as many threads as channels/radios used. So in your case there should be only 1 thread.
4. Make sure your CF is on a level surface while booting, otherwise its self-test might fail during sensor calibration which would cause it to never take off.

rbellido1, 2017-09-28:
Hi Whoenig,
Finally,I was able to establish radio communication between the 2CFs and one crazyradio. I reflashed all the CFs using your method and  I also noticed that I was using for the broadcast_address "E7E7E7E7E7" instead of "FFE7E7E7E7". I still get the warning messages but I think I need to improve Vicon coverage by rearranging the cameras in order to get rid off these messages.
Thank you again for all your help! I will close this post.
Robert Bellido