USC-ACTLab/crazyswarm #29: Build Crazyswarm Failed

EdXian, 2017-10-01:
Hi team ,
I have tried  to build the project with `.\build.sh` but failed  many times eventhough i add the vicon_sdk to 
following path : 
[image]
finally I got error : 

```
[  5%] Linking CXX shared library /home/pinxian/crazyswarm/ros_ws/devel/lib/liblibmotioncapture.so
/usr/bin/ld: skipping incompatible /home/pinxian/crazyswarm/ros_ws/src/crazyflie_ros/externalDependencies/libmotioncapture/externalDependencies/vicon_sdk/lib64/libViconDataStreamSDK_CPP.so when searching for -lViconDataStreamSDK_CPP
/usr/bin/ld: cannot find -lViconDataStreamSDK_CPP
/usr/bin/ld: skipping incompatible /home/pinxian/crazyswarm/ros_ws/src/crazyflie_ros/externalDependencies/libmotioncapture/externalDependencies/vicon_sdk/lib64/libViconDataStreamSDK_CPP.so when searching for -lViconDataStreamSDK_CPP
/usr/bin/ld: cannot find -lViconDataStreamSDK_CPP
collect2: error: ld returned 1 exit status
crazyflie_ros/externalDependencies/libmotioncapture/CMakeFiles/libmotioncapture.dir/build.make:415: recipe for target '/home/pinxian/crazyswarm/ros_ws/devel/lib/liblibmotioncapture.so' failed
make[2]: *** [/home/pinxian/crazyswarm/ros_ws/devel/lib/liblibmotioncapture.so] Error 1
CMakeFiles/Makefile2:1049: recipe for target 'crazyflie_ros/externalDependencies/libmotioncapture/CMakeFiles/libmotioncapture.dir/all' failed
make[1]: *** [crazyflie_ros/externalDependencies/libmotioncapture/CMakeFiles/libmotioncapture.dir/all] Error 2
make[1]: *** Waiting for unfinished jobs....
[  5%] Linking CXX shared library /home/pinxian/crazyswarm/ros_ws/devel/lib/liblibobjecttracker.so
[  5%] Built target libobjecttracker
Makefile:138: recipe for target 'all' failed
make: *** [all] Error 2
Invoking "make -j2 -l2" failed

```
Can you guys give some suggestion or give me  download link of full package?
thanks!!

Ubuntu 16.04
ROS :　kinetic

whoenig, 2017-10-02:
Are you planning to use VICON? If though, you will need to contact VICON support to ask for the DataStream SDK version 1.5 (i.e. older than what they currently advertise online). If you are not planning to use VICON, you can compile the crazyswarm stack without VICON support.

The-SS, 2017-11-21:
Hi,
I am having a similar problem, but I get two other errors.
I added DataStreamClient.h and ran ./build.sh. I got the following errors:
``` 
Scanning dependencies of target listLogVariables
[ 46%] Building CXX object crazyflie_ros/crazyflie_tools/CMakeFiles/listLogVariables.dir/src/listLogVariables.cpp.o
/home/hivemind/cs_ws_2/crazyswarm/ros_ws/src/crazyflie_ros/externalDependencies/libmotioncapture/src/vicon.cpp:4:40: fatal error: vicon_sdk/DataStreamClient.h: No such file or directory
compilation terminated.
crazyflie_ros/externalDependencies/libmotioncapture/CMakeFiles/libmotioncapture.dir/build.make:86: recipe for target 'crazyflie_ros/externalDependencies/libmotioncapture/CMakeFiles/libmotioncapture.dir/src/vicon.cpp.o' failed
make[2]: *** [crazyflie_ros/externalDependencies/libmotioncapture/CMakeFiles/libmotioncapture.dir/src/vicon.cpp.o] Error 1
CMakeFiles/Makefile2:1049: recipe for target 'crazyflie_ros/externalDependencies/libmotioncapture/CMakeFiles/libmotioncapture.dir/all' failed
make[1]: *** [crazyflie_ros/externalDependencies/libmotioncapture/CMakeFiles/libmotioncapture.dir/all] Error 2
make[1]: *** Waiting for unfinished jobs....
/home/hivemind/cs_ws_2/crazyswarm/ros_ws/src/crazyflie_ros/crazyflie_manager/src/manager.cpp:10:47: fatal error: crazyflie_driver/UploadTrajectory.h: No such file or directory
compilation terminated.
crazyflie_ros/crazyflie_manager/CMakeFiles/crazyflie_manager.dir/build.make:62: recipe for target 'crazyflie_ros/crazyflie_manager/CMakeFiles/crazyflie_manager.dir/src/manager.cpp.o' failed
make[2]: *** [crazyflie_ros/crazyflie_manager/CMakeFiles/crazyflie_manager.dir/src/manager.cpp.o] Error 1
CMakeFiles/Makefile2:3852: recipe for target 'crazyflie_ros/crazyflie_manager/CMakeFiles/crazyflie_manager.dir/all' failed
make[1]: *** [crazyflie_ros/crazyflie_manager/CMakeFiles/crazyflie_manager.dir/all] Error 2
/home/hivemind/cs_ws_2/crazyswarm/ros_ws/src/crazyflie_ros/crazyflie_controller/src/controller.cpp:9:58: fatal error: crazyflie_controller/ExecuteTrajectoryAction.h: No such file or directory
compilation terminated.
crazyflie_ros/crazyflie_controller/CMakeFiles/crazyflie_controller.dir/build.make:62: recipe for target 'crazyflie_ros/crazyflie_controller/CMakeFiles/crazyflie_controller.dir/src/controller.cpp.o' failed
make[2]: *** [crazyflie_ros/crazyflie_controller/CMakeFiles/crazyflie_controller.dir/src/controller.cpp.o] Error 1
CMakeFiles/Makefile2:4665: recipe for target 'crazyflie_ros/crazyflie_controller/CMakeFiles/crazyflie_controller.dir/all' failed
make[1]: *** [crazyflie_ros/crazyflie_controller/CMakeFiles/crazyflie_controller.dir/all] Error 2
[ 46%] Linking CXX executable /home/hivemind/cs_ws_2/crazyswarm/ros_ws/devel/lib/crazyflie_tools/listLogVariables
[ 46%] Built target listLogVariables
Makefile:138: recipe for target 'all' failed
make: *** [all] Error 2
Invoking "make -j4 -l4" failed
```
It does not find vicon_sdk/DataStreamClient.h, crazyflie_driver/UploadTrajectory.h, or crazyflie_controller/ExecuteTrajectoryAction.h.

Any help is appreciated!
Thank you.

PS:
I'm using:
Ubuntu: 16.04 LTS
ROS: Kinetic
Tracking: VICON

whoenig, 2017-11-21:
I recently upgraded to DataStreamSDK 1.7 to make it less painful. Did you put the SDK files in the correct folders (as described in: http://crazyswarm.readthedocs.io/en/latest/installation.html)?

The other missing files might be related to dependency issues. Once the SDK is found, you can try running `catkin_make -k` in the folder `ros_ws`. This will tell catkin to keep going even after an error. Repeat the command 2 or 3 times to rule out dependency issues.

The-SS, 2017-11-21:
Regarding the VICON DataStreamSDK, I downloaded the SDK from their website and I copied the files from `Linux64-boost-1.58.0` folder and added them as described in the installation guide on the crazyswarm webpage. Below is an image that shows how the directories and files are setup.
[image]

The other two errors were dependency errors and were solved by running`catkin_make -k` a couple of times in `ros_ws`.

whoenig, 2017-11-21:
I think the path in the documentation is wrong. It should be `ros_ws/src/crazyflie_ros/externalDependencies/libmotioncapture/externalDependencies/vicon_sdk` (i.e. vicon_sdk should be next to the NatNetLinux folder). Let me know if that fixes the issue and I'll update the docs accordingly.

The-SS, 2017-11-21:
Yes, that solved the problem. Now the code compiles without giving any errors.
So, the correct directory to place the vicon_sdk folder is: `ros_ws/src/crazyflie_ros/externalDependencies/libmotioncapture/externalDependencies/vicon_sdk` as you suggested.
Also, when updating the docs, I think it would be helpful if you could replace `git clone https://github.com/USC-ACTLab/crazyswarm.git` by `git clone --recursive https://github.com/USC-ACTLab/crazyswarm.git` to initialize and update all submodules.

Thank you.

whoenig, 2017-11-21:
Thanks for the update. The path is fixed in the online documentation.

We initialize the submodules in `build.sh`. I think there is a problem with the crazyflie-firmware dependencies if you use the `--recursive` flag (it downloads more submodules than needed for the testing infrastructure they use).

The-SS, 2017-11-21:
I tried running `./build.sh` once, but it did not initialize and update all the submodules properly. Thus, I tried using `--recursive`. 
I built the package, again, using `./build.sh` and this time it worked well. Nevertheless, I did have to run `catkin_make -k` twice in `ros_ws` then go back and rerun `./build.sh`.

Thank you.

jpreiss, 2017-12-18:
We will work on fixing the dependencies in the catkin_make file such that running `catkin_make -k` twice is not necessary.

closed 2020-06-13 by jpreiss