USC-ACTLab/crazyswarm #30: Optitack integration

itzzartem, 2017-10-17:
Does the crazyswarm support rigid body's sent from optitrack or only point cloud data (marker positions)?

whoenig, 2017-10-17:
It supports both. The name of the rigid body has to match the ID of the CF (e.g. "cf1" for a CF with ID 1, address 0xE7E7E7E701).

itzzartem, 2017-10-17:
Does it go through multicast or vrpn connection for server ip?

itzzartem, 2017-10-17:
Ok well I am using the multicast address which is the NatNet SDK address but i continue to get a warning "[ WARN] [1508272203.868192904]: VICON Latency high: 6.64778e+22 s.
  Latency: : 6.64778e+22 s.

[ WARN] [1508272203.868293052]: No updated pose for motion capture object cf1"
and there is nothing displayed in the rviz environment.
The rigid body is name "cf1" and the config file is only set up for 1 crazyflie. 
How can I fix this?

whoenig, 2017-10-17:
Sounds like it couldn't establish a connection to your optitrack PC. @naus3a, do you have any troubleshooting ideas?

naus3a, 2017-10-18:
as @whoenig said, it sounds like a network issue;I never tested multicast or VPN and, while  it it should work in theory, I can imagine a couple of possible problems:

- depending on their configuration, some routers will not allow multicast packets to to travel from a subnet to another. Use a packet sniffer, or some other network diagnostic tool to check that you're receiving the packets at all.

- vpn tunneling can cause a certain latency: the software needs very low latency levels to work properly, so check your ping too.

Finally, in order a good way to learn if it's a network issue or not, it would be a good idea to test your setup with a simple network setup: just your cameras and your computer, connected through a switch.

itzzartem, 2017-10-18:
I actually ran into the first point you mentioned earlier on when working with the mocap_ros package. I am currently running everything with a direct LAN connection between the two computers and can get rigid body data both through multicast and VRPN. Since your package was developed using the NatNet SDK I believe it is using multicast to broadcast the data. That's why i'm a little confused on where the networking issue is coming from.

naus3a, 2017-10-18:
The NatNet side of things was wrapped around NatNetLinux, because (as far as we could see) the official NatNet SDK did not support Linux. Not sure if this could make any difference.

If you can see NatNet data being received by your machine, I cannot think of any other obvious check apart from double checking the ROS configuration file (which was probably the first thing you did).

I'll think about it and post again if I have an intuition.

itzzartem, 2017-10-24:
An update on my issue. Heres the result when I run the hover_launch. If I use the multi-cast address It hangs at connecting with the following
"auto-starting new master
process[master]: started with pid [22474]
ROS_MASTER_URI=http://localhost:11311

setting /run_id to 62f6d4a2-b8fb-11e7-8b0b-d481d7cba180
process[rosout-1]: started with pid [22487]
started core service [/rosout]
process[crazyflie_server-2]: started with pid [22504]
process[crazyflie_manager-3]: started with pid [22505]
process[rviz-4]: started with pid [22512]
[ INFO] [1508877549.959016715]: Wait for services...
[ INFO] [1508877549.959805674]: waitForService: Service [/emergency] has not been advertised, waiting...
process[rqt-5]: started with pid [22522]
[ INFO] [1508877550.005646994]: waitForService: Service [/emergency] is now available.
[ INFO] [1508877550.018068985]: Manager ready.
NatNetClient connecting: 192.168.10.5 to 239.255.42.99...
WARNING: Could not set receive buffer size. Asked for 1048576B got 0B "

When I run using the multi cast ip address with the port, it seems to connect but not recieve's data. This seems to be an incorrect connection also.

setting /run_id to ce15cbee-b8fb-11e7-8b0b-d481d7cba180
process[rosout-1]: started with pid [24081]
started core service [/rosout]
process[crazyflie_server-2]: started with pid [24098]
process[crazyflie_manager-3]: started with pid [24099]
process[rviz-4]: started with pid [24100]
process[rqt-5]: started with pid [24106]
[ INFO] [1508877729.700672740]: Wait for services...
[ INFO] [1508877729.701562586]: waitForService: Service [/emergency] has not been advertised, waiting...
[ INFO] [1508877729.743405086]: waitForService: Service [/emergency] is now available.
[ INFO] [1508877729.755799244]: Manager ready.
NatNetClient connecting: 192.168.10.5 to 239.255.42.99:1511...
WARNING: Could not set receive buffer size. Asked for 1048576B got 0B
[Client] Server Software: Motive
[Client] NatNetVersion: 2.10
[Client] ServerVersion: 1.10
NatNet server version: 4294967295
ch: 1
[ INFO] [1508877729.826220521]: Adding CF: cf1 (radio://0/37/2M/E7E7E7E701, cf1)...
[ INFO] [1508877729.830808103]: CF ctor: 0.004449 s
[ INFO] [1508877729.839350750]: [cf1] Requesting parameters...
[ INFO] [1508877729.848166429]: Params: 86
[ INFO] [1508877730.119407305]: [cf1] reqParamTOC: 0.288509 s
[ INFO] [1508877730.119440834]: [cf1] Ready. Elapsed: 0.288566 s
[ INFO] [1508877730.119455630]: CF run: 0.288672 s
[ INFO] [1508877730.134232043]: [cf1] Update parameters
[ INFO] [1508877730.182955983]: Update params: 0.063458 s
[ INFO] [1508877730.183147434]: Started 1 threads
[ WARN] [1508877730.183259273]: No updated pose for motion capture object cf1
[ WARN] [1508877730.183334142]: VICON Latency high: 1.15595e+26 s.
  Latency: : 1.15595e+26 s.

[ WARN] [1508877730.183373354]: No updated pose for motion capture object cf1
[ WARN] [1508877730.183473795]: No updated pose for motion capture object cf1
[ WARN] [1508877730.183523960]: No updated pose for motion capture object cf1
[ WARN] [1508877730.183935176]: No updated pose for motion capture object cf1
[ WARN] [1508877730.183993039]: No updated pose for motion capture object cf1
[ WARN] [1508877730.184042779]: No updated pose for motion capture object cf1 "

From what I have worked for I think the one that hangs is the one that would be the correct connection. I'm running this through a direct LAN connection which I assume should work fine with multi-casting. Any other ideas would be helpful. thanks.

itzzartem, 2017-11-03:
I could not get it to work so I switched to the libobjecttracker and using the same ip setup get the correct info in rviz visualization. Closing this.

closed 2017-11-03 by itzzartem