USC-ACTLab/crazyswarm #31: crazyflie does not take off

hgchew, 2017-10-28:
I have followed the instructions to set up the crazyswarm with optitrack.  The firmwares have been updated and the cf1 appears in rvis, but it does not take off.  
[ INFO] [1509200982.106534044]: [cf1] Takeoff
[ INFO] [1509200984.693103497]: NextPhase!
[ INFO] [1509200984.704865818]: [cf1] Land
The above is logged with individual_hover, but the crazyflie remains static on the ground.  
Do you have any suggestions on how to debug this?

whoenig, 2017-10-29:
* Do I understand correctly that you roslaunch hover_swarm in one terminal, and individual_hover (the python script) in another one?
* Do you have an LED ring installed? If so, you should check the colors: Green: communication ok; blinking: low battery; red: no communication established
* Check if the CFs self-test passed. If the gyro couldn't calibrate, everything looks normal, but you won't be able to take-off. One way is to look at the red LED blinking frequency (I believe 1Hz is fine; failure if it is blinking at 2hz)

hgchew, 2017-10-29:
- yes. I did roslaunch in one and individual_hover on another
- unfortunately, I do not have the ring. 
- is gyro calibrated 2hz? That is the original spec unless it is different for crazyswarm firmware. 0.5hz is uncalibrated.

whoenig, 2017-10-30:
The crazyswarm firmware has the same LED blinking behavior as the original firmware.

If he LEDs indicate no issue and there is no error in the ROS log at startup, I can only imagine that the CF doesn't receive broadcasting commands. How did you flash the firmware? Try reflashing the NRF-firmware using `make cload`. Also, does the python GUI (choser.py) work, i.e. can you check the battery voltage and reboot? Those functions rely on the updated NRF firmware as well.

hgchew, 2017-10-30:
Yes, chooser works correctly. 
I had to use the cfclient to update the nrf using the zip method as make cload does not complete as it just loops the verify error near the end of the flash.

whoenig, 2017-10-31:
If you use the latest official SDK there shouldn't be that verify error (we plan to upgrade our submodule soon!). Since it is quick it is worth a shot, but if choser.py works, the likelyhood that everything was flashed correctly is high (there is no verification in either flashing method right now...)

One thing you should try is adding a logging variable to see if you receive the broadcast commands. A good variable might be "vicon.x" (see https://github.com/USC-ACTLab/crazyflie-firmware/blob/crazyswarm_master/src/modules/src/position_external.c#L200-L211). You can enable logging in hover_swarm.launch using the following steps:

1.) Add the logging variable(s) here: https://github.com/USC-ACTLab/crazyswarm/blob/master/ros_ws/src/crazyswarm/launch/hover_swarm.launch#L18
2.) Enable logging: https://github.com/USC-ACTLab/crazyswarm/blob/master/ros_ws/src/crazyswarm/launch/hover_swarm.launch#L87
3.) Update the plotting utility (or use rostopic echo to look at raw data): https://github.com/USC-ACTLab/crazyswarm/blob/master/ros_ws/src/crazyswarm/launch/hover_swarm.launch#L116

hgchew, 2017-11-03:
I did manage to plot the data, but I'm not sure what each variable refers to.
I think I am logging stabilizer.x, ctrltarget.x, vicon.x, and similar .z.  Am I correct to assume that ctrltarget is the target position, and vicon is the optitrack position?  What is stabilizer? However, physically moving the cf around does not change the vicon.x data.
The plot for ctrltarget.x and vicon.x remains at 0, while stabilizer.x seems to be increasing exponentially.

What do you think the problem is?

whoenig, 2017-11-03:
Stabilizer is the data after the kalman filter. If vicon.x remains at 0, that means that the CF is not receiving any of the information from optitrack. Thus, stabilizer drifts significantly (double integration of IMU data).

Since you are seeing that your CF is tracked in Rviz, there must be some issue with the communication. I would suggest that you try reflashing both firmwares using `make cload`. (I recently pushed a fix for the other error you were seeing by switching to the latest official SDK).

itzzartem, 2017-11-03:
I seem to be having this same issue. Chooser works fine,cf1 tracks correctly in rviz, when a scrip it run it calls for takeoff but crazyflie doesn't seem to be communicating. Flashed using make cload about a week ago.
Also if I try to run the takeoff service separately with hover_swarm running I get 
"ERROR: Unable to load type [crazyflie_driver/Takeoff].
Have you typed 'make' in [crazyflie_driver]"
Not sure if takeoff should work separately here. Might be my issue if it should.
Will try re-flashing today to see if it fixes my issue.

whoenig, 2017-11-03:
@itzzartem Do you have an LED ring to see the communication status? For your second error: you need to run `source ros_ws/devel/setup.bash` in the command prompt you are using. If you have an xbox360 controller, you can also press the Start button to take off and back button to land.

itzzartem, 2017-11-03:
Unfortunately I do not. When i source the terminal and try to run the service I get "ERROR: Service [/cf1/takeof] is not available."

whoenig, 2017-11-03:
Spelling mistake (takeoff with two f)? You can also at the list of currently advertised services.

itzzartem, 2017-11-03:
Yep it was a spelling mistake. What would the arguments be and their format?
Also i re-flashed the cf with the firmware and the result is the same. It seems to the connect to the CF at the launch of the server since the crazyflie makes the connection beep.

hgchew, 2017-11-04:
It seems the some update fixed it.  I did a pull yesterday, rebuild, reflash, and now the swarm works.  So I'm not sure what fixed it.  :)  At least it's now working

murali2kk, 2017-11-23:
I have been trying to run crazyswarm with one crazyflie, and have been unsuccessful. My issue seems to be similar to what has been described above:

* I can start all the nodes with hover_swarm.launch:
  * I use `libobjecttracker`. I can visualize the pointcloud along with the frame for cf1.
  * There are errors initially by the ICP, and stop after a couple of seconds of launching. I assume this is just because of all the nodes not yet being started, and can be safely ignored.
  * After connecting to the crazyflie, it starts with the usual good beeps. I do not have a LED ring, the status on the crazyflie seems okay:
    * The one near M1 is blinking all fine (red at 2Hz). 
    * Both the red and green LEDs near M4 blink continuously. This is similar when I try teleop, which works.
  * The log variables are improper clearly:
    * The `stabilizer.x` and `stabilizer.z`  keep increasing.
    * The `ctrltarget.x` and `ctrltarget.z`are nan.
    * The `vicon.x` and `vicon.z` are 0.
    * A point of note is that I use OptiTrack. Since I can visualize both the pointcloud and the cf frame in rviz, could there still be something yet amiss there?
* I try to start the `individual_hover`, and/or try to takeoff with the joystick. The command passes through and is reflected on the terminal. But nothing happens. There is no error/warning anywhere. The services `/cf1/*` are available.

From what is written by others, I suppose the pose data from the mocap is not getting through to the cf.

I have flashed the firmware. The `chooser.py` works.

Any suggestions to debug?

whoenig, 2017-11-26:
How did you flash the firmware? A couple of people successfully fixed the issue by flashing from the command line using `make cload`.

murali2kk, 2017-11-27:
Hi @whoenig : yes, I flashed both the crazyflie firmwares using `make cload`, and I updated the crazyradio firmware too from the crazyswarm repository. I assume the submodules are at the correct commits for it to work.

whoenig, 2017-11-27:
The submodules point to the correct commits (but the respective `crazyswarm_master` branches should work, too). When you move the Crazyflie around manually, do the `vicon.x/y/z` coordinates update correctly? Also, to avoid issues with ICP, I suggest creating an object named "cf1" (if your crazyflie has id1) in the OptiTrack software and change `object_tracking_type` to `motionCapture` (see https://github.com/USC-ACTLab/crazyswarm/blob/master/ros_ws/src/crazyswarm/launch/hover_swarm.launch#L55). This will use the tracking from OptiTrack instead of the custom ICP tracking.

murali2kk, 2017-11-30:
I am trying your suggestions.

1. I note now that the `vicon.x/y/z` are not updated when I move it manually. This is strange, because it tracks the initial position correctly across multiple runs (if I move it to a different starting location than what I mention in the `all49.yaml`,  it is picking up the correct pose information.) It just doesn't update when I move it. Also `/pointCloud` is updated continuously. I am not sure how to debug this from here. 

2. I need to look through using `motionCapture`. Something seems to be amiss there, this doesn't work at all. The `vicon.x/y/z` are always `0.0` when I do this. Since you do not have an OptiTrack, I shall dig deeper into this a bit, and let you know.

whoenig, 2017-11-30:
1. The point cloud is updating, but is the underlying object tracked correctly (if you use the custom objecttracker)? If that is true, there might be a communication issue, otherwise it is a tracking issue.
2. Lets discuss that in issue #32

murali2kk, 2017-12-01:
When I use the `libobjecttracker`, rivz shows me updated pose if I move it (both the point cloud and the cf1 frame). But this does no go to the cf, thus the `vicon.*` variables are constant.

I will look at this once again early next week and let you know where exactly in the code it gets lost.

itzzartem, 2017-12-01:
I forgot to post an update after I got it all working. Seems the issue was in the crazyradio and not the crazyflie. After rebuilding and re-flashing everything It still did not work. I remembered that I stated working with a new crazyradio and did not flash it's firmware. After flashing it the crazyflie now takes off. I am using the libobjecttracker and need to relax the constaints due to the body failing to converge at  times. Where can I adjust the constraints?

murali2kk, 2017-12-02:
@itzzartem : I did not understand that. So was your crazyradio faulty? Did you flash the firmware on the crazyradio on the new one? It would really help if you can elaborate the exact steps or things that you did which got it working for you? Thanks!

itzzartem, 2017-12-02:
@murali2kk I have multiple radios that I have been using with different projects. I switched out the radio that I was using with this project with another one that was running the newest firmware from bitcraze. That firmware was causing the connection issue. I flashed it with the firmware from the crazy swarm package and this fixed my issue.

murali2kk, 2017-12-07:
Hi @whoenig : I was able to dig a bit deeper. The pose is found correctly by `libobjecttracker`. The /tf is updated. I do not have any latency issues. In fact, I looked up the data up after it reaches [`CrazyflieBroadcaster::sendPositionExternalBringup`](https://github.com/USC-ACTLab/crazyflie_ros/blob/dcf82d762e549bc0fb4bc3a8de5f9fbe5ea46035/crazyflie_cpp/src/Crazyflie.cpp#L1335), and is indeed correct.

I believe there is some issue in just communicating this to the cf. My suspect is that I am using some part of the code updated/un-updated. 

I have flashed the firmwares of all the three from what versions I get when I clone the repository and running `./build.sh`:
1. crazyflie-firmware using `make cload`
2. crazyflie2-nrf-firmware using `make cload`
3. crazyradio-firmware using the commands:
`python ../usbtools/launchBootloader.py`  
`python ../usbtools/nrfbootload.py flash bin/cradio.bin`

Any further suggestions?

whoenig, 2017-12-08:
The CF uses two different messaging modes: regular packets and broadcasting. Since most of the management code seems to work for you, the regular packets seem to go through, while the broadcasting perhaps doesn't. Broadcasting is implemented in the custom radio firmware and nrf firmware. However, you said that the CF gets the initial vicon.x/y/z correctly? That would mean that at least some broadcasting commands go through. Are you using the Crazyradio or Crazyradio PA?

murali2kk, 2017-12-08:
I am using the Crazyradio PA. 

The initial `vicon.x/y/z` go through, but only sometimes. I do not know what exactly happens, but when I launch `hover_swarm.launch`, sometimes the initial pose is correct, otherwise it is`[0.0, 0.0, 0.0]`.

whoenig, 2017-12-09:
Do you reboot the CF between tries? Otherwise you might get "stale" data in the logging variable. Does the channel you use work fine otherwise (i.e. you don't see any significant packet loss)? I haven't seen that behavior myself. The LED ring or the debug adapter might be able to help find out when packets actually get transmitted and when not. Do you have a second CF or radio to try with?

murali2kk, 2017-12-09:
Rebooting does not ensure that the data in the logs are updated every time. I did not always reboot it, I shall do that from now. 

When I try working with the crazyflie_ros repo to hover a single cf, it would fine. I do not see any losses. So I assume the channel is fine.

I am afraid I do not have access to either the LED ring or the debug adapter. I do have an extra cf and radio. I shall give that a shot tomorrow and update.

murali2kk, 2017-12-11:
Hi @whoenig : Thank you for your support thus far. I am still unable to resolve the issue.

I now have access to an LED ring, and I was able to test with a new set of cf with it, and a new radio. The problem still persists. But the LEDs are all ref in color, which ideally implies that there is a connection issue. I tried many other channels over a range, did not help. 

It is surprising, because with the default firmware, the `hover_vrpn.launch` works perfectly fine for me. 

What else can you suggest me?

murali2kk, 2017-12-14:
Yes, I meant that with the default firmwares `hover_vrpn.launch` works fine, there are no communication issues.

When I use `crazyflie_tools scan`, the output is `Found Crazyradio with version 99.55`. The following was the output when I flashed the crazyradio firmware:
```
('Found nRF24LU1 bootloader version', '18.0')
Flashing:
  Flashing 7456 bytes...
Flashing done!
Verifying:
  Reading bin/cradio.bin...
  Reading 7456 bytes from the flash...
Verification succeded!

Note that the updated crazyradio firmware works fine with the old versions of crazyflie-firmware and crazyflie2-nrf-firmware (tested with crazyflie_demo of crazyflie_ros).

whoenig, 2017-12-14:
This version is correct and has broadcasting enabled (if you self compile 99.54 would be an older version). This radio is fully compatible with all other SW - it simply adds a new feature. Can you confirm that while you run `hover_swarm.py` that the green LED on the Crazyradio is constantly on? If that is the case than the transmitting side should be fine.

murali2kk, 2017-12-15:
I did self compile. When I run the `hover_swarm.launch`, the green LED seems to be on, but so is the red LED from what I see. I have the LED ring too now, and you can see it here:

[image]

whoenig, 2017-12-15:
There can be some red blinking (red = no ack received) initially, but not during broadcasting, because broadcasts do not except/wait for an acknowledgement. Thus, I think the issue is more on the sender, and not the receiver side.

I assume you are running Ubuntu 16.04? Which SDCC version did you use for compiling the radio firmware (I am using SDCC 3.5.0)?

Also, I just noticed that @itzzartem noted earlier that there is an issue with the latest official radio firmware (which we unfortunately seem to reference). Try rolling back to the latest version than we actually compiled and tested: https://github.com/bitcraze/crazyradio-firmware/commit/2192e59af64ea4928b132442a8b5053c9c2c26cc.

murali2kk, 2017-12-18:
I am using Ubuntu 16.04.3 LTS, and SDCC 3.5.0 to compile. I tested with the specific branch of crazyradio-firmware you suggested, and that did not help either. 

Are there pre-compiled versions of all the three firmwares anywhere that I can test with? May be there is something amiss when I compile?

whoenig, 2017-12-18:
I put (tested) binaries here: https://github.com/USC-ACTLab/crazyswarm/tree/master/prebuilt. You should be able to copy those to the appropriate places and use the flashing tools as before. I also added some documentation for using a **not well tested** flashing tool that is part of `crazyflie_ros` here: http://crazyswarm.readthedocs.io/en/latest/usage.html#crazyflie-preparation

murali2kk, 2017-12-19:
This does not help either. But there a small change -- when I try to upload trajectories, the cf crashes (it goes to the status where the [self-test fails](https://www.bitcraze.io/getting-started-with-the-crazyflie-2-0/#infostep10)). 

So, I reverted to my compiled version of crazyflie2-nrf-firmware, and I am back to square one.

Can I know what are the different latencies that get printed when I set the option of  `print_latency: True`? Because I see a couple of `0`s there. A sample output of that is here:
```
Latencies
: 19.0007 ms
Run All Groups: 0.202371 ms
Group 0 objectTracking: 0.102806 ms
Group 0 broadcasting: 0.05999 ms
Total 19.203 ms
Avg 2894
2.66529,0,0,0,0.506297,0.195055,0.101241,0,0,0,0,3.17158,
```

whoenig, 2017-12-19:
How did you flash the CF firmwares? I saw this issue yesterday sometimes when flashing using `crazyflie_ros`, but everything works as expected if I flashed using `make cload` (or if you wish Bitcraze GUI). Also, make sure you flash while the CF is close to the radio - the bitcraze tools do not have a verification step.

The latencies are supposed to show you how long each part in the pipeline takes. In your output it looks like you don't get from VICON a detailed step-by-step output, but only a total number (19 ms is reasonable). If you use VICON Tracker it can split the 19ms into 4 categories and tell you for each of them how long it took. The long line with numbers is just an average over all samples so far, and again, the numbers are low enough to not cause any issue.

One thing that is important though is that your framerate for VICON is set to 100Hz (or lower).

murali2kk, 2017-12-20:
I always flashed `make cload`. I did try to flash it with the GUI today for the crazyflie-firmware and it did not change anything really. I keep the cf about 30cm away from the radio (across the laptop). 

That is good to know that at least the tf and mocap side is not the issue for sure.

My OptiTrack was set to 120Hz. I assume you meant higher? Because if I reduce it to below 100Hz, I get a NatNet frame timeout error.

whoenig, 2017-12-20:
I did mean lower - it essentially needs to be slow enough such that we can broadcast everything before the next frame from the mocap arrives. Otherwise we'll be starting broadcasting old/stale information. However, if you try with a single CF, 120 Hz shouldn't be an issue either.

I was under the impression that the latency measurement was actually not fully implemented/working for the OptiTrack backend; thus, I am not sure how reliable the data is you got.

You mentioned earlier that 

>  when I try to upload trajectories, the cf crashes

What happens if you do not try to upload trajectories, i.e. does the LED ring light up in green after executing `hover_swarm.launch`?

murali2kk, 2017-12-21:
Then, probably using a higher rate with OptiTrack is okay. Because the NatNetLinux does not seem to really have a buffer, so it always reads the latest data, I think. Moreover, from the visualization in rviz, the motion of the cf by hand co-relates.

No, the LED ring is always red for me, except for a very brief interval at the beginning, when it is setting up I suppose. It never turns green. Nothing happens when I ask it takeoff (both by launching the `individual_hover.py` and/or clicking the joystick. I see the command popping up on the terminal once for each press.

whoenig, 2017-12-22:
The firmware has a safety check and will not fly if it doesn't receive mocap data. It does not receive the data, otherwise the LED ring would turn green (assuming you didn't change the following setting: https://github.com/USC-ACTLab/crazyswarm/blob/master/ros_ws/src/crazyswarm/launch/hover_swarm.launch#L41). 

Does your radio now show a continuous green light after you run `hover_swarm.launch`? If yes, the issue is likely on the NRF side (and you might need a debugger to see if packets arrive). If not, the issue could still be with your radio.

I assume you bought all the components (radio and CF) from a legitimate source (seeed etc.)? It is just strange because we seem to have many users that used the Crazyswarm successfully with both OptiTrack and Vicon.

murali2kk, 2017-12-22:
Okay. No, I haven't changed that part. 

It shows green for just about a second or less. 

Yes, they are bought through a legitimate source (through the university). Moreover crazyflie_ros works perfectly fine. I do see that people have got this working too. I shall see if I can get my hands on a debugger and check.

whoenig, 2017-12-24:
The radio shows green for only about a second? Then the problem is likely on the sender side and you should make sure (by adding printf's or some other logging) that the packets are attempted to be send in the crazyswarm_server (https://github.com/USC-ACTLab/crazyflie_ros/blob/edc45401577aea4b611421cdbdf3b612c17185d3/crazyflie_driver/src/crazyflie_server.cpp#L733).

murali2kk, 2018-01-03:
I did log the variables using print commands. You can see a sample of the log, and a video of the startup of the cf attached. Unless something is changes just before it gets broadcast, this looks okay to me, let me know if you think otherwise.

[image]

[image]

whoenig, 2018-01-03:
Thanks for the video!

1. It looks like you automatically start up the CF on connection. We usually use chooser.py to reboot a little before we start `hover_swarm.launch`. I am not sure if that might be a problem, but would be worth a shot.
2. Make sure that the LEDs on the CF show the expected behavior (see "Understanding LEDs" in https://www.bitcraze.io/getting-started-with-the-crazyflie-2-0/; "The blue LEDs (2 and 3) are fully lit and the front right LED (1) is blinking red twice every second." and "The front left LED (4) is flickering in red and/or green.")
3. How does the LED on the Crazyradio PA look like?

murali2kk, 2018-01-04:
1. Tried that too. Doesn't really help. Although as I had mentioned before, even the initial correct`/cf/log/vicon.*` happens only sometimes when the system is launched.

2. This is perfectly fine, showing the correct LED behavior.

3. After the launch, you can see the status of the LEDs below. 

With the logging to be true, there seems to be some data transfer with the red LED on the Crazyradio PA blinking, and the red and green LEDs (4) on the cf blinking too.
[image]

When I do not log there is no activity with those LEDs. 
[image]

Perhaps this means the data is not being broadcast at all?

whoenig, 2018-01-04:
The LED behavior is exactly as expected for the two scenarios for the radio. The red LED turns on if there was a packet drop (i.e. a packet was sent from the radio, but no ack received). In case of logging enabled, we have to send some packets the "traditional" way (i.e. with acks enabled). However, if you disable logging, only broadcasts are used (after connecting) and the radio LED will be solid green.

However, on the CF you should see the green LED on the front left side blinking in both cases (see https://github.com/USC-ACTLab/crazyflie-firmware/blob/crazyswarm_master/src/hal/src/radiolink.c#L143). Since you are not seeing it, the most likely case is that there is some problem with your NRF firmware. Unfortunately, that is difficult to debug - you would need the Bitcraze debug adapter kit and a JTAG debugger. We use a SEGGER JLINK EDU which allows to add printfs relatively easily. My best guess is that there was a problem during flashing, but I guess you tried reflashing numerous times already...

murali2kk, 2018-01-05:
I see. Then this is weird because others seem to have got this working. I have tried with different cfs and radios. My suspicion is that I am doing something wrong, just don't know what it is. 

Moreover why does the cf get the pose estimate initially, at least sometimes? I shall try to get my hands on a debugger if it is possible.

whoenig, 2018-01-05:
I am puzzled by this as well. One other random idea: Do you use USB 3.0 or USB 2.0 ports? I have seen strange issues with some devices on USB 3.0 (also depending on the exact computer model). Perhaps plug the radio into a USB 2.0 port, or use an older USB 2.0 hub in between your computer and the Crazyradio.

murali2kk, 2018-01-12:
I was using USB 3.0. But changing it did not work either. I have a couple of questions.

1. Is my procedure correct: 
    * I start the cf, reboot it now with the `chooser.py`,
    * launch the `hover_swarm.launch`, 
    * then try to take off from the joystick and/or,
    * start the `individual_hover.py`.
1. What are the fail safes in the crazyflie-firmware that do not let it take off?
1. When I press the take off, what messages should I expect on the rosout?
1. Does the pose of the cf change anything when I set it up? I have the initial position updated correctly in the yaml file. But does the orientation matter? Does aligning it with the world frame suffice?

Using the debugger will have to wait for me. I shall get to that only towards the end of this month.

Thanks!

whoenig, 2018-01-14:
1. Yes, the procedure is correct.
2. There should be only three (see https://github.com/USC-ACTLab/crazyflie-firmware/blob/crazyswarm_master/src/modules/src/stabilizer.c#L227-L230)
  a) Sensor data invalid (i.e. gyro didn't calibrate)
  b) CF is upside down (as measured by the accelerometer)
  c) No recent (that is, within the last 500ms) external tracking information received
3. Just a request for takeoff. This just tells you that it received the joystick/script command and sent a takeoff command to the CF. The CF might not receive the command or one of the fail-safes (see 2.) occurred.
4. If the CF was recognized correctly (as shown in rviz) you should be good. We typically place the CFs facing forward with the positive x-axis, but libobjecttracker does handle arbitrary orientations as well.

srnetsr, 2018-03-11:
Hi @murali2kk ,
you said 

> My OptiTrack was set to 120Hz. I assume you meant higher? Because if I reduce it to below 100Hz, I get a NatNet frame timeout error.

I am just curious how did you set Hz of OptiTrack?
Today, I got "NATNET FRAME TIMEOUT" error message.
It did not happen before.
So I am just curious whether it is caused by network unstable issue or not.
If you can advise me to set Hz of OptiTrack, then I would like to check it.

Thank you.

murali2kk, 2018-03-13:
Hi @srnetsr ,
Motive has an option to set the frame rate ([FPS](https://v20.wiki.optitrack.com/index.php?title=Devices_pane)). You can vary that to make sure that the NatNet frame timeout error does not come. I do not remember the exact fps I used, but yes varying fixed it. 

Just to note again, Motive 2.0 does not work with NatNetLinux, so use an older version.

Hope this helps!

srnetsr, 2018-03-13:
Thanks, @murali2kk 
sometime it happens, but cf2 flew well with that error before.
weird..
Anyhow, I appreciate it.

murali2kk, 2018-04-11:
Hi @srnetsr 

Can you tell me if you were able to resolve and fly them with OptiTrack? If so, what changes fixed the this issue to you?
 
Thanks!

whoenig, 2019-11-12:
This is a very old issue - closing. Note that Optitrack support has been improved significantly in the latest version.

closed 2019-11-12 by whoenig