USC-ACTLab/crazyswarm #32: connection problem

virtualclone, 2017-11-29:
Hi,I'm not sure if I have connected to the crazyflie
I do the same setting as [http://crazyswarm.readthedocs.io/en/latest/usage.html](url)
I update the firmware(newest)by _make cload_
and in **hover_swarm.launch**,
```
# tracking
      motion_capture_type: "optitrack" # one of vicon,optitrack
      object_tracking_type: "motionCapture" # one of motionCapture,libobjecttracker
      #vicon_host_name: "vicon"
      optitrack_local_ip: "192.168.1.203"
      optitrack_server_ip: "192.168.1.99"
```
and I can use **chooser.py** to receive battery and reboot
but when I roslaunch **hover_swarm.launch**,and use the joy to takeoff
It doesn't takeoff,it shows this
```
roslaunch crazyswarm hover_swarm.launch
... logging to /home/ncrl/.ros/log/96b3bad2-d4ab-11e7-8ca5-88d7f65412c4/roslaunch-ncrl-10390.log
Checking log directory for disk usage. This may take awhile.
Press Ctrl-C to interrupt
Done checking log file disk usage. Usage is <1GB.

started roslaunch server http://192.168.1.203:38281/

SUMMARY
========

PARAMETERS
 * /crazyflie_manager/csv_file: /home/ncrl/crazys...
 * /crazyflie_manager/timescale: 0.8
 * /crazyflie_server/broadcast_address: FFE7E7E7E7
 * /crazyflie_server/dynamicsConfigurations/0/maxFitnessScore: 0.001
 * /crazyflie_server/dynamicsConfigurations/0/maxPitch: 1.4
 * /crazyflie_server/dynamicsConfigurations/0/maxPitchRate: 20.0
 * /crazyflie_server/dynamicsConfigurations/0/maxRoll: 1.4
 * /crazyflie_server/dynamicsConfigurations/0/maxRollRate: 20.0
 * /crazyflie_server/dynamicsConfigurations/0/maxXVelocity: 2.0
 * /crazyflie_server/dynamicsConfigurations/0/maxYVelocity: 2.0
 * /crazyflie_server/dynamicsConfigurations/0/maxYawRate: 10.0
 * /crazyflie_server/dynamicsConfigurations/0/maxZVelocity: 3.0
 * /crazyflie_server/enable_logging: True
 * /crazyflie_server/enable_parameters: True
 * /crazyflie_server/firmwareParams/ctrlMel/i_range_m_z: 1500
 * /crazyflie_server/firmwareParams/ctrlMel/i_range_xy: 2.0
 * /crazyflie_server/firmwareParams/ctrlMel/i_range_z: 0.4
 * /crazyflie_server/firmwareParams/ctrlMel/kR_xy: 70000
 * /crazyflie_server/firmwareParams/ctrlMel/kR_z: 60000
 * /crazyflie_server/firmwareParams/ctrlMel/kd_omega_rp: 200
 * /crazyflie_server/firmwareParams/ctrlMel/kd_xy: 0.2
 * /crazyflie_server/firmwareParams/ctrlMel/kd_z: 0.4
 * /crazyflie_server/firmwareParams/ctrlMel/ki_m_z: 500
 * /crazyflie_server/firmwareParams/ctrlMel/ki_xy: 0.05
 * /crazyflie_server/firmwareParams/ctrlMel/ki_z: 0.05
 * /crazyflie_server/firmwareParams/ctrlMel/kp_xy: 0.4
 * /crazyflie_server/firmwareParams/ctrlMel/kp_z: 1.25
 * /crazyflie_server/firmwareParams/ctrlMel/kw_xy: 20000
 * /crazyflie_server/firmwareParams/ctrlMel/kw_z: 12000
 * /crazyflie_server/firmwareParams/ctrlMel/mass: 0.032
 * /crazyflie_server/firmwareParams/ctrlMel/massThrust: 132000
 * /crazyflie_server/firmwareParams/flightmode/posCtrl: 1
 * /crazyflie_server/firmwareParams/ring/effect: 14
 * /crazyflie_server/firmwareParams/ring/headlightEnable: 0
 * /crazyflie_server/firmwareParams/ring/solidBlue: 20
 * /crazyflie_server/firmwareParams/ring/solidGreen: 0
 * /crazyflie_server/firmwareParams/ring/solidRed: 0
 * /crazyflie_server/force_no_cache: False
 * /crazyflie_server/genericLogTopicFrequencies: [10]
 * /crazyflie_server/genericLogTopic_log1_Variables: ['stabilizer.yaw'...
 * /crazyflie_server/genericLogTopics: ['log1']
 * /crazyflie_server/interactive_object: active-wand
 * /crazyflie_server/markerConfigurations/0/numPoints: 4
 * /crazyflie_server/markerConfigurations/0/offset: [0.0, -0.01, -0.04]
 * /crazyflie_server/markerConfigurations/0/points/0: [0.0177184, 0.013...
 * /crazyflie_server/markerConfigurations/0/points/1: [-0.0262914, 0.05...
 * /crazyflie_server/markerConfigurations/0/points/2: [-0.0328889, -0.0...
 * /crazyflie_server/markerConfigurations/0/points/3: [0.0431307, -0.03...
 * /crazyflie_server/motion_capture_type: optitrack
 * /crazyflie_server/numDynamicsConfigurations: 1
 * /crazyflie_server/numMarkerConfigurations: 1
 * /crazyflie_server/object_tracking_type: motionCapture
 * /crazyflie_server/optitrack_local_ip: 192.168.1.203
 * /crazyflie_server/optitrack_server_ip: 192.168.1.99
 * /crazyflie_server/print_latency: False
 * /crazyflie_server/save_point_clouds: ~/pointCloud.ot
 * /crazyflie_server/world_frame: /world
 * /crazyflie_server/write_csvs: False
 * /crazyflies: []
 * /joy/dev: /dev/input/js0
 * /rosdistro: kinetic
 * /rosversion: 1.12.7

NODES
  /
    crazyflie_manager (crazyflie_manager/crazyflie_manager)
    crazyflie_server (crazyflie_driver/crazyflie_server)
    joy (joy/joy_node)
    rqt (rqt_gui/rqt_gui)
    rqt_plot_yaw (rqt_plot/rqt_plot)
    rviz (rviz/rviz)

ROS_MASTER_URI=http://192.168.1.203:11311

core service [/rosout] found
process[crazyflie_server-1]: started with pid [10408]
NatNetClient connecting: 192.168.1.203 to 192.168.1.99...
WARNING: Could not set receive buffer size. Asked for 1048576B got 0B
WARNING: Package "cfclient" does not follow the version conventions. It should not contain leading zeros (unless the number is 0).
process[joy-2]: started with pid [10475]
process[crazyflie_manager-3]: started with pid [10476]
process[rviz-4]: started with pid [10494]
[ INFO] [1511926598.658810492]: Wait for services...
process[rqt-5]: started with pid [10511]
[ INFO] [1511926598.661881451]: Manager ready.
process[rqt_plot_yaw-6]: started with pid [10518]
[ INFO] [1511926598.837264133]: Opened joystick: /dev/input/js0. deadzone_: 0.050000.
[ INFO] [1511926610.147639957]: Land!
[ INFO] [1511926610.955938389]: Land!
[ INFO] [1511926611.398637994]: Takeoff!
[ INFO] [1511926612.034207578]: Takeoff!
[ INFO] [1511926612.254008227]: Takeoff!
[ INFO] [1511926612.412697073]: Takeoff!
[ INFO] [1511926612.561124305]: Takeoff!
[ INFO] [1511926612.720287962]: Takeoff!
[ INFO] [1511926612.878865092]: Takeoff!
[ INFO] [1511926613.037298735]: Takeoff!
[ INFO] [1511926613.561745422]: Takeoff!
[ INFO] [1511926613.751122276]: Takeoff!
[ INFO] [1511926613.899476596]: Takeoff!
[ INFO] [1511926614.058046134]: Takeoff!
[ INFO] [1511926614.226837242]: Takeoff!
```

and I can see crazyflie in rviz,and I move it , it will also move in rviz(but I need to 

**roslaunch vrpn_client_ros sample.launch server:=192.168.1.99** in another terminal,is it normal?)

and I rostopic list,the topics are
```
/clicked_point
/diagnostics
/initialpose
/joy
/move_base_simple/goal
/pointCloud
/rosout
/rosout_agg
/tf
/tf_static
/virtual_interactive_object
/vrpn_client_node/cf1/pose

```
Are these right?

thanks for your replying!

whoenig, 2017-11-29:
* You should definitely not need to run `vrpn_client_ros`. The crazyflie server is connecting to your OptiTrack directly. Make sure that both local and server IP are set correctly.
* According to the log, the server does not even attempt to connect to your CF. How does `crazyflies.yaml` look like? [This could also be related to the OptiTrack connection issue though, because the server tries to connect to OptiTrack first, before establishing the connection to the CFs]

virtualclone, 2017-11-30:
- server IP is on the desktop with Optitrack(motive),and it's windows
  local IP is on the desktop with Ubuntu(another desktop),
  and I'm not sure that whether connection between them has problems or not,I'll check

- My crazyflie's channel is `//0/100/2M/E7E7E7E701`
 `crazyflies.yaml` is
```
 crazyflies:
- channel: 100
  id: 1
  initialPosition: [0.0, 0.0, 0.0]
```

thanks for your replying!

virtualclone, 2017-11-30:
Now,there is another problem,it keeps showing this warning

```
[ WARN] [1512016596.039300447]: No updated pose for motion capture object cf1
[ WARN] [1512016596.047483596]: No updated pose for motion capture object cf1
[ WARN] [1512016596.056344300]: No updated pose for motion capture object cf1
[ WARN] [1512016596.065058472]: No updated pose for motion capture object cf1

```
it seems that I have connected to the crazyflie but I can't get pose data from optitrack
but I can get pose data from optitrack by vrpn_client_ros

Is there wrong setting in motive(optitrack)?
because crazyswarm uses Natnet,different from vrpn
thanks for your reply!!!

murali2kk, 2017-11-30:
@virtualclone : did you try using the `libobjecttracker` instead of `motionCapture` (of course you will have to update the `/crazyflie_server/markerConfigurations/0/points/` according to your arrangement)? If so what happens then?

Note: Actually this and [issue31](https://github.com/USC-ACTLab/crazyswarm/issues/31) seem to be the same.

whoenig, 2017-11-30:
One problem with the optitrack support is that it uses a 3rd party library on Linux (the official one only supports Windows). This 3rd party library was initially developed a few years back, and might be not fully compatible with the latest motive version.

How do you guys use optitrack with ROS otherwise? I am aware of VRPN, but VRPN does not support raw marker data. We could develop a VRPN backend for external object tracking though.

@naus3a: Did you ever test using tracking by optitrack? If so, which motive version did you test it with? Are you aware of a newer/different SDK for Linux/Mac for OptiTrack?

virtualclone, 2017-12-01:
@murali2kk ,thanks for your reply,I try `libobjecttracker` , and update `markerConfigurations`
and it seems to have the same problem
```
ncrl@ncrl:~/crazyswarm$ roslaunch crazyswarm hover_swarm.launch
... logging to /home/ncrl/.ros/log/45a46b3c-d587-11e7-a8ba-88d7f65412c4/roslaunch-ncrl-28307.log
Checking log directory for disk usage. This may take awhile.
Press Ctrl-C to interrupt
Done checking log file disk usage. Usage is <1GB.

started roslaunch server http://192.168.1.203:45899/

SUMMARY
========

PARAMETERS
 * /crazyflie_manager/csv_file: /home/ncrl/crazys...
 * /crazyflie_manager/timescale: 0.8
 * /crazyflie_server/broadcast_address: FFE7E7E7E7
 * /crazyflie_server/dynamicsConfigurations/0/maxFitnessScore: 0.001
 * /crazyflie_server/dynamicsConfigurations/0/maxPitch: 1.4
 * /crazyflie_server/dynamicsConfigurations/0/maxPitchRate: 20.0
 * /crazyflie_server/dynamicsConfigurations/0/maxRoll: 1.4
 * /crazyflie_server/dynamicsConfigurations/0/maxRollRate: 20.0
 * /crazyflie_server/dynamicsConfigurations/0/maxXVelocity: 2.0
 * /crazyflie_server/dynamicsConfigurations/0/maxYVelocity: 2.0
 * /crazyflie_server/dynamicsConfigurations/0/maxYawRate: 10.0
 * /crazyflie_server/dynamicsConfigurations/0/maxZVelocity: 3.0
 * /crazyflie_server/enable_logging: True
 * /crazyflie_server/enable_parameters: True
 * /crazyflie_server/firmwareParams/ctrlMel/i_range_m_z: 1500
 * /crazyflie_server/firmwareParams/ctrlMel/i_range_xy: 2.0
 * /crazyflie_server/firmwareParams/ctrlMel/i_range_z: 0.4
 * /crazyflie_server/firmwareParams/ctrlMel/kR_xy: 70000
 * /crazyflie_server/firmwareParams/ctrlMel/kR_z: 60000
 * /crazyflie_server/firmwareParams/ctrlMel/kd_omega_rp: 200
 * /crazyflie_server/firmwareParams/ctrlMel/kd_xy: 0.2
 * /crazyflie_server/firmwareParams/ctrlMel/kd_z: 0.4
 * /crazyflie_server/firmwareParams/ctrlMel/ki_m_z: 500
 * /crazyflie_server/firmwareParams/ctrlMel/ki_xy: 0.05
 * /crazyflie_server/firmwareParams/ctrlMel/ki_z: 0.05
 * /crazyflie_server/firmwareParams/ctrlMel/kp_xy: 0.4
 * /crazyflie_server/firmwareParams/ctrlMel/kp_z: 1.25
 * /crazyflie_server/firmwareParams/ctrlMel/kw_xy: 20000
 * /crazyflie_server/firmwareParams/ctrlMel/kw_z: 12000
 * /crazyflie_server/firmwareParams/ctrlMel/mass: 0.032
 * /crazyflie_server/firmwareParams/ctrlMel/massThrust: 132000
 * /crazyflie_server/firmwareParams/flightmode/posCtrl: 1
 * /crazyflie_server/firmwareParams/ring/effect: 14
 * /crazyflie_server/firmwareParams/ring/headlightEnable: 0
 * /crazyflie_server/firmwareParams/ring/solidBlue: 20
 * /crazyflie_server/firmwareParams/ring/solidGreen: 0
 * /crazyflie_server/firmwareParams/ring/solidRed: 0
 * /crazyflie_server/force_no_cache: False
 * /crazyflie_server/genericLogTopicFrequencies: [10]
 * /crazyflie_server/genericLogTopic_log1_Variables: ['stabilizer.x', ...
 * /crazyflie_server/genericLogTopics: ['log1']
 * /crazyflie_server/interactive_object: active-wand
 * /crazyflie_server/markerConfigurations/0/numPoints: 3
 * /crazyflie_server/markerConfigurations/0/offset: [0.0, -0.01, -0.04]
 * /crazyflie_server/markerConfigurations/0/points/0: [0.011611, -0.069...
 * /crazyflie_server/markerConfigurations/0/points/1: [-0.007863, 0.052...
 * /crazyflie_server/markerConfigurations/0/points/2: [-0.066476, -0.00...
 * /crazyflie_server/motion_capture_type: optitrack
 * /crazyflie_server/numDynamicsConfigurations: 1
 * /crazyflie_server/numMarkerConfigurations: 1
 * /crazyflie_server/object_tracking_type: libobjecttracker
 * /crazyflie_server/optitrack_local_ip: 192.168.1.203
 * /crazyflie_server/optitrack_server_ip: 192.168.1.99
 * /crazyflie_server/print_latency: False
 * /crazyflie_server/save_point_clouds: ~/pointCloud.ot
[... 102 lines cut ...]
[ WARN] [1512101913.198991761]: No updated pose for motion capture object active-wand
[ WARN] [1512101913.199021502]: No updated pose for CF cf1 for 1512101913.198968 s.
[ INFO] [1512101913.814852323]: Takeoff!
[ WARN] [1512101914.200160054]: No updated pose for motion capture object active-wand
[ WARN] [1512101914.200194441]: No updated pose for CF cf1 for 1512101914.200107 s.
[ WARN] [1512101915.201370722]: No updated pose for motion capture object active-wand
[ WARN] [1512101915.201458285]: No updated pose for CF cf1 for 1512101915.201327 s.
[ WARN] [1512101916.202631035]: No updated pose for motion capture object active-wand
[ WARN] [1512101916.202685319]: No updated pose for CF cf1 for 1512101916.202609 s.
[ WARN] [1512101917.203952537]: No updated pose for motion capture object active-wand
[ WARN] [1512101917.204128590]: No updated pose for CF cf1 for 1512101917.203887 s.
[ WARN] [1512101918.205550876]: No updated pose for motion capture object active-wand
[ WARN] [1512101918.205634323]: No updated pose for CF cf1 for 1512101918.205476 s.
[ WARN] [1512101919.206932144]: No updated pose for motion capture object active-wand
[ WARN] [1512101919.207010358]: No updated pose for CF cf1 for 1512101919.206907 s.
[ WARN] [1512101920.208163608]: No updated pose for motion capture object active-wand
[ WARN] [1512101920.208228763]: No updated pose for CF cf1 for 1512101920.208137 s.
[ WARN] [1512101921.209174733]: No updated pose for motion capture object active-wand
[ WARN] [1512101921.209204599]: No updated pose for CF cf1 for 1512101921.209142 s.
[ WARN] [1512101922.210400282]: No updated pose for motion capture object active-wand
[ WARN] [1512101922.210432971]: No updated pose for CF cf1 for 1512101922.210344 s.
[ WARN] [1512101923.211484110]: No updated pose for motion capture object active-wand
[ WARN] [1512101923.211538123]: No updated pose for CF cf1 for 1512101923.211441 s.

```

whoenig, 2017-12-01:
@virtualclone You can get rid of the active-wand warning by putting a `#` in front of https://github.com/USC-ACTLab/crazyswarm/blob/master/ros_ws/src/crazyswarm/launch/hover_swarm.launch#L60.

In rviz, do you see the point cloud of your markers? As you mentioned, it looks like you are connecting just fine, but there is something wrong with the tracking. If you see the point cloud, at least the communication with Motive would be working.

virtualclone, 2017-12-01:
@whoenig , I don't see point cloud in rviz 
[image]
[image]
and I set like this,
I use three markers on my crazyflie and set `markerConfigurations`
```
numMarkerConfigurations: 1
      markerConfigurations:
        "0":
          numPoints: 3
          offset: [0.0, -0.01, -0.04]
          points:
            "0": [0.026096,-0.074674,0.004834]
            "1": [0.000946,0.045349,0.006044]
            "2": [-0.054784,-0.015370,0.005743]
```

naus3a, 2017-12-01:
@whoenig unfortunately, as far as I know, the official Otitrack tools supports Windows only, so we had to look for a Linux alternative.
NatNetLinux was the most promising project we could find, so we made our fork and basically always used that.

murali2kk, 2017-12-01:
I was just testing the `optitrack` class in the `libmotioncapture`  package independently. There seems to be some communication issue there when I try to read rigidbody pose data. The number of rigid bodies are 0. 

@naus3a : did you ever test it with the rigidbody estimate of the cf from NatNetLinux? Or did you always use the `libobjecttracker`?

But with `libobjecttracker` I do not see such a problem, the point cloud is being updated continuously. 

Note: The updated Motive 2.0 now uses NatNet 3.0, and the bit stream syntax is changed. So NatNetLinux throws up a segfault there. Yet, we are all using Motive 1.10, and NatNet 2.1 on the server side, so that should technically not be an issue.

I shall dig deeper early next week to find more.

virtualclone, 2017-12-01:
@murali2kk , can you use `libobjecttracker` to control the crazyflie?
because I set like what I said on the above reply.
but as you can see,I cannot see my crazyflie in rviz.
Is there anything wrong with my setting?
thanks!

naus3a, 2017-12-01:
@murali2kk we used libojecttracker most of the time, but I remember we did a few NatNetLinux only experiments when we first investigated the library.

murali2kk, 2017-12-02:
@virtualclone : Your settings on the crazyswarm front must be correct if you are seeing this part when you launch:

```
NatNetClient connecting: 192.168.1.203 to 192.168.1.99...
WARNING: Could not set receive buffer size. Asked for 1048576B got 0B
[Client] Server Software: Motive
[Client] NatNetVersion: 2.10
[Client] ServerVersion: 1.10
NatNet server version: 1661053120
```

On the OptiTrack front, check for your data stream settings on Motive -- I set the labelled and unlabelled markers to be True, with the Rigid bodies off, and using Multicast option. I would suggest fiddling around there to see if you can visualize the pointCloud on rviz. (I can tell you what exactly I used after I return to work next week).

Note: I have not been able to control my cf either, yet. I am at a stage where I can visualize the point cloud and the cf on rviz.

jpreiss, 2017-12-18:
@murali2kk @virtualclone are you still having problems using OptiTrack?

murali2kk, 2017-12-19:
The `motionCapture` for OptiTrack cannot work. The function [`getObjects`](https://github.com/USC-ACTLab/libmotioncapture/blob/5ea83a8969e981c170976904f942952f686a4b0b/src/optitrack.cpp#L38) is incomplete. I tried to fix it, but seems that NatNetLinux used here does not get the name of the rigidbody but the id manually set on Motive. So, one needs to map the id to the cf-name to fix this. I would do it at a later point, if I get crazyswarm working.

I got the `libobjecttracker` working though. A couple of notes for anyone who uses OptiTrack, and could not get this working:
  - This is incompatible with the latest `Motive (2.0)` because it uses `NatNet 3.0.1` which has a different bit-steam syntax. 
  - Use an older version (Motive 1.10.3 works for me). 
  - Have the rigid body *un-ticked* on Motive so that the point cloud is streamed. 
  - Update to the latest version of `libmotioncapture` branch of the [NatNetLinux](https://github.com/naus3a/NatNetLinux/commit/083601a8ed6ea1276e849a58497d60d91b18ae86).

whoenig, 2018-01-05:
I updated NatNetLinux to the latest version of libmotioncapture and added the other instructions to the documentation (http://crazyswarm.readthedocs.io/en/latest/usage.html#optitrack).

closed 2018-01-05 by whoenig

srnetsr, 2018-03-06:
@murali2kk 
Hi, I have one question about "Have the rigid body un-ticked on Motive so that the point cloud is streamed."

Do you mean that it needs to set "Stream Rigid Bodies" in Streamin > OptiTrack Streaming Engine > Streaming Setting as "False"?

murali2kk, 2018-03-13:
@srnetsr 

I used the following which worked, at least with two cfs. I created the rigid bodies, but unchecked them in the Assets pane. I was still streaming everything though. 

I am not sure if this is equivalent to making the stream setting to false, because then, Motive might not stream the markers associated with this rigid body.