USC-ACTLab/crazyswarm #33: crazyflie not connecting

The-SS, 2017-12-07:
Hi
I've been facing some problems getting the crazyswarm package running. In short, the crazyflies are not connecting and the topic `/pointCloud` does not have any data.
I installed the package based on [this link](http://crazyswarm.readthedocs.io/en/latest/installation.html), and I placed the vicon_sdk files in the correct path: `/crazyswarm/ros_ws/src/crazyflie_ros/externalDependencies/libmotioncapture/externalDependencies`.
I'm using vicon, which runs on a server computer with hostname: viconmotiontrac. So, I `hover_swarm.launch` and `/crazyflie_ros/crazyflie_manager/launch/hover_vicon.launch` and replaced the the hostname (vicon) by viconmotiontrac.
My Crazyflie.yaml file has the following data:
```
 crazyflies:
- channel: 80
  id: 1
  initialPosition: [0.0, 0.0, 0.0]
```
which is what I had set up.
When I run `roslaunch crazyswarm hover_swarm.launch`, I get the following:
```
SUMMARY
========

PARAMETERS
 * /crazyflie_manager/csv_file: /home/hivemind/cs...
 * /crazyflie_manager/timescale: 0.8
 * /crazyflie_server/broadcast_address: FFE7E7E7E7
 * /crazyflie_server/dynamicsConfigurations/0/maxFitnessScore: 0.001
 * /crazyflie_server/dynamicsConfigurations/0/maxPitch: 1.4
 * /crazyflie_server/dynamicsConfigurations/0/maxPitchRate: 20.0
 * /crazyflie_server/dynamicsConfigurations/0/maxRoll: 1.4
 * /crazyflie_server/dynamicsConfigurations/0/maxRollRate: 20.0
 * /crazyflie_server/dynamicsConfigurations/0/maxXVelocity: 2.0
 * /crazyflie_server/dynamicsConfigurations/0/maxYVelocity: 2.0
 * /crazyflie_server/dynamicsConfigurations/0/maxYawRate: 10.0
 * /crazyflie_server/dynamicsConfigurations/0/maxZVelocity: 3.0
 * /crazyflie_server/enable_logging: False
 * /crazyflie_server/enable_parameters: True
 * /crazyflie_server/firmwareParams/ctrlMel/i_range_m_z: 1500
 * /crazyflie_server/firmwareParams/ctrlMel/i_range_xy: 2.0
 * /crazyflie_server/firmwareParams/ctrlMel/i_range_z: 0.4
 * /crazyflie_server/firmwareParams/ctrlMel/kR_xy: 70000
 * /crazyflie_server/firmwareParams/ctrlMel/kR_z: 60000
 * /crazyflie_server/firmwareParams/ctrlMel/kd_omega_rp: 200
 * /crazyflie_server/firmwareParams/ctrlMel/kd_xy: 0.2
 * /crazyflie_server/firmwareParams/ctrlMel/kd_z: 0.4
 * /crazyflie_server/firmwareParams/ctrlMel/ki_m_z: 500
 * /crazyflie_server/firmwareParams/ctrlMel/ki_xy: 0.05
 * /crazyflie_server/firmwareParams/ctrlMel/ki_z: 0.05
 * /crazyflie_server/firmwareParams/ctrlMel/kp_xy: 0.4
 * /crazyflie_server/firmwareParams/ctrlMel/kp_z: 1.25
 * /crazyflie_server/firmwareParams/ctrlMel/kw_xy: 20000
 * /crazyflie_server/firmwareParams/ctrlMel/kw_z: 12000
 * /crazyflie_server/firmwareParams/ctrlMel/mass: 0.032
 * /crazyflie_server/firmwareParams/ctrlMel/massThrust: 132000
 * /crazyflie_server/firmwareParams/flightmode/posCtrl: 1
 * /crazyflie_server/firmwareParams/ring/effect: 14
 * /crazyflie_server/firmwareParams/ring/headlightEnable: 0
 * /crazyflie_server/firmwareParams/ring/solidBlue: 20
 * /crazyflie_server/firmwareParams/ring/solidGreen: 0
 * /crazyflie_server/firmwareParams/ring/solidRed: 0
 * /crazyflie_server/force_no_cache: False
 * /crazyflie_server/genericLogTopicFrequencies: [10]
 * /crazyflie_server/genericLogTopic_log1_Variables: ['stabilizer.yaw'...
 * /crazyflie_server/genericLogTopics: ['log1']
 * /crazyflie_server/interactive_object: active-wand
 * /crazyflie_server/markerConfigurations/0/numPoints: 4
 * /crazyflie_server/markerConfigurations/0/offset: [0.0, -0.01, -0.04]
 * /crazyflie_server/markerConfigurations/0/points/0: [0.0177184, 0.013...
 * /crazyflie_server/markerConfigurations/0/points/1: [-0.0262914, 0.05...
 * /crazyflie_server/markerConfigurations/0/points/2: [-0.0328889, -0.0...
 * /crazyflie_server/markerConfigurations/0/points/3: [0.0431307, -0.03...
 * /crazyflie_server/motion_capture_type: vicon
 * /crazyflie_server/numDynamicsConfigurations: 1
 * /crazyflie_server/numMarkerConfigurations: 1
 * /crazyflie_server/object_tracking_type: libobjecttracker
 * /crazyflie_server/print_latency: False
 * /crazyflie_server/save_point_clouds: ~/pointCloud.ot
 * /crazyflie_server/vicon_host_name: viconmotiontrac
 * /crazyflie_server/world_frame: /world
 * /crazyflie_server/write_csvs: False
 * /crazyflies: [{'initialPositio...
 * /joy/dev: /dev/input/js0
 * /rosdistro: kinetic
 * /rosversion: 1.12.7

NODES
  /
    crazyflie_manager (crazyflie_manager/crazyflie_manager)
    crazyflie_server (crazyflie_driver/crazyflie_server)
    joy (joy/joy_node)
    rqt (rqt_gui/rqt_gui)
    rqt_plot_yaw (rqt_plot/rqt_plot)
    rviz (rviz/rviz)

ROS_MASTER_URI=http://localhost:11311

core service [/rosout] found
process[crazyflie_server-1]: started with pid [5552]
process[joy-2]: started with pid [5576]
process[crazyflie_manager-3]: started with pid [5579]
process[rviz-4]: started with pid [5591]
process[rqt-5]: started with pid [5607]
process[rqt_plot_yaw-6]: started with pid [5612]
[ INFO] [1512674701.052109800]: Wait for services...
[ INFO] [1512674701.061766397]: Manager ready.
[ INFO] [1512674701.085324756]: Opened joystick: /dev/input/js0. deadzone_: 0.050000.

```

The crazyflie does not start. And when I run `rostopic echo /pointCloud` there is nothing there.

Am I missing something?

Thank you.

whoenig, 2017-12-08:
* Do other packages (e.g. `vicon_bridge`) work correctly with your setup (to exclude network issues)
* Are the markers visible in VICON Tracker? Specifically, we only use unlabeled markers, so make sure that you don't have any object tracking in Tracker enabled.

The-SS, 2017-12-08:
`vicon_bridge` works with my setup. But, if I'm not mistaken, `crazyswarm` does not use`vicon_bridge`. Just to make sure, is there anywhere in particular that the IP address of the VICON server computer must be used?

Regarding the markers and VICON, I'm using VICON Nexus. The markers are visible. I create segments corresponding to the crazyflies, but I do not label the markers.

whoenig, 2017-12-09:
I never used Nexus, so I am not sure what the behavior is (and what you mean by "segments"). 
Perhaps try the following changes:
1.) https://github.com/USC-ACTLab/libmotioncapture/blob/master/src/vicon.cpp#L33: Also call `pImpl->client.EnableMarkerData();`
2.) https://github.com/USC-ACTLab/libmotioncapture/blob/master/src/vicon.cpp#L112-L120
Copy this segment directly below and use `GetMarkerCount()` and `GetMarkerGlobalTranslation()` instead of the `Unlabeled` calls.

This will take both labeled and unlabeled markers into account, in case VICON Nexus has a different interpretation of unlabeled markers (in Tracker markers that are not associated with objects/segments are considered unlabeled).

The-SS, 2017-12-15:
Actually, I found out that the problem was that a connection was not being established between the computer and the VICON server computer. I was using the hostname of the server, but a connection was not being established. When I replaced the hostname by the IP address of the VICON server, a connection was established, the crazyflies were added, and I saw the crazyflies on the RVIZ graph.

Note: Using `GetMarkerCount()` and `GetMarkerGlobalTranslation()` instead of the Unlabeled calls gave build errors because they are not defined. However,  `GetLabeledMarkerCount()` and `GetLabeledMarkerGlobalTranslation()` are defined. Nonetheless, I did not have to change them.

closed 2017-12-15 by The-SS