USC-ACTLab/crazyswarm #34: Firmware for BigQuad

aldrinchua8, 2017-12-20:
Good Day, I will be using a bigquad expansion deck for my project. I would just like to ask if I can enable the BigQuad expansion deck in the NRF51 firmware because I would like to use the crazyswarm repository for my project. Thanks!

whoenig, 2017-12-20:
The BigQuad Deck is supported and we tested it a few month back. You won't need to adjust the NRF51 firmware, but recompile the STM32 firmware with `-DENABLE_BQ_DECK` (following the [wiki](https://wiki.bitcraze.io/projects:crazyflie2:expansionboards:bigquad)). However, you will need to tune the controller for your frame; while doing so, make sure that you proceed with caution and use safety equipment.

closed 2017-12-20 by whoenig

aldrinchua8, 2017-12-21:
Thanks for the quick reply! We are actually on the part on testing the crazyswarm and finding it hard to stabilize a single crazyflie without the BigQuad. I would just like to ask on how to tune the controller?