USC-ACTLab/crazyswarm #35: Tuning the Crazyflie

aldrinchua8, 2017-12-21:
Good Day. I am new to the crazyflie equipment and hopefully that someone will be able to help. I would just like to know on how the tune the PID of the crazyswarm? I am currently working on a single drone using this tutorial http://crazyswarm.readthedocs.io/en/latest/ the drone would not stabilize. The advise that we got was to tune the quad but I do not know how. if you could help us or link us to a tutorial, it would mean a lot. Thanks! :)

whoenig, 2017-12-22:
What modification/payload does your CF have? The controller is pre-tuned for 33g, that is a payload of LED extension and 4 mocap markers. Unless you made a lot of HW modifications, retuning shouldn't be necessary. What kind of stability issues do you see?

All tuning values are here: https://github.com/USC-ACTLab/crazyswarm/blob/master/ros_ws/src/crazyswarm/launch/hover_swarm.launch#L23-L39. Start by measuring the mass and finding the right mass thrust by turning off the i-gain for z (ki_z=0.0). You should also enable logging to compare desired with the true state. You can lower the position gains first to tune the attitude controller.

aldrinchua8, 2017-12-23:
Hi, Thanks for the quick reply as always :)
We haven't done any modifications yet because we are still testing the crazyswarm repository. Whenever we would try to launch the crazyflie, it would go up for about an inch then it would suddenly accelerate and fly in random directions. 

We are using optitrack and getting its coordinate via Natnet SDK in windows. We just followed the guide available in the github. Also, there are no modified parts for the crazyflie yet.

I think the problem lies somewhere along the retrieving of the coordinates from Natnet. Are there any ways to verify that what we are sending to the hover_swarm.launch is in fact the coordinates that we are getting in Natnet SDK from optitrack in windows?

whoenig, 2017-12-23:
Sounds like your coordinate system is somehow switched and not an controller tuning issue. I think there might be a setting in motive for the coordinate system to use. In order to check, launch `hover_swarm.launch`, but don't take off. Move the CF around manually and see if what you do matches the visualization in rviz.

aldrinchua8, 2017-12-24:
Hi, we just tested it and the visualization follows the rviz. 

There is just something weird happening. Whenever we would initialize the hover_swarm.launch then place the cf anywhere within the optitrack area, the visualization in rviz would always place the quad in the origin. Should this be happening?

whoenig, 2017-12-26:
Your initial position should roughly match what you specified in `crazyflies.yaml`, so try avoid putting it anywhere. Does rviz show the markers (as spheres) and did you update the marker configuration according to your setup?

aldrinchua8, 2017-12-28:
Hi, fixed the RVIZ visualization and RVIZ follows the visualization in optitrack.

The cf is still unstable. Should the front of the crazyflie face in any particular direction/axis? or can i face it anywhere? also, I just want to clarify the "marker configuration according to your setup".

whoenig, 2017-12-28:
The CF should be facing the direction of the positive x-axis. If you move it left, it should go in the direction of the positive y-axis, and if you move it up it should go in the direction of the positive z-axis (that is the ROS coordinate system). Make sure that this is the case, because OptiTrack might use a different coordinate system (as far as I know depending on a setting).

With "marker configuration according to your setup", I meant step 3 in http://crazyswarm.readthedocs.io/en/latest/usage.html#configure-marker-arrangement.

aldrinchua8, 2017-12-28:
Hi, we were able to do the instructions. We were also able to change the axis of the optirrack such that the rviz would follow its visualization. The strange thing is that although we changed the axis in the optitrack the values being received is still the same (i.e. the z value would still follow the forward and backward motion, although we already changed that to the upward and downward motion). In RVIZ however, the z follows the upward and downward motion. I would just like to ask on how should we change the values in the crazyflies.yaml? And also, I would just like to verify if the values for crazyflies.yaml and the coordinates of marker in hover_swarm.launch is arranged as (x,y,z) format? 

Thanks! :)

whoenig, 2017-12-28:
1. There is an axis flip in libmotioncapture (see https://github.com/USC-ACTLab/libmotioncapture/blob/master/src/optitrack.cpp#L24-L25), that might explain some of the discrepancies you are seeing?
2. Adjust the position in all49.yaml, restart chooser.py, and disable/enable your CF in order to update crazyflies.yaml. You should only need to update the position in [x,y,z] format, as seen by ROS/RVIZ.
3. The coordinates are in [x,y,z] format, as seen by ROS/RVIZ (you should be able to see the values when you look at the pointcloud topic).

aldrinchua8, 2018-01-03:
Hi there, we tried adjusting everything and it seems that some improvements were seen. The RVIZ below shows the axis of the cf. 
[image]
I would just like to ask, if what we are seeing is the expected output in RVIZ? The cf would still not hover when we launch the individual_hover.py file. I am wondering why the axis of the cf is not aligned with the markers being shown in RVIZ.

Also, I observed that the cf is always going backward (with respect to the front position of the cf) at whatever axis is is placed. Most of the times, it would flip backwards.

Thanks

whoenig, 2018-01-03:
The "cf1" frame should be on top of the markers. Also "flip backwards" still sounds like your coordinate system is not correct.

How does your marker configuration look like? (see https://github.com/USC-ACTLab/crazyswarm/blob/master/ros_ws/src/crazyswarm/launch/hover_swarm.launch#L64-L70). And perhaps take a picture on how you placed the markers on the CF alongside?

aldrinchua8, 2018-01-05:
Hi there, 
The photo below shows the coordinates that we get from motive.
[image]

we just write the coordinates that we see in the pointcloud in the hover_swarm.launch. I would just like to ask on where do we get the numbers to write in all49.yaml and crazyflies.yaml? the values that we use here are the ones that natnet sdk from the windows computer. 

thanks!

itzzartem, 2018-01-05:
Perhaps I can help you with this since I currently have it working.
For the coordinate transform as stated by Whoenig
"There is an axis flip in libmotioncapture (see https://github.com/USC-ACTLab/libmotioncapture/blob/master/src/optitrack.cpp#L24-L25), that might explain some of the discrepancies you are seeing?"
you need to change it to:
pImpl->axisMultiplier = Point3f(1, 1, 1);//0.001,0.001,0.001);
pImpl->axisOrder = Point3f(2,0,1);
if you have the up axis set to "Y up: in motive. (default setting)
When specifying the marker locations in the config file you need to use the coordinates in Rviz and not motive. i.e. the coordinated in motive are given as (x,y,z), you need to put them in as (z,x,y) if you use the coordinate transform stated above.

aldrinchua8, 2018-01-16:
Hi there,
I have been able to run the indivudual_hover.py successfully; however this only happened once. After the successful run, the quadcopter became unstable again. I think that it has something to do with the tuning?? because the quadcopter tend to go to the direction of the two blue LED lights. Please help. Thanks!

whoenig, 2018-01-16:
You shouldn't need tuning unless you use a custom frame/different batteries/much higher payload. Perhaps your propellers are damaged and need replacement or balancing. You can exclude hardware issues by using the official firmware and flying manually using a joystick.

aldrinchua8, 2018-01-19:
Hi, here is a GIF of what is happening
[image]
[image]
From what I observe, once the cf gets to its specified height, it starts to wobble and be unstsable. What do you think is wrong? Thanks!

whoenig, 2018-01-19:
What is the metal structure in the middle? Make sure that it doesn't cause occlusion from your mocap system. Do you see the same behavior without that structure?

aldrinchua8, 2018-01-20:
The structure does not cause occlusion. I placed it there so that i can secure the cf when flying. Without the metal stand, the cf flies to a certain height then would sway all around the place until it goes out of reach of the vision of the optitrack. I attached strings on the cf, then attached it to the structure so that it when the cf goes out of control, it would not be damaged. 

Yes, I do see the same behavior without the structure. The only difference, is that nothing is stopping the cf from going out of control once it reaches the specified height. Thanks!

whoenig, 2018-01-22:
In my experience, strings often have some weird effect on the controller since they might exert additional forces if pulled. However, given that you have problems without strings/structure as well, I think you experience something unrelated.

How does rviz look like during those flights? In particular, is the tracking fine during takeoff? You can do a screen capture or rosrecord the tf and pointcloud to be able to look at it in detail. 

One method we used at the beginning is using the mocap software to track the object (which is typically more robust in terms of occlusions or intermediate tracking issues). Since you are using OptiTrack, this is not fully supported yet. You would need to implement the getObjects function (see https://github.com/USC-ACTLab/libmotioncapture/blob/master/src/optitrack.cpp#L38-L53). This way you will know if it is more likely a tracking issue or a control/HW issue.

emmasmith100, 2019-01-09:
I want to retune the crazyflie mellinger and PID controllers since I attached a frame with some considerable mass and medium sized markers to it. I had to do this, since it didn't stay stable under the default controller settings (as I expected).

I changed the mass to the new value in the .yaml file. Now it is stable, but only if the duration in the python scripts is set large enough (5 or higher). It does not work if the position reference is instantaneously applied (small duration).

When retuning both the PID and mellinger controller (using repectively kalman and kalman USC as estimators), I saw some things that are not intuitive to me. I tested some simple experiments where the crazyflie only takes-off, moves in the x direction, comes back to initial xy position and finally lands in its initial position.
- even if ki_z is set to 0 (also if it is the default value), reducing kp_z by a factor of 10 or more does not make the uav slowly go to its setpoint height. Same if i increase the damping kd_z by 10. I would expect the uav to go very slowly up, but it all doesn't change that much. There are some oscillations, but it always goes quite quickly up. Seems to be the same in the x direction.
- because the system is a double integrator, I don't see why we have to add the integrator in the controller. Shouldn't it automatically have a zero steady state error using only a PD control? Is there any reason why it is better to add the integrator?
What do you think about those points?

Some other questions related to the controller and tuning
-Also what is the use of the parameter massThrust? When does this case gets false? setpoint->mode.z == modeDisable in controller_mellinger.c. what are the different modes? where to set them?
- in controller_mellinger.c you have calculated the 3 control moments M.x, M.y, M.z and the thrust. But where can I find the code to relate these to the actual inputs to the 4 motors?
-In the above discussion you told "you can exclude hardware issues by using the official firmware and flying manually using a joystick." How can I fly manually with a joystick in the crazyswarm? So should i put firmware: "crazyswarm" to firmware: "bitcraze"? What else should I do to test this? I assume then the xbox is giving a velocity instead of a position setpoint? Is it also possible to make the xbox give the position setpoint?

Thanks a lot for helping me out!

emmasmith100, 2019-01-10:
Thanks you for the valuable comments!
Which type of trajectory planning is used in the `takeoff` and `GoTo` commands? Do you have any documentation (with some equations) on this? Can you show me the code where I can find the trajectory planning part? Since in all files names `GoTo` I don't see that particular planning part.
Is the trajectory planning calculated onboard the crazyflie?

Is there a way to bypass the trajectory planning calculations using the python API?
I would first of all just like to tune the onboard PID/mellinger controller without having a trajectory planner on top. I just want to send via the python script a position (x,y,z) and yaw reference to the PID/mellinger controller. I mean applying the reference instantaneously as a step input to the controller.
Thanks you so much for your quick answers!

whoenig, 2019-01-10:
The trajectory planning is done onboard, see https://github.com/bitcraze/crazyflie-firmware/blob/master/src/modules/src/planner.c#L127-L147 and https://github.com/bitcraze/crazyflie-firmware/blob/master/src/modules/src/pptraj.c. I don't think there is any documentation apart from the code, but the equations are fairly easy to identify in the files.

You can do the trajectory evaluation in Python and send the desired state over the radio (up to 4 CFs per radio, only). You can find an example here: https://github.com/USC-ACTLab/crazyswarm/blob/master/ros_ws/src/crazyswarm/scripts/cmdFullState.py.

jpreiss, 2019-01-10:
Keep in mind that our controller is tuned for precisely tracking a smooth trajectory, under the assumption that deviation from the trajectory will never be too large. It is a high gain controller. A very big step input in position might cause instability. You can lower the position gains or maybe limit the magnitude of `target_thrust` to handle step changes in the setpoint. (If you get the thrust limiting working nicely, please submit a pull request. The magnitude limit should be a radio-controllable parameter like the controller gains.)

emmasmith100, 2019-01-11:
@whoenig , I don't fully understand why sending the full desired state is limited up to 4 CFs per radio. It doesn't look like this is using much bandwidth for radio communication, or am I wrong?  I think you mean with the full desired state in the code `cmdFullState.py`: `pos, vel, acc, yaw, omega`?
Maybe I was not clear enough, but I just want to send the desired position reference and desired yaw angle (so no unnecessary information as vel, acc, omega) at for example a rate of 100Hz. So is there a function/code to just send this information: desired x,y,z and yaw (while off-course still sending the actual pos, vel used as state feedback to the crazyflies)? Otherwise to have a position ref+yaw I would have to set vel, acc, omega to zero, but this would still use more bandwidth than necessary.

So, what is the reason you achieved to obtain a larger number of crazyflies per radio (I read somewhere a ratio up to 15) if you use the onboard trajectory planning. In your case (using onboard planning) you should also send position and yaw reference, and a duration parameter. Or is this data sent only once when executing the code and not continuously sent at a rate of 100Hz, making the difference in bandwidth? So how can you reduce the communication bandwidth by a factor of 4 (15 vs 4)?

@jpreiss , yes indeed, a very big step input in position + yaw can cause instability. I would like to see where the limits of the PID / mellinger controllers are by tuning the gains manually. In this way I can get a better understanding of the controlled system.

In the end (after understanding the behavior and how to tune the existing PID and mellinger controller for a x,y,z position+yaw setpoint, I would like to add another controller and maybe a planner in the firmware. So that this 3th controller would be called from the `hover_swarm.launch` file like this, `controller: 3 # 1: PID, 2: mellinger, 3: the new controller`.
Therefore I think I have to change these files in the firmware folder:
- add a file with my controller code in (which is similar to `controller_mellinger.c` in crazyswarm/crazyflie-firmware/src/modules)
- add some lines to controller.c and controller.h
Is there anything else I have to do to add my own controller? Maybe something in the ros_ws?
How does the value (1,2 or 3) assigned to controller in `hover_swarm.launch` affect the firmware? Where can I say the new controller should be corresponding to value 3?

Is there a place in `hover_swarm.launch` where I can set the type of planner when I would add a new planner? Is it under `commander: enHighLevel: 1` or is this something else?

whoenig, 2019-01-13:
You are correct that the high-level planner generates setpoints onboard at a high rate (250 Hz), which only taking commands such as take-off, goTo, or executeTrajectory (with the trajectory being pre-uploaded and evaluated on-board). Sending setpoints at a high-frequency limits the bandwidth drastically (CFs radio is optimized for low latency, not high bandwidth). The firmware and crazyflie_ros have functions that can only send positions to save bandwidth, but the crazyswarm_server does not have this support currently (it would be easy to implement and contributions are welcome). For now, I suggest just setting the components you don't need to 0 - it sounds like your controller experiments can be done even with a single CF.

To add a controller, you add a new Type at the end in the controllerTypes enum (https://github.com/bitcraze/crazyflie-firmware/blob/master/src/modules/interface/controller.h#L31-L36), and update the code in controller.c. There is no change required on the host-computer side.

There is currently no firmware support for multiple planners. You can see how the current high-level planner is enabled here: https://github.com/bitcraze/crazyflie-firmware/blob/master/src/modules/src/commander.c#L86-L88.

emmasmith100, 2019-01-15:
Thanks for the answers!
As you suggested I adapted cmdFullState.py and set everything I don't need (like velocity, acc) to 0.
I just give a fixed desired position (x,y,z) and desired yaw for some duration.
Two weird things can happen (I don't always get the same problems depending on the values I put) when I run the code attached below (and indicated with comments where it occurs): 
1) the crazyflie suddenly seems to fall down (same effect as if I would push for emergency) 
or 
2) would drift for some time after execution of the executeSetpoint function.

Sometimes this sudden fall down happens after I press the xbox button for the first time, sometimes it does not, while I didn't change anything of software/hardware configuration. 
I don't get any occlusion warnings. The step references I use are very small, and it can even happen when you give a desired position step equal to your current position the cf can fall down.
Also in between 2 different GoTo commands I was previously able to use warm.input.waitUntilButtonPressed(). Doing the same between GoTo and the function executeSetpoint below seems to give these problems.

Any idea what part of the code is causing this weird behavior and how I can get rid of this problem?
I would like to give the step reference by executeSetpoint() from the moment I press the xbox button.

Thank you for the help!!

code:
import numpy as np
import rospy

from pycrazyswarm import *
import uav_trajectory

def executeSetpoint(reverse = False, rate = 100, setpointPos=np.array([0,0,0]), setpointYaw = 0.0, duration=10): # I made this function
    rate = rospy.Rate(rate)

    start_time = rospy.Time.now()
    while not rospy.is_shutdown():
        t = (rospy.Time.now() - start_time).to_sec()
        print(t)
        if t > duration:
            break

        #if reverse:
        #    print(t)
        #else:
        #    print(t)
        for cf in allcfs.crazyflies:
            cf.cmdFullState(
                setpointPos,
                np.array([0.0,0.0,0.0]),       # don't send velocity
                np.array([0.0,0.,0.0]),       # don't send acceleration
                setpointYaw,         # later you can specify also a yaw angle
                np.array([0.0,0.0,0.0]))     # don't send rot speed
       
        rate.sleep()

if __name__ == "__main__":
    swarm = Crazyswarm()
    timeHelper = swarm.timeHelper
    allcfs = swarm.allcfs

    rate = 100 #50
    takeoffheight=0.5
    takeoffpos=np.array([0.0,-0.5,takeoffheight])
    takeoffyaw=0.0
    desiredPos = np.array([0.0,-0.6,0.7])
    desiredYaw = 0.0 # yaw in radians
    
    for cf in allcfs.crazyflies:
	allcfs.takeoff(takeoffheight, 5.0)

whoenig, 2019-01-15:
Does `cmdFullState.py` work for you as expected?

In your code, you don't seem to use your `executeSetpoint` function?

The behavior you describe might indicate that there is a communication issue (e.g., either mocap feedback or setpoints are not received at a high enough rate on-board).

jpreiss, 2020-06-13:
@emmasmith100 were you able to resolve this?

jpreiss, 2021-01-30:
closing due to inactivity.

closed 2021-01-30 by jpreiss