USC-ACTLab/crazyswarm #36: Tracking Problem

The-SS, 2017-12-25:
Hi everyone. I'm trying to get one crazyflie to hover, but I'm having problems with tracking.
I am using VICON to track the crazyflies. I've tried both `object_tracking_type` options, but I get problems with both.

When using ` libobjecttracker` as `object_tracking_type`, I get the following problem:
```
SUMMARY
========

PARAMETERS
 * /crazyflie_manager/csv_file: /home/hivemind/cs...
 * /crazyflie_manager/timescale: 0.8
 * /crazyflie_server/broadcast_address: FFE7E7E7E7
 * /crazyflie_server/dynamicsConfigurations/0/maxFitnessScore: 0.001
 * /crazyflie_server/dynamicsConfigurations/0/maxPitch: 1.4
 * /crazyflie_server/dynamicsConfigurations/0/maxPitchRate: 20.0
 * /crazyflie_server/dynamicsConfigurations/0/maxRoll: 1.4
 * /crazyflie_server/dynamicsConfigurations/0/maxRollRate: 20.0
 * /crazyflie_server/dynamicsConfigurations/0/maxXVelocity: 2.0
 * /crazyflie_server/dynamicsConfigurations/0/maxYVelocity: 2.0
 * /crazyflie_server/dynamicsConfigurations/0/maxYawRate: 10.0
 * /crazyflie_server/dynamicsConfigurations/0/maxZVelocity: 3.0
 * /crazyflie_server/enable_logging: False
 * /crazyflie_server/enable_parameters: True
 * /crazyflie_server/firmwareParams/ctrlMel/i_range_m_z: 1500
 * /crazyflie_server/firmwareParams/ctrlMel/i_range_xy: 2.0
 * /crazyflie_server/firmwareParams/ctrlMel/i_range_z: 0.4
 * /crazyflie_server/firmwareParams/ctrlMel/kR_xy: 70000
 * /crazyflie_server/firmwareParams/ctrlMel/kR_z: 60000
 * /crazyflie_server/firmwareParams/ctrlMel/kd_omega_rp: 200
 * /crazyflie_server/firmwareParams/ctrlMel/kd_xy: 0.2
 * /crazyflie_server/firmwareParams/ctrlMel/kd_z: 0.4
 * /crazyflie_server/firmwareParams/ctrlMel/ki_m_z: 500
 * /crazyflie_server/firmwareParams/ctrlMel/ki_xy: 0.05
 * /crazyflie_server/firmwareParams/ctrlMel/ki_z: 0.05
 * /crazyflie_server/firmwareParams/ctrlMel/kp_xy: 0.4
 * /crazyflie_server/firmwareParams/ctrlMel/kp_z: 1.25
 * /crazyflie_server/firmwareParams/ctrlMel/kw_xy: 20000
 * /crazyflie_server/firmwareParams/ctrlMel/kw_z: 12000
 * /crazyflie_server/firmwareParams/ctrlMel/mass: 0.032
 * /crazyflie_server/firmwareParams/ctrlMel/massThrust: 132000
 * /crazyflie_server/firmwareParams/flightmode/posCtrl: 1
 * /crazyflie_server/firmwareParams/ring/effect: 14
 * /crazyflie_server/firmwareParams/ring/headlightEnable: 0
 * /crazyflie_server/firmwareParams/ring/solidBlue: 20
 * /crazyflie_server/firmwareParams/ring/solidGreen: 0
 * /crazyflie_server/firmwareParams/ring/solidRed: 0
 * /crazyflie_server/force_no_cache: False
 * /crazyflie_server/genericLogTopicFrequencies: [10]
 * /crazyflie_server/genericLogTopic_log1_Variables: ['stabilizer.yaw'...
 * /crazyflie_server/genericLogTopics: ['log1']
 * /crazyflie_server/markerConfigurations/0/numPoints: 4
 * /crazyflie_server/markerConfigurations/0/offset: [0.0, -0.01, -0.04]
 * /crazyflie_server/markerConfigurations/0/points/0: [-1.64561450481, ...
 * /crazyflie_server/markerConfigurations/0/points/1: [-1.41054797173, ...
 * /crazyflie_server/markerConfigurations/0/points/2: [-1.41226184368, ...
 * /crazyflie_server/markerConfigurations/0/points/3: [-1.37784779072, ...
 * /crazyflie_server/motion_capture_type: vicon
 * /crazyflie_server/numDynamicsConfigurations: 1
 * /crazyflie_server/numMarkerConfigurations: 1
 * /crazyflie_server/object_tracking_type: libobjecttracker
 * /crazyflie_server/print_latency: False
 * /crazyflie_server/save_point_clouds: ~/pointCloud.ot
 * /crazyflie_server/vicon_host_name: 10.0.0.3:801
 * /crazyflie_server/world_frame: /world
 * /crazyflie_server/write_csvs: False
 * /crazyflies: [{'initialPositio...
 * /joy/dev: /dev/input/js0
 * /rosdistro: kinetic
 * /rosversion: 1.12.7

[... 73 lines cut ...]
[ WARN] [1514243758.363781096]: Object tracker initialization failed - check that position is correct, all markers are visible, and marker configuration matches config file
[ WARN] [1514243758.363915686]: No updated pose for CF cf1 for 1514243758.362515 s.
[ WARN] [1514243758.394175456]: Initialize did not succeed (fitness too low).
[ WARN] [1514243758.394261109]: Object tracker initialization failed - check that position is correct, all markers are visible, and marker configuration matches config file
[ WARN] [1514243758.394298951]: No updated pose for CF cf1 for 1514243758.392438 s.
[ WARN] [1514243758.395672323]: Initialize did not succeed (fitness too low).
[ WARN] [1514243758.395740321]: Object tracker initialization failed - check that position is correct, all markers are visible, and marker configuration matches config file
[ WARN] [1514243758.395777351]: No updated pose for CF cf1 for 1514243758.394408 s.
[ WARN] [1514243758.423618981]: Initialize did not succeed (fitness too low).
[ WARN] [1514243758.423845751]: Object tracker initialization failed - check that position is correct, all markers are visible, and marker configuration matches config file
```
The warning goes on forever.
In RVIZ, I can see the markers reproduced as seen by VICON.  I got the marker coordinates from `pointCloud` and I used them as the `markerConfigurations` in `hover_swarm.launch`. I also modified the initial location of the crazyflie in `crazyflies.yaml` per the marker coordinates (the average of the markers). However, the coordinates system axis attached to the crazyflie does not appear in VICON.

So, based on the output when running  `roslaunch crazyswarm hover_swarm.launch` there are two problems observed:
1- The error with finding parameters `Could not find param ctrlMel/i_range_m_z` ...
2- The problem with the object tracker initialization failing.

What is causing these problems and what should I do about them?

Thank you.

Note:
The markers are asymmetrical.
I read issue [#10](https://github.com/USC-ACTLab/crazyswarm/issues/10) and I updated the cf firmware, yet I still get the error with finding parameters.

whoenig, 2018-01-05:
1. That is related to the firmware not being our custom firmware. You need to flash the STM32 firmware that is part of the Crazyswarm distribution.
2. You can try using the motionCapture object tracking mode first, using VICON Tracker to track your object (name the object cf1). For your tracking issue it seems like you didn't put the CF at the origin when writing down the pointCloud coordinates. Specifically, all your 4 marker coordinates are larger than a meter, while your `markerConfigurations/0/offset` is close to 0.

The-SS, 2018-01-26:
When I flashed both STM and NRF, this time using `make cload` for both, the problem with the parameters was resolved and the crazyflie finally received commands and began to operate.
As for the second issue, it was due to the placement of the crazyflie away from the center, as you explained.

closed 2018-01-26 by The-SS