USC-ACTLab/crazyswarm #37: crazyswarm and vicon connection issue

lincoln1587, 2017-12-26:
HI, 
     I'm trying to get the crazyswarm to hover with the hover_swarm accordingly, but failed and I can't see any cf on rviz, I think it's a vicon connectin problem.

- I am using vicon system with both motion Capture or libobjecttracker
- I run the chooser.py fine
- I run the crazyflie_demo package fine 
- I do the configuration according to http://crazyswarm.readthedocs.io/en/latest/usage.html#basic-flight

The message are shown below

SUMMARY
========

PARAMETERS
 * /crazyflie_manager/csv_file: /home/fat-lab/cra...
 * /crazyflie_manager/timescale: 0.8
 * /crazyflie_server/broadcast_address: FFE7E7E7E7
 * /crazyflie_server/dynamicsConfigurations/0/maxFitnessScore: 0.001
 * /crazyflie_server/dynamicsConfigurations/0/maxPitch: 1.4
 * /crazyflie_server/dynamicsConfigurations/0/maxPitchRate: 20.0
 * /crazyflie_server/dynamicsConfigurations/0/maxRoll: 1.4
 * /crazyflie_server/dynamicsConfigurations/0/maxRollRate: 20.0
 * /crazyflie_server/dynamicsConfigurations/0/maxXVelocity: 2.0
 * /crazyflie_server/dynamicsConfigurations/0/maxYVelocity: 2.0
 * /crazyflie_server/dynamicsConfigurations/0/maxYawRate: 10.0
 * /crazyflie_server/dynamicsConfigurations/0/maxZVelocity: 3.0
 * /crazyflie_server/enable_logging: False
 * /crazyflie_server/enable_parameters: True
 * /crazyflie_server/firmwareParams/ctrlMel/i_range_m_z: 1500
 * /crazyflie_server/firmwareParams/ctrlMel/i_range_xy: 2.0
 * /crazyflie_server/firmwareParams/ctrlMel/i_range_z: 0.4
 * /crazyflie_server/firmwareParams/ctrlMel/kR_xy: 70000
 * /crazyflie_server/firmwareParams/ctrlMel/kR_z: 60000
 * /crazyflie_server/firmwareParams/ctrlMel/kd_omega_rp: 200
 * /crazyflie_server/firmwareParams/ctrlMel/kd_xy: 0.2
 * /crazyflie_server/firmwareParams/ctrlMel/kd_z: 0.4
 * /crazyflie_server/firmwareParams/ctrlMel/ki_m_z: 500
 * /crazyflie_server/firmwareParams/ctrlMel/ki_xy: 0.05
 * /crazyflie_server/firmwareParams/ctrlMel/ki_z: 0.05
 * /crazyflie_server/firmwareParams/ctrlMel/kp_xy: 0.4
 * /crazyflie_server/firmwareParams/ctrlMel/kp_z: 1.25
 * /crazyflie_server/firmwareParams/ctrlMel/kw_xy: 20000
 * /crazyflie_server/firmwareParams/ctrlMel/kw_z: 12000
 * /crazyflie_server/firmwareParams/ctrlMel/mass: 0.032
 * /crazyflie_server/firmwareParams/ctrlMel/massThrust: 132000
 * /crazyflie_server/firmwareParams/flightmode/posCtrl: 1
 * /crazyflie_server/firmwareParams/ring/effect: 14
 * /crazyflie_server/firmwareParams/ring/headlightEnable: 0
 * /crazyflie_server/firmwareParams/ring/solidBlue: 20
 * /crazyflie_server/firmwareParams/ring/solidGreen: 0
 * /crazyflie_server/firmwareParams/ring/solidRed: 0
 * /crazyflie_server/force_no_cache: False
 * /crazyflie_server/genericLogTopicFrequencies: [10]
 * /crazyflie_server/genericLogTopic_log1_Variables: ['stabilizer.yaw'...
 * /crazyflie_server/genericLogTopics: ['log1']
 * /crazyflie_server/markerConfigurations/0/numPoints: 4
 * /crazyflie_server/markerConfigurations/0/offset: [0.0, -0.0, -0.0]
 * /crazyflie_server/markerConfigurations/0/points/0: ['0.0431009 -0.04...
 * /crazyflie_server/markerConfigurations/0/points/1: ['-0.0222183 0.07...
 * /crazyflie_server/markerConfigurations/0/points/2: ['-0.0227949 -0.0...
 * /crazyflie_server/markerConfigurations/0/points/3: ['-0.0536936 -0.0...
 * /crazyflie_server/motion_capture_type: vicon
 * /crazyflie_server/numDynamicsConfigurations: 1
 * /crazyflie_server/numMarkerConfigurations: 1
 * /crazyflie_server/object_tracking_type: motionCapture
 * /crazyflie_server/print_latency: False
 * /crazyflie_server/save_point_clouds: ~/pointCloud.ot
 * /crazyflie_server/vicon_host_name: vicon
 * /crazyflie_server/world_frame: /world
 * /crazyflie_server/write_csvs: False
 * /crazyflies: [{'initialPositio...
 * /joy/dev: /dev/input/js0
 * /rosdistro: kinetic
 * /rosversion: 1.12.7

NODES
  /
    crazyflie_manager (crazyflie_manager/crazyflie_manager)
    crazyflie_server (crazyflie_driver/crazyflie_server)
    joy (joy/joy_node)
    rqt (rqt_gui/rqt_gui)
    rqt_plot_yaw (rqt_plot/rqt_plot)
    rviz (rviz/rviz)

ROS_MASTER_URI=http://localhost:11311

core service [/rosout] found
process[crazyflie_server-1]: started with pid [15454]
process[joy-2]: started with pid [15455]
process[crazyflie_manager-3]: started with pid [15456]
process[rviz-4]: started with pid [15470]
process[rqt-5]: started with pid [15484]
[ INFO] [1514259259.186278966]: Wait for services...
[ INFO] [1514259259.187908400]: waitForService: Service [/emergency] has not been advertised, waiting...
process[rqt_plot_yaw-6]: started with pid [15500]
[ INFO] [1514259259.208758812]: waitForService: Service [/emergency] is now available.
[ INFO] [1514259259.213837566]: Manager ready.
[ INFO] [1514259259.362692335]: Opened joystick: /dev/input/js0. deadzone_: 0.050000.

I am wondering did I miss any modification or doing anything incorrect.

Thanks

Yang

whoenig, 2017-12-26:
This does indeed sound like a connection issue. You can try if you can connect your vicon system using `vicon_bridge`. Also, make sure that your hostname (`vicon`) is correct and try using the IP address instead in case there is an issue with the name resolution.

lincoln1587, 2017-12-26:
Thank you so much, it works after I change the host name to the ip addr.

closed 2017-12-26 by lincoln1587