USC-ACTLab/crazyswarm #38: Adding crazyflies

itzzartem, 2018-01-05:
I seem to be running into a stability issue when running multiple crazyflies. If I use the chooser and fly a figure 8 pattern with each crazyflie (currently a swarm of 9) I get a good hover as well as a smooth figure 8. When I attempt this with two its not as smooth but still executes fairly well. The problems begin when I use 3 or more. When I attempt it with three, I can sometimes get the to hover but after a few seconds one of them flies away in some random direction. I am running this on optitrack using the libmotioncapture with the same marker configuration on each crazyflie. The crazyflies are all on the 2Mhz band and on channel 37 (seems to be the best channel here). Any ideas where my problem is arising from? Also for reference I seemed to have the same issue when flying more than 2 on the crazyflie_ros package.

whoenig, 2018-01-06:
This could be either an issue with the communication (high packet drop rate) or an issue with tracking.

1. How did you determine channel 37 to be good? A nice tool is https://github.com/bitcraze/crazyflie-lib-python/blob/master/examples/radio-test.py.
2. Do you have the LED ring installed? It is color-coded and tells you if the communication is good (green) or bad (red).
3. Are all markers tracked correctly when you fly 3 or more? You could record the pointcloud topic and tf during the execution during a crash. A replay of those topics should allow you to see if the tracking is good at least before/at the beginning of the crash.

The original crazyflie_ros only works well with up to 2 CFs per radio, because the controller is running on the PC (rather the CF), and it doesn't use broadcasting commands.

itzzartem, 2018-01-06:
I determined the best channel using the tool that was specified. Unfortunately I do not have the led ring. I am adding 6 more cameras today to improve tracking and will try recording the data afterwards to see if the problem lays in tracking or communication. Also there sometimes seems to be a slight problem in the ICP from time to time where it calculates the position slightly incorrectly, or on original startup the orientation is calculated incorrectly. I am using 5 markers that are not placed symmetrically. Also I am using the same marker configuration for all the crazyflies but it seems the markers on a few of them are slightly off (not noticeably)(currently not using these for the current testing) and the ICP struggles to converge. What would be the best approach to this problem? Should I relax the ICP or define marker configurations for each one in the launch file?

whoenig, 2018-01-06:
How do you know that the position is calculated incorrectly? On startup, we try several possible orientations (https://github.com/USC-ACTLab/libobjecttracker/blob/master/src/object_tracker.cpp#L186) and take the best one. You can try increasing that number and see if that improves your tracking. We only use four markers and have no such problems.

It is expected that you can't place the markers perfectly identical. There is currently no full support for different marker configurations per CF (we are working on that though). You can try adjusting the fitness score (which is the sum of the error squared between measured marker position and expected marker position based on the current transformation): https://github.com/USC-ACTLab/crazyswarm/blob/master/ros_ws/src/crazyswarm/launch/hover_swarm.launch#L82. Naturally, this number needs to be higher if you use 5 instead of 4 markers.

You could also try the marker configuration that works well for us. You can find a picture showing the markers in J. A. Preiss*, W. Hönig*, G. S. Sukhatme, and N. Ayanian. "Crazyswarm: A Large Nano-Quadcopter Swarm", in Proc. IEEE International Conference on Robotics and Automation, Singapore, 2017, Figure 1 (preprint available at http://act.usc.edu/publications.html). We use 7.9mm spherical markers, although that depends on your motion capture system.

itzzartem, 2018-02-18:
What would be the best way to change the channel on a crazyflie on the crazyswarm firmware?

whoenig, 2018-02-18:
Connect your CF over USB and change it using the Bitcraze Python Client.

itzzartem, 2018-03-28:
So just an update, I got multiple crazy flies working now. After fixing the tracking issue there seem to be crazyflie specific problems. Not sure if its due to propellers coming loose or motors, but I was able to find 4 that were able to execute the figure eight with my marker configuration, proving that there is no communication or tracking issue left. Closing this now.

closed 2018-03-28 by itzzartem

srnetsr, 2018-04-10:
Hi, @whoenig , @itzzartem 

I am trying to use radio-test.py
However, I got the error messages as shown in the following.

```
~/crazyflie-lib-python$ sudo python examples/radio-test.py 
Traceback (most recent call last):
  File "examples/radio-test.py", line 19, in <module>
    import cflib.drivers.crazyradio as crazyradio
ImportError: No module named cflib.drivers.crazyradio
```

Could you teach me how to use radio-test.py ?

Thank you in advance

whoenig, 2018-04-10:
You have to install the CF python lib on your system:

```
$ cd crazyflie-lib-python
$ pip3 install --user -e .
```

srnetsr, 2018-04-10:
It works!
I appreciate it.

srnetsr, 2018-04-12:
Hi @whoenig ,
do you have any idea about it?

Thank you in advance..

srnetsr, 2018-04-17:
@itzzartem , thank you for your reply.
But it did not work. I got the same error, and I found [https://github.com/bitcraze/crazyflie-lib-python/pull/21](https://github.com/bitcraze/crazyflie-lib-python/pull/21)
@whoenig , I have a question.
I followed [http://crazyswarm.readthedocs.io/en/latest/usage.html](http://crazyswarm.readthedocs.io/en/latest/usage.html), and flash nrf & stm using your instruction.
The question is that what is the different between [https://github.com/bitcraze/crazyflie2-nrf-firmware](https://github.com/bitcraze/crazyflie2-nrf-firmware) and `cf2_nrf.bin` file in `crazyswarm/prebuilt` ?

Thank you.

whoenig, 2018-04-17:
There is no difference - the NRF51 firmware is identical to the latest official one. The prebuilt binary is only there for simplified reproducibility.

srnetsr, 2018-04-17:
Thanks for confirmation.
Based on [bitcraze/crazyflie-lib-python#21](bitcraze/crazyflie-lib-python#21) and information you gave, I tested all cases, and finally what is the problem.
The problem is related with the address of cf2.
I set the address of cf2 as `0xE7E7E7E701`.
This address does not work!
I switched it to `0xE7E7E7E7E7`, and it worked.

Thank you guys!