USC-ACTLab/crazyswarm #39: Getting error when running the chooser.py

aldrinchua8, 2018-01-13:
Hi!
I am getting an error when running the chooser.py. The error can be seen in the screenshot.
[image]
basically, I had this working with my laptop, but now I am using my desktop and can't seem to fixed this.

I changed the channel of the cf with cfclient using the crazyradio. I then changed the channel in the all49.yaml and the crazyflies.yaml. The crazyradio can connect just fine in the cfclient, but the chooser.py shows the erros. Please Help. Thanks!

-Aldrin

jpreiss, 2018-01-13:
You just need to install the pyyaml package. https://github.com/yaml/pyyaml

aldrinchua8, 2018-01-14:
Hi, 
I have been able to get past the yaml problem, but a new error occurs.
[image]
I have the cf connected to the cfclient via the crazyradio. then I would run the chooser.py and the error would occur.