USC-ACTLab/crazyswarm #40: Cannot flash radio

aldrinchua8, 2018-01-14:
Hi, 
I an getting an error when I try to flash the radio using the "option 2" method. The error is shown below
[image]
I do have the crazyradio upgraded to 99.55 but I am not sure if I am using the crazyradio PA or not. because in the "README" file, it seems that the you can choose if it is crazyradio PA or not. how can i check if I have the crazyradio PA?

whoenig, 2018-01-14:
1. You need to run the command as sudo or add a udev-file (see "Flying Multiple UAVs Using ROS", Chapter in Robot Operating System (ROS): The Complete Reference (Volume 2), Springer, 2017. ; pre-print at http://act.usc.edu/publications.html).

2. There should be some text on your radio indicating "Crazyradio" or "Crazyradio PA".

aldrinchua8, 2018-01-14:
Thanks! adding sudo made it work!

closed 2018-01-14 by aldrinchua8