USC-ACTLab/crazyswarm #41: crazyflie_demo hover_vicon.launch  doesn't work

lincoln1587, 2018-01-16:
I run the launch flie "crazyflie_demo hover_vicon.launch", it doesn't work. And i used the firmware in the prebuild folder. Could you help me? thanks

Traceback (most recent call last):
  File "/home/crazyswarm/ros_ws/src/crazyflie_ros/crazyflie_demo/scripts/controller.py", line 99, in <module>
    controller = Controller(use_controller, joy_topic, csv_file)
  File "/home/crazyswarm/ros_ws/src/crazyflie_ros/crazyflie_demo/scripts/controller.py", line 14, in __init__
    rospy.wait_for_service('update_params')
  File "/opt/ros/kinetic/lib/python2.7/dist-packages/rospy/impl/tcpros_service.py", line 159, in wait_for_service
    raise ROSInterruptException("rospy shutdown")
rospy.exceptions.ROSInterruptException: rospy shutdown

whoenig, 2018-01-16:
This is not a supported launch file. The crazyswarm only works with `hover_swarm.launch`.

lincoln1587, 2018-01-16:
hello, whoenig
So which project or firmware works for the hover_vicon.launch? Have you opened the source? Thanks

whoenig, 2018-01-16:
The original crazyflie_ros supports `hover_vicon.launch` (see https://github.com/whoenig/crazyflie_ros).

closed 2021-01-30 by jpreiss