USC-ACTLab/crazyswarm #42: Using the firmware of crazyradio of crazyswarm for basic python in windows

aldrinchua8, 2018-01-19:
Hi there, Some of my team is working on the swarm of crazyflies project in windows only and using python to program the drones. The problem that we encounter there is that when the crazyradio are beside each other, the cf does not respond. But if the crazyradio are atleast 5inches apart, it starts to work properly. The problem is that we can only connect two crazyflies in each radio. Since I have been trying to work with crazyswarm, they asked me if I could somehow extract the program/firmware to communicate with multiple drones. And I was wondering if this is possible, that the program that was used for crazyswarm connection to the crazyradio be transferred to windows Python only. Thanks!

whoenig, 2018-01-19:
It depends on what you want to control. The Crazyswarm can handle so many CFs per radio because we moved state estimation and control onboard, i.e. you will need (our) custom firmware. The communication itself now supports broadcasts which is easy to implement in python using the python binding to libusb. However, re-implementing the other control logic to control the swarm seems cumbersome (in particular you somehow need to track the CFs and send the pose updates).

aldrinchua8, 2018-01-20:
Noted! Will show this to them! Thanks!

closed 2018-01-20 by aldrinchua8