USC-ACTLab/crazyswarm #43: Problem hovering multiple crazyflies

The-SS, 2018-01-31:
Hi,

I'm having trouble flying multiple crazyflies (cfs) at once.

Using `hover_swarm.launch` I successfully got crazyflie 1 (cf1) and crazyflie7 (cf7) to hover independently. Both have approximately the same marker placement. I would place each at [0,0,0] and run `hover_swarm.launch` and each would hover.

Then, to get multiple cfs hovering, I made the following changes to `crazyflie.yaml`:
```
crazyflies:
- channel: 80
  id: 1
  initialPosition: [0.0, 0.5, 0.0]
- channel: 86
  id: 7
  initialPosition: [0.0, 0.0, 0.0]
```
I ran `hover_swarm.laucnh` but I got the following:
```
[ INFO] [1517418080.991244623]: Wait for services...
[ INFO] [1517418080.999312318]: waitForService: Service [/takeoff] has not been advertised, waiting...
[ INFO] [1517418081.018830845]: Opened joystick: /dev/input/js0. deadzone_: 0.050000.
[ INFO] [1517418081.020966378]: waitForService: Service [/takeoff] is now available.
[ INFO] [1517418081.033966664]: Manager ready.
crazyflies.size = 2
crazyflie id and channel are: 1 80
crazyflie id and channel are: 7 86
[ INFO] [1517418081.034869807]: All CFs are ready!
running runFast
Connecting to VICON... 
Connected to VICON! 
[ INFO] [1517418081.079185555]: motion capture type defined
ch: 2
radio: 1
radio: 0
libusb: warning [add_to_flying_list] failed to arm first timerfd (errno 9)
```  
I tried uploading trajectories and pressing the start button, but nothing happened.

Am I doing something wrong?

Thank you.

note: I am using only one crazyradio.

whoenig, 2018-01-31:
By design you need one radio per channel, because each channel/radio runs in its own thread. So either plug in a second radio, or change the channel for both CFs to 80 (or 86).

Please confirm if that fixes your problem. If so, we should add some proper error checking for that case...

The-SS, 2018-01-31:
Yes, that fixed my problem. 
Thank you.

closed 2018-01-31 by The-SS

samandehghan, 2018-03-02:
Hi,
I could not hover 2 crazyflie by using hover_swarm.launch. I could see crazyflies position in RViz and there is no error. I used 
crazyflies:
- channel: 100
  id: 1
  initialPosition: [1.5, 1.5, 0.0]
- channel: 100
  id: 7
  initialPosition: [1.5, -1.5, 0.0]
in crazyyaml and I could see battery level in chooser.py but when I execute python figure8_canned.py, I got this massage and nothing happened:
sam@sam-HP-Spectre-x360-Convertible-15-bl1XX:~/msc/crazyswarm/ros_ws/src/crazyswarm/scripts$ python figure8_canned.py

sam@sam-HP-Spectre-x360-Convertible-15-bl1XX:~/msc/crazyswarm/ros_ws/src/crazyswarm/scripts$ python figure8_canned.py
Traceback (most recent call last):
  File "figure8_canned.py", line 8, in <module>
    swarm = Crazyswarm()
  File "/home/sam/msc/crazyswarm/ros_ws/src/crazyswarm/scripts/pycrazyswarm/crazyswarm.py", line 24, in __init__
    self.input = genericJoystick.Joystick(self.timeHelper)
  File "/home/sam/msc/crazyswarm/ros_ws/src/crazyswarm/scripts/pycrazyswarm/genericJoystick.py", line 39, in __init__
    self.js.open(0)
  File "/home/sam/msc/crazyswarm/ros_ws/src/crazyswarm/scripts/pycrazyswarm/linuxjsdev.py", line 227, in open
    self._js[device_id].open()
KeyError: 0

I upload both firmware nrf and main processor and crazyradio and actually everything works in rviz.
should I do anything more?
Thanks

whoenig, 2018-03-02:
That looks like you have some joystick attached, but it is has not the device id 0 (which is what we currently assume). What joystick devices do you have (`/sys/class/input/js*`)? If you have a joystick plugged in, which one is the one you want to use?

samandehghan, 2018-04-23:
Thank you whoenig
that problem solved and you were right. 
but i tried to run the original code again and this time I got this error. 
after I run python figure8 can.py, this error pop up:
Traceback (most recent call last):
  File "figure8_canned.py", line 8, in <module>
    swarm = Crazyswarm()
  File "/home/sam/final/crazyswarm/ros_ws/src/crazyswarm/scripts/pycrazyswarm/crazyswarm.py", line 19, in __init__
    import crazyflie
  File "/home/sam/final/crazyswarm/ros_ws/src/crazyswarm/scripts/pycrazyswarm/crazyflie.py", line 9, in <module>
    from crazyflie_driver.srv import Hover
ImportError: No module named crazyflie_driver.srv
would you please help me how to solve this one
Thanks
Saman

whoenig, 2018-04-23:
You need to source the ROS workspace in the shell you are trying to run the script from, i.e.,
```
crazyswarm/ros_ws$ source devel/setup.bash
crazyswarm/ros_ws$ cd src/crazyswarm/scripts
```

samandehghan, 2018-04-24:
Thanks Whoenig,
now I gt this error which I could not find any reason for that. 
terminal's output is that:
sam@sam-HP-Spectre-x360-Convertible-15-bl1XX:~/final2/crazyswarm/ros_ws/src/crazyswarm/scripts$ python figure8_csv.py
Traceback (most recent call last):
  File "figure8_csv.py", line 20, in <module>
    cf.uploadTrajectory(0, 0, traj1)
  File "/home/sam/final2/crazyswarm/ros_ws/src/crazyswarm/scripts/pycrazyswarm/crazyflie.py", line 83, in uploadTrajectory
    self.uploadTrajectoryService(trajectoryId, pieceOffset, pieces)
  File "/opt/ros/kinetic/lib/python2.7/dist-packages/rospy/impl/tcpros_service.py", line 435, in __call__
    return self.call(*args, **kwds)
  File "/opt/ros/kinetic/lib/python2.7/dist-packages/rospy/impl/tcpros_service.py", line 515, in call
    responses = transport.receive_once()
  File "/opt/ros/kinetic/lib/python2.7/dist-packages/rospy/impl/tcpros_base.py", line 727, in receive_once
    p.read_messages(b, msg_queue, sock) 
  File "/opt/ros/kinetic/lib/python2.7/dist-packages/rospy/impl/tcpros_service.py", line 353, in read_messages
    self._read_ok_byte(b, sock)
  File "/opt/ros/kinetic/lib/python2.7/dist-packages/rospy/impl/tcpros_service.py", line 336, in _read_ok_byte
    raise ServiceException("service [%s] responded with an error: %s"%(self.resolved_name, str))
rospy.service.ServiceException: service [/cf7/upload_trajectory] responded with an error: Could not find MemoryTypeTRAJ!

first I had 2 crazyflies and I after that i tested just one and I got this error. Should I change initial position?

whoenig, 2018-04-24:
Looks like you upgraded to the latest crazyswarm master, but didn't update your firmware? The new crazyswarm requires a different firmware (which is now largely identical to the latest official firmware).

samandehghan, 2018-04-24:
I updated firmware based on this master, I updated everything. should I get the firmware from somewhere else?
I followed the instruction in readthedocs.io and I updated the pre compiled firmware using of rosrun (second option). should I do anything else?

whoenig, 2018-04-24:
It should work fine with the method described on readthedocs. Could you double check that the binary you are using (https://github.com/USC-ACTLab/crazyswarm/blob/master/prebuilt/cf2.bin) is the one that was updated yesterday?

samandehghan, 2018-04-24:
you were right. the binary I was using was different from this one. but, now after I put new cf2.bin in its directory, I could not flash STM. time out error pop up, even, I clone it again and I did everything from the beginning but this error appear every time. It started to flash the STm but after couple of minutes, time out error appear. it has never happened and every time after I source devel directory, I could upload the ore compiled firmware. Am I doing something wrong?

whoenig, 2018-04-24:
This is strange - I assume your CF is close to the radio? If you own a JTAG debugger, you can also flash the CF over the debug connector. Another option is to use the DFU flash option (https://wiki.bitcraze.io/projects:crazyflie2:development:dfu).

samandehghan, 2018-04-24:
Thank you whoenig,
now everything is working and there is no error. just crazyflie steaks on the ground and it does not fly. can I fly them without using joysteak?

whoenig, 2018-04-24:
If you don't have a joystick you can try using one of the python scripts (e.g., niceHover.py).

zserlin, 2019-01-09:
How do you add multiple crazyflies to the hover_swarm code? I may be missing something but I have not found anywhere where this is explained. I have added them to the .yaml file but I dont understand how broadcast to them.

whoenig, 2019-01-10:
This issue is closed - please open a new issue.

There is no need to change the hover_swarm launch file. All you need to to is to update allCrazyflies.yaml *and* select the subset of CFs you are planning to use in chooser.py.