USC-ACTLab/crazyswarm #47: ImportError: cannot import name cffirmware

srnetsr, 2018-03-06:
Hi, I updated crazyswarm files, and try to run
`$ python figure8_canned.py --sim`
and get the error as shown in the following
```
Traceback (most recent call last):
  File "figure8_canned.py", line 5, in <module>
    from pycrazyswarm import *
  File "/home/asrl_ubuntu/crazyswarm/ros_ws/src/crazyswarm/scripts/pycrazyswarm/piecewise.py", line 1, in <module>
    from .cfsim import cffirmware as firm
ImportError: cannot import name cffirmware
```
Please help me to resolve this issue.

Thank you.

whoenig, 2018-03-06:
Did you build the firmware backend (either by running `build.sh` or by

```
cd ros_ws/src/crazyswarm/scripts/pycrazyswarm/cfsim
make
```
)?

srnetsr, 2018-03-06:
I did run "make" in the mentioned directory, but I see the error message
```
swig -python -I../../../../../../crazyflie-firmware/src/modules/interface cffirmware.i
cffirmware.i:12: Error: Unable to find 'avoidtarget.h'
Makefile:6: recipe for target 'swig' failed
make: *** [swig] Error 1
```

srnetsr, 2018-03-06:
I built the crazyswarm package by following [http://crazyswarm.readthedocs.io/en/latest/installation.html#simulation-and-physical-robots](url)
and executed```
crazyswarm/ros_ws$ catkin_make -k
crazyswarm$ ./build.sh
```

whoenig, 2018-03-07:
Should be fixed now. Thanks for reporting!

srnetsr, 2018-03-07:
I found some errors more.

When I run `crazyswarm/ros_ws$ catkin_make -k`,

**1.**
there are some errors which is related "Qualisys2Ros/include/RTProtocol.h & .cpp"
See the following which shows one of warnings. Others are all the similar to this.
```
In file included from /home/asrl_ubuntu/crazyswarm/ros_ws/src/crazyflie_ros/externalDependencies/libmotioncapture/externalDependencies/Qualisys2Ros/include/RTProtocol.cpp:4:0:
/home/asrl_ubuntu/crazyswarm/ros_ws/src/crazyflie_ros/externalDependencies/libmotioncapture/externalDependencies/Qualisys2Ros/include/RTProtocol.h:480:51: warning: ISO C++ forbids converting a string constant to ‘char*’ [-Wwrite-strings]
                            _TCHAR* tTrue = ("True"), _TCHAR* tFalse = ("False"));
                                                   ^
```

**2.**
This is about "crazyflie_controller" part.
```
/home/asrl_ubuntu/crazyswarm/ros_ws/src/crazyflie_ros/crazyflie_controller/src/controller.cpp:9:58: fatal error: crazyflie_controller/ExecuteTrajectoryAction.h: No such file or directory
compilation terminated.
crazyflie_ros/crazyflie_controller/CMakeFiles/crazyflie_controller.dir/build.make:62: recipe for target 'crazyflie_ros/crazyflie_controller/CMakeFiles/crazyflie_controller.dir/src/controller.cpp.o' failed
make[2]: *** [crazyflie_ros/crazyflie_controller/CMakeFiles/crazyflie_controller.dir/src/controller.cpp.o] Error 1
make[2]: Target 'crazyflie_ros/crazyflie_controller/CMakeFiles/crazyflie_controller.dir/build' not remade because of errors.
CMakeFiles/Makefile2:4668: recipe for target 'crazyflie_ros/crazyflie_controller/CMakeFiles/crazyflie_controller.dir/all' failed
make[1]: *** [crazyflie_ros/crazyflie_controller/CMakeFiles/crazyflie_controller.dir/all] Error 2
```

**3.**
The last one is about "mocap_helper"
```
In file included from /home/asrl_ubuntu/crazyswarm/ros_ws/src/crazyflie_ros/crazyflie_driver/src/crazyflie_server.cpp:56:0:
/home/asrl_ubuntu/crazyswarm/ros_ws/src/crazyflie_ros/externalDependencies/libobjecttracker/include/libobjecttracker/cloudlog.hpp: In member function ‘void libobjecttracker::PointCloudPlayer::play(libobjecttracker::ObjectTracker&) const’:
/home/asrl_ubuntu/crazyswarm/ros_ws/src/crazyflie_ros/externalDependencies/libobjecttracker/include/libobjecttracker/cloudlog.hpp:103:57: warning: format ‘%d’ expects argument of type ‘int’, but argument 2 has type ‘std::size_t {aka long unsigned int}’ [-Wformat=]
     printf("\n  %d  ------------------------------\n", i);
                                                         ^
/home/asrl_ubuntu/crazyswarm/ros_ws/src/crazyflie_ros/externalDependencies/libobjecttracker/include/libobjecttracker/cloudlog.hpp: In member function ‘void libobjecttracker::PointCloudDebugger::convert(libobjecttracker::ObjectTracker&, std::vector<boost::shared_ptr<pcl::PointCloud<pcl::PointXYZ> > >&)’:
/home/asrl_ubuntu/crazyswarm/ros_ws/src/crazyflie_ros/externalDependencies/libobjecttracker/include/libobjecttracker/cloudlog.hpp:139:57: warning: format ‘%d’ expects argument of type ‘int’, but argument 2 has type ‘std::size_t {aka long unsigned int}’ [-Wformat=]
     printf("\n  %d  ------------------------------\n", i);
                                                         ^
/home/asrl_ubuntu/crazyswarm/ros_ws/src/crazyflie_ros/externalDependencies/libobjecttracker/include/libobjecttracker/cloudlog.hpp:152:55: warning: format ‘%d’ expects argument of type ‘int’, but argument 2 has type ‘std::vector<libobjecttracker::Object>::size_type {aka long unsigned int}’ [-Wformat=]
      printf("Object vector size: %d\n", objects.size());
                                                       ^
```

whoenig, 2018-03-07:
1. Is a warning in third-party software, so we can't fix it.
2. Should go away once you build a second time (`catkin_make -k`). It's a missing dependency and we are working on fixing that in an upcoming version.
3. Are warnings that we are aware of and will fix in an upcoming version.

Thanks for the feedback!

closed 2018-03-08 by srnetsr