USC-ACTLab/crazyswarm #48: OptiTrack error with "No updated pose for motion capture object cf1"

srnetsr, 2018-03-06:
Hi,
I finished making cf2 fly with the following package
[](https://github.com/whoenig/crazyflie_ros)
(At this time, I used ros_vrpn package.)
and now I am doing integration cf2 with crazyswarm package.

However, I got some error as shown in the following
```
~/crazyswarm$ roslaunch crazyswarm hover_swarm.launch
... logging to /home/ubuntu/.ros/log/ebfd7e84-2176-11e8-b6bc-74d435137038/roslaunch-ubuntu-3516.log
Checking log directory for disk usage. This may take awhile.
Press Ctrl-C to interrupt
Done checking log file disk usage. Usage is <1GB.

started roslaunch server http://ubuntu:36713/

SUMMARY
========

PARAMETERS
 * /crazyflieTypes/default/batteryVoltageWarning: 3.8
 * /crazyflieTypes/default/batteryVoltateCritical: 3.7
 * /crazyflieTypes/default/bigQuad: False
 * /crazyflieTypes/default/dynamicsConfiguration: 0
 * /crazyflieTypes/default/firmwareParams/ctrlMel/i_range_m_z: 1500
 * /crazyflieTypes/default/firmwareParams/ctrlMel/i_range_xy: 2.0
 * /crazyflieTypes/default/firmwareParams/ctrlMel/i_range_z: 0.4
 * /crazyflieTypes/default/firmwareParams/ctrlMel/kR_xy: 70000
 * /crazyflieTypes/default/firmwareParams/ctrlMel/kR_z: 60000
 * /crazyflieTypes/default/firmwareParams/ctrlMel/kd_omega_rp: 200
 * /crazyflieTypes/default/firmwareParams/ctrlMel/kd_xy: 0.2
 * /crazyflieTypes/default/firmwareParams/ctrlMel/kd_z: 0.4
 * /crazyflieTypes/default/firmwareParams/ctrlMel/ki_m_z: 500
 * /crazyflieTypes/default/firmwareParams/ctrlMel/ki_xy: 0.05
 * /crazyflieTypes/default/firmwareParams/ctrlMel/ki_z: 0.05
 * /crazyflieTypes/default/firmwareParams/ctrlMel/kp_xy: 0.4
 * /crazyflieTypes/default/firmwareParams/ctrlMel/kp_z: 1.25
 * /crazyflieTypes/default/firmwareParams/ctrlMel/kw_xy: 20000
 * /crazyflieTypes/default/firmwareParams/ctrlMel/kw_z: 12000
 * /crazyflieTypes/default/firmwareParams/ctrlMel/mass: 0.032
 * /crazyflieTypes/default/firmwareParams/ctrlMel/massThrust: 132000
 * /crazyflieTypes/default/markerConfiguration: 0
 * /crazyflieTypes/large/batteryVoltageWarning: 11.4
 * /crazyflieTypes/large/batteryVoltateCritical: 11.1
 * /crazyflieTypes/large/bigQuad: True
 * /crazyflieTypes/large/dynamicsConfiguration: 0
 * /crazyflieTypes/large/firmwareParams/ctrlMel/i_range_m_z: 1500
 * /crazyflieTypes/large/firmwareParams/ctrlMel/i_range_xy: 2.0
 * /crazyflieTypes/large/firmwareParams/ctrlMel/i_range_z: 4.0
 * /crazyflieTypes/large/firmwareParams/ctrlMel/kR_xy: 30000
 * /crazyflieTypes/large/firmwareParams/ctrlMel/kR_z: 40000
 * /crazyflieTypes/large/firmwareParams/ctrlMel/kd_omega_rp: 100
 * /crazyflieTypes/large/firmwareParams/ctrlMel/kd_xy: 3.0
 * /crazyflieTypes/large/firmwareParams/ctrlMel/kd_z: 3.0
 * /crazyflieTypes/large/firmwareParams/ctrlMel/ki_m_z: 500
 * /crazyflieTypes/large/firmwareParams/ctrlMel/ki_xy: 1.0
 * /crazyflieTypes/large/firmwareParams/ctrlMel/ki_z: 2.0
 * /crazyflieTypes/large/firmwareParams/ctrlMel/kp_xy: 8.0
 * /crazyflieTypes/large/firmwareParams/ctrlMel/kp_z: 8.0
 * /crazyflieTypes/large/firmwareParams/ctrlMel/kw_xy: 9000
 * /crazyflieTypes/large/firmwareParams/ctrlMel/kw_z: 10000
 * /crazyflieTypes/large/firmwareParams/ctrlMel/mass: 0.38
 * /crazyflieTypes/large/firmwareParams/ctrlMel/massThrust: 5400
 * /crazyflieTypes/large/markerConfiguration: 2
 * /crazyflieTypes/medium/batteryVoltageWarning: 7.6
 * /crazyflieTypes/medium/batteryVoltateCritical: 7.4
 * /crazyflieTypes/medium/bigQuad: True
 * /crazyflieTypes/medium/dynamicsConfiguration: 0
 * /crazyflieTypes/medium/firmwareParams/ctrlMel/i_range_m_z: 1500
 * /crazyflieTypes/medium/firmwareParams/ctrlMel/i_range_xy: 2.0
[... 147 lines cut ...]
[ INFO] [1520365530.271920285]: Started 1 threads
[ WARN] [1520365530.272146565]: No updated pose for motion capture object cf1
[ WARN] [1520365530.272321530]: No updated pose for motion capture object cf1
[ WARN] [1520365530.272465823]: No updated pose for motion capture object cf1
[ WARN] [1520365530.272586391]: No updated pose for motion capture object cf1
[ WARN] [1520365530.272700633]: No updated pose for motion capture object cf1
[ WARN] [1520365530.272812226]: No updated pose for motion capture object cf1
[ WARN] [1520365530.272920487]: No updated pose for motion capture object cf1
[ WARN] [1520365530.273068233]: No updated pose for motion capture object cf1
```

The .launch file setting is 
```
# tracking
      motion_capture_type: "optitrack" # one of vicon, optitrack
      object_tracking_type: "motionCapture" # one of motionCapture, libobjecttracker
      # vicon_host_name: "vicon"
      optitrack_local_ip: "130.85.170.237" # ubuntu computer
      optitrack_server_ip: "130.85.170.251" # optitrack computer
      save_point_clouds: ~/pointCloud.ot
```

Please help me to resolve this issue.

Thank you.

whoenig, 2018-03-06:
The optitrack backend doesn't support using motive to track objects at this point (PRs are welcome; we do not have an optitrack system to test). You will need to switch directly to `libobjecttracker` as `object_tracking_type`. The documention has some pointers on how to define a marker configuration for your CF.

srnetsr, 2018-03-06:
I understood.

Thank you.

srnetsr, 2018-03-06:
I used the following setting in hover_swarm.launch
```
# tracking
      motion_capture_type: "optitrack" # one of vicon,optitrack
      object_tracking_type: "libobjecttracker" # one of motionCapture,libobjecttracker
      # vicon_host_name: "vicon"
      optitrack_local_ip: "***.**.**.***" # ubuntu computer
      optitrack_server_ip: "***.**.**.*** # optitrack computer
```
Also, I updated the marker arrangement information into crazyflieTypes.yaml using roslaunch crazyswarm mocap_helper.launch
Example)
```
markerConfigurations:
  "0":  # for standard Crazyflie
    numPoints: 4
    offset: [0.0, -0.01, -0.04] 
    points:
      "0": [0.000664575,-0.00346782,0.0394031]
      "1": [-0.0022814,-0.0702006,0.0210295]
      "2": [-0.0683127,-0.00555807,0.0219418]
      "3": [0.0629354,-0.00361527,0.0230691]
```

closed 2018-03-06 by srnetsr