USC-ACTLab/crazyswarm #49: CF2 does not fly (+ does not fly correctly)

srnetsr, 2018-03-07:
Hi,
I am integrating optitrack and cf2 using this crazyswarm package.
I think my setting is almost done.
However, CF2 does not take off.
```
<The setting of hover_swarm.launch>
     # tracking
      motion_capture_type: "optitrack" # one of vicon, optitrack
      object_tracking_type: "libobjecttracker" # one of motionCapture, libobjecttracker
      # vicon_host_name: "vicon"
      optitrack_local_ip: "130.xx.xxx.xxx" # ubuntu computer
      optitrack_server_ip: "130.xx.xxx.xxx" # optitrack computer
```
I executed the following commands.
```
$ python chooser.py
In chooser window, click "Clear" and "Fill", then click "reboot"
$ roslaunch crazyswarm hover_swarm.launch 
```
After that, CF2 appeared in rvis window. (There were two frames; "cf1" and "world")
Part of terminal window's line is shown in the following.
```
[ INFO] [1520450020.300528125]: [cf1] Update parameters
[ERROR] [1520450020.301070905]: Could not find param kalmanUSC/rstWithExtPos
[ERROR] [1520450020.303812255]: Could not find param traj/auto_yaw
[ INFO] [1520450020.349110276]: Update params: 0.065297 s
[ INFO] [1520450020.349302876]: Started 1 threads
```
(What is the param errors?)

Then, I push the start button of joystick, and terminal window showed 
`[ INFO] [1520450065.633917239]: Takeoff!`.
But, CF2 did not move at all.

I attach the LED patterns.
1. When I turn on crazyflie: [](https://youtu.be/4_la075u4G4)
2. When I reboot through chooser.py: [](https://youtu.be/PyFIcwxN7Fc)
3. When I run "$roslaunch crazyswarm hover_swarm.launch": [](https://youtu.be/KYxmz74xods) 

Despite of that I executed the following command
`$ python figure8_canned.py`.
Then, CF2 flew to the wrong side and crashed to the wall.
The terminal window outputted the following error message several times.
```
[pcl::IterativeClosestPoint::computeTransformation] Not enough correspondences found. Relax your threshold parameters.
[ WARN] [1520450711.869466057]: ICP did not converge!
[ WARN] [1520450711.869526159]: No updated pose for CF cf1 for 0.049901 s.
[ WARN] [1520450711.877692015]: Dynamic check failed
fitness: 0.00160511 >= 0.001
```
Then, it automatically went to emergency mode after crashing.
```
terminate called after throwing an instance of 'boost::exception_detail::clone_impl<boost::exception_detail::error_info_injector<boost::lock_error> >'
  what():  boost: mutex lock failed in pthread_mutex_lock: Invalid argument
[ INFO] [1520450713.327218320]: emergency requested...
[ INFO] [1520450713.328431185]: Done.
[crazyflie_server-2] process has died [pid 7083, exit code -6, cmd /home/asrl_ubuntu/crazyswarm/ros_ws/devel/lib/crazyflie_driver/crazyflie_server __name:=crazyflie_server __log:=/home/asrl_ubuntu/.ros/log/3df1d818-223d-11e8-a760-74d435137038/crazyflie_server-2.log].
log file: /home/asrl_ubuntu/.ros/log/3df1d818-223d-11e8-a760-74d435137038/crazyflie_server-2*.log
```

Please help me to resolve this issue.

whoenig, 2018-03-07:
1. When you choose reboot in chooser.py the crazyflie should spin up the motors again, just like in your first video. Note, that you can only use chooser.py while the server is *not* running (since only one process can use the Crazyradio at a time.) If it doesn't work, make sure you use the provided (or latest) NRF51 firmware.
2. The parameter errors you are seeing are caused by a wrong firmware version (STM32). Please flash the ones that are in the prebuilt folder (or build the provided firmware yourself).
3. In your last video the CF is already in some sort of error state (fast blinking of the red LED). This could again be related to the wrong firmware version or some other event that happened between the reboot and the roslaunch command.
4. I am not sure why the joystick would not work, but the script does - they simple send the very same command to the crazyflie_server. If you have an LED-ring, it will show you the communication status between the CF and the server.
5. For the crashed flight, did you check that the coordinate system is following the ROS standard? There are some settings in Motive about the coordinate system that you might need to adjust. A good test is to manually move the CF around while the server is running and verify in rviz that the position feedback is correct.

srnetsr, 2018-03-08:
- After re-flashing the firmware, the parameter errors is resolved. Thank you.
- I think the coordinate system is correct as shown in the figure.
[image]
- I executed
```
$ roslaunch crazyswarm hover_swarm.launch 
$ python figure8_canned.py
```
Doesn't cf2 hover at the first time?
I just executed commands and did not press any buttons.
But cf2 flew and crashed on the ground......it did not fly correctly...
FYI, I achieved hovering with [crazyflie ros package](https://github.com/whoenig/crazyflie_ros) with optitrack...

whoenig, 2018-03-08:
The LED pattern is fine. crazyflie_ros and crazyswarm use different ways on how they communicate with Motive. Your problem looks like your coordinate system is flipped. Make sure that forward is positive x (forward is where the CF's antenna is), left is positive y, and up is positive z. The current optitrack backend transforms the points from motive, so you need to make sure that rviz shows them correctly accordingly. There are settings in Motive to change the coordinate system (since I never used it myself I can't be more specific.)

srnetsr, 2018-03-09:
Hi,
thank you for your reply.

After you mentioned the frame, I think that is the issue in my system.
I put cf2 making the forward is positive x in "world frame"
It is shown in the figure in the following figure.

[image]

In addition, I moved cf2 to its front left side as shown in the following figure.

[image]

As you can see, its left is not positive y!!
That is the problem.............I am figuring out how to change this coordinate fram...
If you have any idea, then please let me know.
Also, if I found the way how to change y direction, then I will post it.

For your information, the optitrack program has its own setting for z-axis.
In "Streaming" menu, there is "Up Axis" option which can be chosen between "Z Up" and "Y Up".
I made successful flight using [crazyflie ros package](https://github.com/whoenig/crazyflie_ros) with [vrpn_ros_client](https://github.com/ros-drivers/vrpn_client_ros) and "Z Up" option.

Thank you.

srnetsr, 2018-03-09:
Interestingly, coordinate frame is weird...
I set "Z Up" in Motive program. and begun .launch file.
The initial point (origin) was obtained as
    points:
      "0": [-0.00246003,0.0672245,0.0241833]
      "1": [-0.00295699,0.00139483,0.0367536]
      "2": [-0.0654743,-0.000146568,0.023399]
      "3": [0.00273966,-0.0636448,0.0219547]
Then, I moved cf2 to front left side.
However, in rviz window showed the position of cf2 as the following figure
**(The red is x-axis, green is y-axis, and blue is z-asix)**

[image]

The output of mocap_helper.launch is
    points:
      "0": [-0.22823,0.321825,0.023681]
      "1": [-0.231844,0.256385,0.0374854]
      "2": [-0.294067,0.257335,0.023227]
      "3": [-0.228993,0.190902,0.0238054]

srnetsr, 2018-03-10:
Hi,
I just found what was wrong...the coordinate frame direction.
I recognized that the world frame is fixed, so I needed to rotate cf2 to the right direction.
I just confused it with when I used vrpn_ros_client package....
So I found right direction rotating cf2, then finally got the successful result with "python figure8_canned.py"

Thank you for all your support and development, @whoenig !!!
I really appreciate it.