USC-ACTLab/crazyswarm #50: [OptiTrack error] No updated pose for motion capture object cf1

srnetsr, 2018-03-23:
I have to discuss about the error "No updated pose for motion capture object cf1" when you are integrating cf2 with optitrack.

If you set `object_tracking_type` in hover_swarm.launch file as `libobjecttracker`
With this option, sometime optitrack camera misses one or more markers during its tilting.
When a marker is occluded its ID will change. That might be confusing ROS.

This happened many times to me with only four cameras, so cf2 fell onto the ground...

Hence, we need to set `object_tracking_type ` as `motionCapture` using the rigid body option...
When I use it, however, the error `No updated pose for motion capture object cf1` happened...

So what should I do resolve this issue?
Is there anyone who resolve this issue?
I would like to discuss about it and resolve it.

Thank you so much.

whoenig, 2018-03-25:
For optitrack the rigid body option is not yet supported (PRs welcome; we don't have a optitrack system to implement & verify). Thus, you'll need to either implement that support or rely on libobjecttracker. It might also be possible to implement a generic VRNP backend for rigid body only that could be used with any VRPN provider (such as VICON or OptiTrack).

I am not sure what you mean by "When a marker is occluded its ID will change"? Libobjecttracker can handle occasional missing markers as long as it is for a short period of time only (and not at the beginning).

srnetsr, 2018-03-27:
I meant that when a marker is missed by the camera, then the software, developed by OptiTrack Camera, assign new number to the marker.

I confirmed that during running figure 8 flight, at the beginning of its tilting, one marker is occluded by the other marker for few seconds, then the software assigned the new number to the marker.

I contacted to OptiTrack company, then they said the system setting is good, and they recommend to use rigid body option to resolve my issue.

I am figuring out the crazyswarm code, if you have any recommendation and advice then please let me know.

Thank you.

closed 2018-04-11 by srnetsr

murali2kk, 2018-04-11:
Hi @srnetsr 

I did not explore using the rigid body option much. If you got it working, please let me know what resolved the issue.

If I understand correctly, the libobject tracker should not depend on marker id. I did not have a problem with that part, but then I have never been able to fly it with the crazyswarm package that I tried last.

whoenig, 2018-04-11:
There is a PR for the rigid body option with explanation [here](https://github.com/USC-ACTLab/libmotioncapture/pull/3). I'll likely be able to merge and integrate it into the official documentation on Friday.