USC-ACTLab/crazyswarm #51: Qi

klauswq, 2018-03-30:
I had these error can not be solved, please help:

/usr/bin/ld: cannot find -lViconDataStreamSDK_CPP
collect2: error: ld returned 1 exit status
crazyflie_ros/crazyflie_driver/CMakeFiles/crazyflie_server.dir/build.make:360: recipe for target '/home/qi/catkin_ws/src/crazyswarm/ros_ws/devel/lib/crazyflie_driver/crazyflie_server' failed
make[2]: *** [/home/qi/catkin_ws/src/crazyswarm/ros_ws/devel/lib/crazyflie_driver/crazyflie_server] Error 1
make[2]: Target 'crazyflie_ros/crazyflie_driver/CMakeFiles/crazyflie_server.dir/build' not remade because of errors.
CMakeFiles/Makefile2:1930: recipe for target 'crazyflie_ros/crazyflie_driver/CMakeFiles/crazyflie_server.dir/all' failed
make[1]: *** [crazyflie_ros/crazyflie_driver/CMakeFiles/crazyflie_server.dir/all] Error 2
make[1]: Target 'all' not remade because of errors.
Makefile:138: recipe for target 'all' failed
make: *** [all] Error 2
make: Target 'default_target' not remade because of errors.
Invoking "make -k -j4 -l4" failed

srnetsr, 2018-03-30:
Hi,
did you put Vicon SDK in the crazyswarm folder?
If you did not, then please refer [https://github.com/USC-ACTLab/crazyswarm/issues/29]()
If you did, then please run `catkin_make -k` in the ~/crazyswarm/ros_ws/ folder as described in the above link.

klauswq, 2018-03-30:
Hi, i had put Vicon SDK in the correct crazyswarm folder
[image]

and i also tried run catkin_make -k  several times in ros_ws folder, but the error i mentioned before still exists.
[image]

whoenig, 2018-04-01:
I am not sure if that is causing your problem, but the crazyswarm folder should not be part of another ROS workspace (it comes with its own workspace in the ros_ws subfolder).

klauswq, 2018-04-01:
Wow！this fixes the problem! Thx

closed 2018-12-07 by whoenig