USC-ACTLab/crazyswarm #52: "enable_logging" function

srnetsr, 2018-04-05:
Hi,

what is "enable_logging" function for?

Can I check the flight history with this function?
If so, could you please explain how to use it?

Thank you in advance.

whoenig, 2018-04-05:
It can be used to enable logging on the CF. We only use it for controller tuning when flying a single CF. Then, you can select some logging variables (you will find the names in the firmware) like this: https://github.com/USC-ACTLab/crazyswarm/blob/master/ros_ws/src/crazyswarm/launch/hover_swarm.launch#L16. You can plot the data using rqt_plot (see https://github.com/USC-ACTLab/crazyswarm/blob/master/ros_ws/src/crazyswarm/launch/hover_swarm.launch#L67).

Logging should be disabled if flying multiple CFs because we do not have enough radio bandwidth to gather sensor data and update the pose information of all CFs.

srnetsr, 2018-04-08:
Thank you for your reply.

closed 2018-04-08 by srnetsr

AhmedA92, 2019-07-07:
Thanks for the information, another question is how to save those log data in a text file? let's say the position information.

whoenig, 2019-07-08:
(Please don't comment on closed issues and open a new issue instead.)
I suggest you use rosbag, and then the rosbag Python API to parse and convert the data to a text file.