USC-ACTLab/crazyswarm #55: allcfs.startEllipse() ingnorred

negvet, 2018-04-12:
In a script, after cf.setEllipse(...) and takeoff goes allcfs.startEllipse(). But nothing happen. All Crazyflies just continue hovering.
The same for pyramid.py and rotgrid.py scripts (i did almost no change).
To the point, I cannot setEllipse while simulation, the output is the following:

> ~/crazyswarm/ros_ws/src/crazyswarm/scripts$ python pyramid9.py --sim
> [[10, 12, 16, 18], [11, 13, 15, 17], [14]]
> Warning: No joystick found!
> Traceback (most recent call last):
>   File "pyramid9.py", line 95, in <module>
>     main()
>   File "pyramid9.py", line 61, in main
>     period = PERIOD)
>   File "/home/drone/crazyswarm/ros_ws/src/crazyswarm/scripts/pycrazyswarm/crazyflieSim.py", line 122, in setEllipse
>     e = self.planner.ellipse
>   File "/home/drone/crazyswarm/ros_ws/src/crazyswarm/scripts/pycrazyswarm/cfsim/cffirmware.py", line 747, in <lambda>
>     __getattr__ = lambda self, name: _swig_getattr(self, planner, name)
>   File "/home/drone/crazyswarm/ros_ws/src/crazyswarm/scripts/pycrazyswarm/cfsim/cffirmware.py", line 74, in _swig_getattr
>     return _swig_getattr_nondynamic(self, class_type, name, 0)
>   File "/home/drone/crazyswarm/ros_ws/src/crazyswarm/scripts/pycrazyswarm/cfsim/cffirmware.py", line 69, in _swig_getattr_nondynamic
>     return object.__getattr__(self, name)
> AttributeError: type object 'object' has no attribute '__getattr__'

whoenig, 2018-04-12:
It looks like I accidentally removed the ellipse support when re-basing the firmware.
I can re-add ellipse support, but I am not sure if it will be part of the official firmware.

negvet, 2018-04-13:
Let me know how to re-add ellipse support if it is easy, otherwise close the issue.
Thanks!

closed 2018-05-17 by negvet