USC-ACTLab/crazyswarm #56: Crazyflie crashes after flying ~14m

pPatrickK, 2018-04-12:
We try to fly the crazyflie for 15 meters on one axis at 0.3m altitude. After ca 14 meters the drone "jumps" higher and crashes into the ground. The drone is still tracked by the motion capture system.

We use the following script:
```
#!/usr/bin/env python

import numpy as np
from pycrazyswarm import *

Z = 0.3

if __name__ == "__main__":
    swarm = Crazyswarm()
    timeHelper = swarm.timeHelper
    allcfs = swarm.allcfs

    allcfs.takeoff(targetHeight=Z, duration=1.0+Z)
    timeHelper.sleep(1.5+Z)
    for cf in allcfs.crazyflies:
        pos = np.array(cf.initialPosition) + np.array([12, 0, Z])
        cf.hover(pos, 0, 25.0)

    print("press button to continue...")
    swarm.input.waitUntilButtonPressed()

    allcfs.land(targetHeight=0.02, duration=1.0+Z)
    timeHelper.sleep(1.0+Z)

```
The hall has a size of 30m x 15m x 10m. We use 38 cameras of the company Vicon for our setup. The dron is a Crazyflie 2. We use two computers, one for the motion capture and one for crazyswarm.

jpreiss, 2018-04-12:
We haven't seen bugs like this, but we also haven't flown quite so far from the origin in our experiments. A couple of questions:

1. Can you reproduce this bug in simulation mode (by appending `--sim` to your python script)?

2. Does the issue persist if you use vicon's rigid body tracking instead of libobjecttracker? (change `object_tracking_type` to `motionCapture` and create an object in vicon tracker with the same name as your Crazyflie, e.g. cf1)

whoenig, 2018-04-12:
I guess this is related to our [hardcoded position limit](https://github.com/USC-ACTLab/crazyflie-firmware/blob/crazyswarm-next/src/modules/interface/packetdef.h#L49) of +/- 8m. For your space change the value to at least 15, and recompile both firmware and the ros workspace. I believe this limit is not exposed in the simulator, so the simulation should still behave correctly.

jpreiss, 2018-04-12:
We should simplify this code to use integer number of millimeters...

whoenig, 2018-04-12:
That's already done [upstream](https://github.com/bitcraze/crazyflie-firmware/pull/310). This will, however, still limit the space to +/-16m, similar to your new fullPoseSetPoint.

pPatrickK, 2018-04-13:
That worked.

Thank you very much for the quick response

Best regards
Patrick

closed 2018-04-14 by pPatrickK