USC-ACTLab/crazyswarm #58: Latency to high

pPatrickK, 2018-04-12:
Hi guys,

the next issue i'm facing is that the latency is too high out of the default parameter in your release.
This error is rarely seen if we use just one cf, but is flooding if we us ~10 cf's.

[33m[ WARN] [1523545145.684234751]: Latency too high! Is 0.125496 s.[0m

If we change eg. the iteration count on the convergence crit. for the icp we can affect the latency.
Is there any experience regarding this topic on your side. 

If you need more information on this please let me know.

Best regards
Patrick

Ps.: 
The hall has a size of 30m x 15m x 10m. We use 38 Vicon cameras. We use Crazyflie's 2. We use two computers, one for the motion capture and one for crazyswarm.

Our main goal is to get a stable system for applied logistics research and fundamental research. Our far goal is to have a heterogeneous swarm with over 100 agents.

jpreiss, 2018-04-12:
Hi Patrick,  please set `print_latency` to true in your launch file so we can isolate which component is too slow.

1. If the vicon system is not calibrated well, the latency from vicon will increase.

2. Based on size of your vicon space, I would be concerned that our 7.9mm markers would barely be larger than one pixel in the vicon camera images. This could add noise that makes vicon tracker itself take longer to deliver the point cloud, or making libobjecttracker slower. Can you check?

3. Currently we use one thread per crazyradio, and each radio thread handles its own object tracking workload. So adding more radios will parallelize object tracking. But 10 on one thread should be fine.

4. You need to run crazyswarm on a fast PC.

pPatrickK, 2018-04-13:
Hi jpreiss,
we will check the calibration and the try the print_latency and more radios. 

We use 20mm markers made of styrofoam and foil (0.4g/marker). In Vicon we get a patch of ~6x6/ ~4x4 pixels, sometimes more. Is that enough ?

I have some picture attached in the other issue.

whoenig, 2018-04-13:
The pixel size should be fine. We use VANTAGE cameras and our markers result in 3 to 4 pixel big blobs in the image. Let us know what print_latency shows. How many radios are you currently using? 10 CFs should be easily handled by a single radio. What kind of machine are you running the crazyswarm package on? Also, make sure that you compiled in Release mode (which should be the case if you use our build script, see https://github.com/USC-ACTLab/crazyswarm/blob/master/build.sh#L43).

pPatrickK, 2018-04-20:
Hey,
we checked the print latency, output is:
```
[ WARN] [1524220193.849475963]: Latency too high! Is 0.012543 s.
Latencies
Data Delivered: 5.02072 ms
Data Synchronized: 0.271074 ms
Data Processed: 4.97974 ms
Delivered To DataStream: 0.16159 ms
Run All Groups: 12.4563 ms
Group 0 objectTracking: 1.18725 ms
Group 0 broadcasting: 11.1225 ms
Group 1 objectTracking: 0.9466 ms
Group 1 broadcasting: 2.49065 ms
Total 22.8894 ms
Avg 1539
10.6903,0,0,0,7.58169,1.23124,5.72822,1.01114,3.96166,0,0,18.272,
Latencies
Data Delivered: 5.33775 ms
Data Synchronized: 0.285125 ms
Data Processed: 4.88255 ms
Delivered To DataStream: 0.152516 ms
Run All Groups: 6.02901 ms
Group 0 objectTracking: 1.20379 ms
Group 0 broadcasting: 4.67049 ms
Group 1 objectTracking: 0.983904 ms
Group 1 broadcasting: 2.82945 ms
Total 16.6869 ms
Avg 1540
```
We use the release build.
We have two channels, one with 12 and one with 8 cfs.

Our hardware setup is:
Ubuntu 16.04 LTS (64 bit)
16GiB RAM
Intel Core I7-6700 @ 3.4 Ghz

Do you have some ideas why the latency is that high?

whoenig, 2018-04-20:
VICON latency looks great. Your processor also keeps up pretty well with ICP. It looks like in the one case where the warning occurred that the broadcasting (i.e., transferring data to the radio) took too long. Basically, if you run your VICON at 100 Hz, ICP and broadcasting need to be finished in < 10ms.

How are your radios connected? Avoid additional USB hubs and plug-in directly to your PC (of course, you can use extension cords.) Try different ports as well: we have one machine where the radios don't work very well on some of the ports for an unknown reason. Also, I assume it's a direct installation on not a VM?

You can also reduce the VICON frequency to give the radios some more time, but I think it is best to try to figure out the root-cause first.

jchen93nj, 2018-09-13:
Hi there,

I got a similar problem. I was trying to run hover_swarm.launch to test the connection but it kept giving me Latency too high warning. We are using Qualisys with 16 cameras, only one crazyflie 2.0 and crazyradio PA. Warning message looks like this:
[ INFO] [1536867496.980133178]: Wait for services...
[ INFO] [1536867496.980673375]: waitForService: Service [/emergency] has not been advertised, waiting...
[ INFO] [1536867497.001597144]: waitForService: Service [/emergency] is now available.
[ INFO] [1536867497.002482927]: Manager ready.
QTM RT Interface connected
Version set to 1.7
return 1
ch: 1
[ INFO] [1536867497.051313051]: Adding CF: cf13 (radio://0/80/2M/E7E7E7E70d, cf13)...
[ INFO] [1536867497.054288654]: CF ctor: 0.002902 s
[ INFO] [1536867497.062174755]: [cf13] Requesting parameters...
[ INFO] [1536867497.070055572]: Found variables in cache.
[ INFO] [1536867497.286106211]: [cf13] reqParamTOC: 0.231778 s
[ INFO] [1536867497.286126149]: Requesting memories...
[ INFO] [1536867497.294084791]: Memories: 4
[ INFO] [1536867497.305696896]: [cf13] Ready. Elapsed: 0.251364 s
[ INFO] [1536867497.305717448]: CF run: 0.251458 s
update commander/enHighLevel to 1
update ring/effect to 15
update ring/headlightEnable to 0
update ring/solidBlue to 255
update ring/solidGreen to 0
update ring/solidRed to 0
update stabilizer/controller to 2
update stabilizer/estimator to 3
update ctrlMel/i_range_m_z to 1500
update ctrlMel/i_range_xy to 2
update ctrlMel/i_range_z to 0.4
update ctrlMel/kR_xy to 70000
update ctrlMel/kR_z to 60000
update ctrlMel/kd_omega_rp to 200
update ctrlMel/kd_xy to 0.2
update ctrlMel/kd_z to 0.4
update ctrlMel/ki_m_z to 500
update ctrlMel/ki_xy to 0.05
update ctrlMel/ki_z to 0.05
update ctrlMel/kp_xy to 0.4
update ctrlMel/kp_z to 1.25
update ctrlMel/kw_xy to 20000
update ctrlMel/kw_z to 12000
update ctrlMel/mass to 0.032
update ctrlMel/massThrust to 132000
[ INFO] [1536867497.316151800]: [cf13] Update parameters
[ INFO] [1536867497.350535717]: Update params: 0.044782 s
[ INFO] [1536867497.350626561]: Started 1 threads
[ WARN] [1536867497.378275984]: Latency too high! Is 0.026925 s.

After setting print_latency to be true it looked like:
Avg 570
0,0,0,0,0.53028,0.190267,0.138763,0,0,0,0,0.53028,
Latencies
Run All Groups: 0.448249 ms
Group 0 objectTracking: 0.173632 ms
Group 0 broadcasting: 0.117917 ms
Total 0.448249 ms
Avg 571
0,0,0,0,0.530136,0.190238,0.138727,0,0,0,0,0.530136,
Latencies
Run All Groups: 0.453914 ms
Group 0 objectTracking: 0.15297 ms
Group 0 broadcasting: 0.126274 ms
Total 0.453914 ms
Avg 572
0,0,0,0,0.530003,0.190172,0.138705,0,0,0,0,0.530003,
Latencies
Run All Groups: 0.394687 ms
Group 0 objectTracking: 0.184795 ms
Group 0 broadcasting: 0.108825 ms
Total 0.394687 ms
Avg 573
0,0,0,0,0.529767,0.190163,0.138653,0,0,0,0,0.529767,

Any ideas of the issue please let me know. Appreciate so much!

Best,
Jun

jpreiss, 2018-09-13:
I think this is just a transient latency issue. It's normal to receive one or two latency warnings on initialization. Unless you are seeing latency warnings continually during operation, all is ok. Your steady state latency looks under 1ms which is really good.

jchen93nj, 2018-09-13:
> I think this is just a transient latency issue. It's normal to receive one or two latency warnings on initialization. Unless you are seeing latency warnings continually during operation, all is ok. Your steady state latency looks under 1ms which is really good.

Hi jpreiss,

Thank you for your quick response. I am glad to hear it's all fine.

Best,
Jun

closed 2018-12-07 by whoenig