USC-ACTLab/crazyswarm #59: Controlling CF flight using a python script

The-SS, 2018-04-27:
Hi

I am trying to write a python script that flies the CFs along a continues trajectory through multiple waypoints, but I am having some problems.

In `ros_ws/src/crazyswarm/scripts/figure8_csv.py` the script loads `figure8.csv` from the same directory. Is there a script in the package that generates the data in the csv file? 

Also, what is that data in the csv file? Is it the optimization coefficients/decision variables obtained from solving the quadratic problem min c<sup>T</sup>Hc+f<sup>T</sup>c   s.t. Ac<=b ([equation (11) in the link](http://www-personal.acfr.usyd.edu.au/spns/cdm/papers/Mellinger.pdf))? I have written a python script that generates those decision variables but when I use them in a python script similar to `figure8_csv.py` the CFs do not follow the designed trajectory. 

Thank you

whoenig, 2018-04-27:
The csv assumes a 7th order polynomial (one for x,y, z, and yaw). Each line is one piece, the columns are documented in the first line. You can generate such polynomials by solving a QP as in the link provided. Other methods also consider the dynamic limits during trajectory generation, e.g., work by Richter and Roy.

You can use the helper tools provided in https://github.com/whoenig/uav_trajectories. This includes a program that generates trajectories based on a list of waypoints and the dynamic limits (the hard leg work was done by people from ETH Zurich based on Richter and Roy's paper; I only provide a simple command line tool based on their work). This repository also includes a visualization script that can show you the curve as well as temporal profiles of velocity, acceleration, etc. The tools in that repository use the same format as the Crazyswarm requires. Finally, you can also use figure8_csv with the `--sim` parameter to see a visualization of your generated trajectory before flying.

The-SS, 2018-04-30:
That may have been the problem. The polynomials I generated are evaluated from [t<sub>i-1</sub>, t<sub>i</sub>] for i = 1, 2, ... n (n being the number of waypoints).

jpreiss, 2018-04-30:
> from [ti-1, ti] for i = 1, 2, ... n (n being the number of waypoints).

It's better for numerical stability to make each piece start at zero.

closed 2018-05-01 by The-SS

RozDavid, 2018-09-17:
Hi @whoenig,
I tried the uav_trajectories library and it works really well. Tough when generating a trajectory for fixed waypoints at every goal all the derivatives are 0 consistently that makes the flight really discontinuous. See my output attached:

[image]

I checked the crazyswarm project and found some matlab scripts for generating a time variable continuous trajectory for the figure8, circle etc. My only problem this script looks a bit complicated for me...
My question would be:

- Is there a setting for the uav_trajectory or other similar library that generates continuous trajectory for all derivatives thus making a smooth flight between the waypoints.

- Is there a documentation or more high level script for the matlab scripts for just entering the waypoints and constraints in time. Or is there a way to do that dynamically inside crazyswarm project?

- Or shall I just understand those and generate the path for myself?

All your works are highly appreciated and looking for your answer!